betaflight/betaflight #4394: CPU usage in osd

Bugg69s, 2017-10-19:
I can't find CPU usage in my OSD setup any more

Omibus f3

DanNixon, 2017-10-19:
I have a feeling it was removed as it was of little use.

CPU load does not vary much during flight and by the time you come to fly you have little need for it (i.e. you should have already verified that your choice of features, loop time, filters, etc do not put too much load on the MCU).

Bugg69s, 2017-10-19:
I Will like it, I have f3 board and I set it up to 8k 8k. Flies really good. But CPU usage is like 65%. I want to see if there are some spikes.

jflyper, 2017-10-19:
@Bugg69s You will not be able to see spikes with OSD, as values are only updated at 60Hz.

Bugg69s, 2017-10-19:
Oh, ok. But can i enable that CPU usage on OSD ?

wind0r, 2017-10-19:
Since Betaflight 3.1 CPU usage only is calculated when in CLI.
This saves a lot of cpu cycles but on the "downside" the OSD element got removed.
Which isnt even that usefull ;) Just check the CPU Usage in CLI. Real usage will be below that value.

Bugg69s, 2017-10-20:
Ok :)

closed 2017-10-20 by Bugg69s