betaflight/betaflight #4406: DYS F4 (Omnibus F4 target) - Can't assign a motor to RSSI pin

StarrrLiteNL, 2017-10-20:
I have a DYS F4 FC (it uses the Omnibus F4 target) with a broken motor output (motor 3).
I have tried to remap that motor to another pin to get my quad to fly again. I have managed to sucesfully remap motor 3 to the PPM pin using the resource command. The PPM pin does not have a DMA channel, so I cannot run DSHOT. I have resoldered the motor 3 wire to the RSSI pin (since that does have a DMA channel), but I cannot assign motor 3 to that pin.

The address for the RSSI pin is A00. As soon as I assign motor 3 to A00 ('resource motor 3 A00' and 'save' ) it will not stick. The weird thing is that motor 4 also gets lost. 
After I rebound motor 3 to A00 I see the following in the CLI:

'Resource' gives me: 
```
resource MOTOR 1 B00
resource MOTOR 2 B01
resource MOTOR 3 A00
resource MOTOR 4 A02
```

'Resource list' however gives:
```
A00: FREE
A01: FREE
A02: FREE
A03: FREE
A04: MPU_CS
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: FREE
A09: FREE
A10: FREE
A11: USB
A12: USB
A15: OSD_CS
B00: MOTOR 1
B01: MOTOR 2
```

As you can see, both motor 3 **and** 4 are missing now.

If I remap motor 3 back to the PPM pin (B14), **both** motors reappear in 'resource list':

```
A00: FREE
A01: FREE
A02: MOTOR 4
A03: FREE
A04: MPU_CS
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: FREE
A09: FREE
A10: FREE
A11: USB
A12: USB
A15: OSD_CS
B00: MOTOR 1
B01: MOTOR 2
B03: FLASH_CS
B04: BEEPER
B05: LED 1
B06: FREE
B07: FREE
B08: FREE
B09: FREE
B10: FREE
B11: FREE
B12: FREE
B13: FREE
B14: MOTOR 3
```

Something is not going right when remapping to A00.

StarrrLiteNL, 2017-10-20:
The same happens when I rebind motor 3 to UART3 RX (B14).
When I rebind it to the PPM pin , it works fine however.

When the motors don't appear in the 'resource list' command, blhelisuite will not recognise them as well, so I think it is not just a display bug in the resource command.

jflyper, 2017-10-20:
As the Betaflight github issues is not a place for user support, please consider using other user support options such as asking the manufacturer of the hardware you are using, RCGroups: https://rcgroups.com/forums/showthread.php?t=2464844, or Slack: https://slack.betaflight.tech or other user support forums & groups (e.g. facebook).

That being said, here are some information for you.
1. You can not assign pins that do not have valid timer entries configured. PA0 does not.
2. PA0 does not have a timer entry, because usable timer collide with LED strip.

and most importantly,

3. If your DYS F4 board has a stacking connector for 4-in-1 esc, there are motor signal pins there.
4. Here is the DYS F4 thread in RCG.
https://www.rcgroups.com/forums/showthread.php?2856541-DYS-F4-OSD-BEC-AiO-Flight-controller

closed 2017-10-20 by jflyper

StarrrLiteNL, 2017-10-20:
What is the easiest way to determine which pins have valid timer entries? Is there a CLI command to check that?

jflyper, 2017-10-20:
For a complete list, you have to read the corresponding source code.
For DYSF4PRO target, it is here
https://github.com/betaflight/betaflight/blob/master/src/main/target/OMNIBUSF4/target.c

jflyper, 2017-10-20:
FYI, the reason the motor 4 disappears is that motor initialization stops when it encounters an error while initializing motor 3 for not having a valid timer channel entry.