betaflight/betaflight #4407: Gyro overflow slew filter not working in BF 3.2.1

etracer65, 2017-10-20:
@martinbudden @gke After seeing many reports of YSTTM reports beginning to appear in the rcgroups thread I retested BF 3.2.1 on my yaw spin rig and found that the slew filter that deals with gyro overflow on the ICM206xx series is not functioning.  I'm not sure where the regression occurred after it was introduced and working in 3.2.0 RC3 (I think).

Attached is a blackbox log from the yaw spin rig for an affected flight controller (target AirbotF4, ICM20608G gyro).

[icm20608g.BBL.zip](https://github.com/betaflight/betaflight/files/1402604/icm20608g.BBL.zip)

Related issues/discussions:
https://github.com/betaflight/betaflight/issues/3959
https://github.com/betaflight/betaflight/pull/3909
https://github.com/betaflight/betaflight/pull/3950
https://github.com/betaflight/betaflight/issues/3893
https://github.com/betaflight/betaflight/pull/4058

I'm wondering if the problem is in the defines and not selecting all the affected gyros in gyro.c.

`#if ((FLASH_SIZE > 128) && (defined(USE_GYRO_SPI_ICM20601) || defined(USE_GYRO_SPI_ICM20689)))`
`#define USE_GYRO_SLEW_LIMITER`
`#endif`

In this case the ICM20608G doesn't appear to be selected but I'm not sure how the target represents the gyro (maybe it falls back to ICM20601?).  I believe this case was the same during the previous slew filter testing so maybe a change in the target gyro detection occurred?  Just tossing out ideas.

etracer65, 2017-10-20:
So I compiled a version with the directive disabled so that the slew filter would be enabled like this:

`//#if ((FLASH_SIZE > 128) && (defined(USE_GYRO_SPI_ICM20601) || defined(USE_GYRO_SPI_ICM20689)))`
`#define USE_GYRO_SLEW_LIMITER`
`//#endif`

This allowed the slew filter to function for the ICM20608G gyro and prevented the gyro overflow/inversion.  Blackbox attached.  So I think we need to revise the conditional compilation to ensure the affected gyros are includes, or possibly enable the slew filter all the time.  Ideally the slew filter only needs to be active for the affected gyros and the best case would be to enable it based on runtime selection of the actual gyro detected.  Doing by target is problematic as some flight controllers come with several gyro choices (or even multiple gyros).  The Airbot F4 I'm using is one of these.  There are variants with MPU6500, MPU9250, MPU6000 and ICM20608G.  There's even the dual gyro version that has both the MPU6000 and ICM20608G (selectable with a CLI command) - but this one uses target AIRBOTF4SD.

[icm20608g_slew_filter.BBL.zip](https://github.com/betaflight/betaflight/files/1403038/icm20608g_slew_filter.BBL.zip)

mikeller, 2017-10-20:
Thanks for looking into this, @etracer65. The ICM20608G is recognised as part of the MPU6500 driver:
https://github.com/betaflight/betaflight/blob/master/src/main/sensors/gyro.c#L223-L228

i.e. the most selective fix for this would probably be 

```
#if ((FLASH_SIZE > 128) && (defined(USE_GYRO_SPI_ICM20601) || defined(USE_GYRO_SPI_ICM20689) || defined(USE_GYRO_SPI_MPU6500)))
#define USE_GYRO_SLEW_LIMITER
#endif
```

Are you able to test this?

etracer65, 2017-10-20:
@mikeller 

>The ICM20608G is recognised as part of the MPU6500 driver

Well that's unfortunate since the MPU6500 doesn't have the overflow/inversion problem and doesn't need the slew filter.

I tested with the change as suggested and the slew filter was active.  I also tested with a MPU6500 and the slew filter didn't seem to hurt anything other than consume extra cycles in the gyro loop.

The gyro detection seems kind of a convoluted mess.  How is it going to behave with a flight controller like target AIRBOTF4SD that has a variant with both a MPU6000 **and** an ICM20608G and the active gyro is selected at runtime using the cli parameter "gyro_to_use"?

http://www.readytoflyquads.com/flip32-f4-dual-gyro-edition-mpu-icm20608mpu6000

mikeller, 2017-10-21:
Yeah, it's not ideal that the slew filter selection is only done at compile time. But the problem was discovered very late in the release process of 3.2, so this was the solution that could be done with the least risk of breaking other things.

Thanks for testing, @etracer65, I've opened #4412 to add MPU6500 to the list of gyro drivers using the slew filter.

gke, 2017-10-21:
Just fluked today to be curious about what happened with the YTTM issue. 

My slew filter's overhead is negligible and the  cleanest solution IMO is to ditch the "special case" define,s which make the code generally look appalling, and be done with it.  If we are still scrabbling for microseconds in this BS chase for faster is better then we have lost the plot. 

Trust me I am an engineer :).

ctzsnooze, 2017-10-22:
I am having lots of yaw spins on all my quads regardless of gyro type.  Quite a few fellow pilots having the same issues on MPU6000 and I was going to ask if the code would just disarm whenever the absolute rate on any axis exceeds 1900 for a period of time.  I never command that kind of rate, so it would signify hitting something.  If it persists for more than some arbitrarily short time, it's usually a yaw spin.  I'm having to disarm manually to save the quad far more often than I'd like, so a disarm might as well be automatic.  Probably instead of a disarm, a better solution would be having the gyro's forced to zero (and probably all PID's set to zero so as not to accumulate I from stick inputs), while in excess of some arbitrarily unacceptable value.  So if I clip a gate and the yaw spin goes insane then it spins but I have no authority and no PID accumulation until the absolute spin rate becomes something that might be manageable eg less than say 1700.  This is needed in addition to the slew rate error fix for 2060X gyros imho.

richard-scott, 2017-10-23:
#4412 has been merged, try build [#16](https://ci.betaflight.tech/job/Betaflight%20Maintenance%203.2/16/) or newer from the Betaflight Maintenance branch.

ctzsnooze, 2017-12-04:
OK so I and several other pilots still get these spins.  It has destroyed one of my quads and I was lucky to retrieve another.  I'm sort of really fed up with it.  I do not recall it happening ever with good old 2.9 code and I don't really know when it started.  But it is a major problem for me and several other pilots running release 3.2

I originally reported the issue with ICM20608.  I am not so good with code, but it seems to not be activated here at 73:

```
#if ((FLASH_SIZE > 128) && (defined(USE_GYRO_SPI_ICM20601) || defined(USE_GYRO_SPI_ICM20689) || defined(USE_GYRO_SPI_MPU6500)))
#define USE_GYRO_SLEW_LIMITER
#endif
```

That appears to not activate for the 20608.  

The issue may not be just an inversion of the gyro data.  It seems to happen also with MPU6000's and we know they do not invert.

Anyway could I have some advice on building some test code that for all F3/F4 boards would activate the slew limiter regardless of gyro type?  I was thinking of just removing all conditions on the #define USE_GYRO_SLEW_LIMITER line, so it was active always.  Would this be a big performance hit for F3 boards using MPU6000's?

I'd also like to be able to test a code change that would cut effectively stop the quad responding to extremely high gyro inputs, ideally any yaw exceeding dps.  The intent is, in the yaw axis, to not respond after inducing a very fast spin.  My logs show the quad never normally exceeding about 1200 dps during normal hard flips and rolls, and far less on yaw.  Yaw is such a weak drive I rarely exceed 400dps.   I would like to make it so that whenever abs(yaw) was above say 1000dps, ie after striking a gate, gyro on yaw was driven to zero.

so at 606 a cumbersome way of doing it could maybe be something like this?

```#if defined(USE_GYRO_SLEW_LIMITER)
int32_t gyroSlewLimiter(gyroSensor_t *gyroSensor, int axis)
{
    int32_t newRawGyro = (int32_t)gyroSensor->gyroDev.gyroADCRaw[axis];
    if axis == YAW{
       if (abs(newRawGyro) > (1<<14)) {
            newRawGyro = 0.0f;
        } else {
           gyroSensor->gyroDev.gyroADCRawPrevious[axis] = newRawGyro;
    }
    }
    return newRawGyro;
}
#endif
```

Note I'm sure the above code is wrong because the axis is an integer and I don't know which integer is yaw.  Also I don't know for sure if 1<<14 is equivalent of about 500 or 1000 dpi?

All help gratefully appreciated.

ctzsnooze, 2017-12-04:
PS I see that in pid.c there is an error rate constraint on yaw but only when crash recovery is on.  I do not fly with it on due to false triggering.  Perhaps I could modify that code for testing the above ideas?

ctzsnooze, 2017-12-04:
Could I just remove all the current crash recovery code and just do this?

```
if (axis == FD_YAW) {
	#define GYRO_POTENTIAL_OVERFLOW_RATE 1000.0f
	if (gyroRate > GYRO_POTENTIAL_OVERFLOW_RATE || gyroRate < -GYRO_POTENTIAL_OVERFLOW_RATE) {
		errorRate = 0.0f;
		axisPID_I[axis] = 0.0f;
	}
}
```

azyner, 2017-12-04:
Yep. I got two this weekend. ICM 20602. Default settings, outside of rates and switches. Acc on, no crash detection.
`Betaflight / MATEKF405 (MKF4) 3.2.2 Nov 11 2017 / 00:35:15 (77782e602) MSP API: 1.36`

martinbudden, 2017-12-04:
>OK so I and several other pilots still get these spins. It destroyed one of my quads yesterday and I was lucky to retrieve another from dense scrub last weekend. I'm sort of really fed up with it. I do not recall it happening ever with good old 2.9 code and I don't really know when it started. But it is a major problem for me and several other pilots running release 3.2

@ctzsnooze ,I agree I don't think this is solely a gyro inversion issue and I also don't think the slew filter is a full solution.

To solve this problem I need lots of black box logs showing the problem occurring. Despite my requests, I've received very few black box logs. Can you please post black box logs and ask the pilots you fly with to do the same.

ctzsnooze, 2017-12-04:
Unfortunately only my Bluejay's have logging, they are ICM20608, I can't log any of my MPU6000's.  And it is very unusual, I can fly quite a lot without a single instance, then suddenly many times, so you have to log every flight and clear the memory all the time.  It's not easy to get logs.  

I've been thinking that it could be emulated by putting very tiny props on a quad, props on upside down, balancing it on the battery and hitting it hard with a shoe.  Or something like that.  Trying to capture it in the wild is very frustrating.

ctzsnooze, 2017-12-04:
Martin can you check the crash recovery code?  It seems strange at the if !MPU6000 line:

`#if !(defined(USE_GYRO_SPI_MPU6000) || defined(USE_GYRO_SPI_ICM20649))
`
That seems to me to say 'if not MPU6000', or if ICM20649... but surely if not MPU6000 then that means all gyros except MPU6000.  I can't quite figure why the 'or ICM20649'.  

Most of mine have been with ICM20608 and when I look at the code in gyro.c that determines which gyros get the slew filter, it is somewhat fussy but sees to *exclude* the ICM20608.

And I have no confidence that while active the slew filter will itself stop I accumulating especially if the pilot does what they naturally would do, which is to attempt to yaw in the opposite direction to the spin.

Just for testing, I've replaced the crash recovery code in pid.c with this:

```
		if (axis == FD_YAW) {
			#define GYRO_POTENTIAL_OVERFLOW_RATE 1000.0f
			if (gyroRate > GYRO_POTENTIAL_OVERFLOW_RATE || gyroRate < -GYRO_POTENTIAL_OVERFLOW_RATE) {
				inCrashRecoveryMode = true;
			} else {
				inCrashRecoveryMode = false;
			}
		}
		if (inCrashRecoveryMode) {
			errorRate = 0.0f;
			// stops all P and D on all axes
			axisPID_I[axis] = 0.0f;
			// clears all I on all axes
			// quad becomes unresponsive, except for throttle, until yaw < 1000
		}
```

and will mess around with it a bit to see what happens.  Basically chops all PIDs whenever yaw spins more than 1000.  

I just need to be able to fly safely for now, I need a temporary fix that grounds the machine rather than making it uncontrollably fly away.

brucesdad13, 2017-12-04:
I will try to submit some BB logs too

martinbudden, 2017-12-04:
I try and have a look at the code this week.

As for just limiting the yaw, that was one of my original proposals. However one of the guys I was talking with said that commanded rates of 1800 dps are used by acro pilots. We need to solve the problem for those guys as well, so we really need to understand the underlying cause.

ctzsnooze, 2017-12-04:
Perhaps a more nuanced variant could be something that for yaw only cut the error by a factor of 10, allowing some yaw authority but severely limiting how much...

```
if (axis == FD_YAW) {
	#define GYRO_POTENTIAL_OVERFLOW_RATE 1000.0f
	if (gyroRate > GYRO_POTENTIAL_OVERFLOW_RATE || gyroRate < -GYRO_POTENTIAL_OVERFLOW_RATE) {
		errorRate = 0.1 * errorRate;
		axisPID_I[axis] = 0.0f;
	}
}
```

ctzsnooze, 2017-12-04:
There's only one or two LOS acro pilots in the world that actually generate those kind of yaw spins for tricks.  For them this is a non-issue, they can disarm and not lose it fairly easily.  If the limit is a CLI option then racers can enable it and those who don't want it can keep it off.

PS I figured a great testing methodology.   I can print some small diameter TPU props with lots of drag and very little thrust.  Like vertical blades, say 2" diameter, very little forward angle.  Enough to provide normal yaw authority, some small amount of pitch/roll authority, but not enough lift to take off even on full throttle.  TPU is so tough you could belt it and the props wouldn't care.

It should then be possible to put the quad on a smooth surface, balance it on the battery, arm, hit it with the heel of a soft shoe to make it spin, and see/log what happens.  Should fairly easily test out the performance of the yaw limiting code etc..

ctzsnooze, 2017-12-06:
Another way is to smoothly reduce yaw error by factor (2000-gyroRate)/2000.  This way for most slow yaws (normal flight yaw rates are less than 200), behaviour will be basically normal.  But if we hit a gate and get a 2000 deg/s spin, error becomes zero.  This would solve the problem smoothly, only negative I can think of is that people deliberately pushing for a >1500 deg/s spin might find it more reliant on I and less on P.

ctzsnooze, 2017-12-07:
I've 3D printed some 40mm diameter props with near vertical blades.  They should generate yaw but very little lift.  I'll mount polycarbonate extension under the motor bolts at right angles to the arm, that pokes out past the props.  I can then hit this hard without hitting the props, inducing a yaw spin.  The special little props won't make it take off but I will be able to log what happens.  Hopefully do this on the weekend.

ctzsnooze, 2017-12-09:
Here's a video of a test simple code fix that zeros all PIDs above 1900 deg/s on yaw in an otherwise current 3.3 build

https://www.youtube.com/watch?v=tLGrroys8jw

This is using the above prop, which gives quite good yaw authority and a tiny bit of lift.

You can see almost normal stick responsiveness to yaw inputs from the radio, then normal opposition to external yaw inputs in the normal range (I rotate it by hand).  I then spin it on the floor by hand, first briefly exceeding 1900, then not quite exceeding 1900, then well exceeding 1900.  In the first and last spin, the PID system goes to zip, throttle stays on, the quad passively spins with no motor speed-up, and only when the yaw spin drops passively below 1900 does the PID system attempt to slow it down (which it does, very quickly).

Code like this would absolutely stop all yaw spins whether due to gyro overflow or just because the quad is spinning 8000 degrees/sec after an impact and the system needs 5s of 50% throttle to slow down.  The quad would, after hitting a gate and spinning > 1900, just spin silently until it became controllable again.  The pilot could cut throttle quickly or slowly, and would not be fighting an insane climbing spin like I had here https://www.youtube.com/edit?o=U&video_id=l6WPUEqIYX8

ctzsnooze, 2017-12-09:
This is the code I put in PID.c, replacing/hijacking the current crash detection code (using its variables)

```
		if (axis == FD_YAW) {
			#define GYRO_POTENTIAL_OVERFLOW_RATE 1900.0f
			if (gyroRate > GYRO_POTENTIAL_OVERFLOW_RATE || gyroRate < -GYRO_POTENTIAL_OVERFLOW_RATE) {
				inCrashRecoveryMode = true;
			} else {
				inCrashRecoveryMode = false;
			}
		}
		if (inCrashRecoveryMode) {
			errorRate = 0.0f;
			// stops all P and D on all axes
			axisPID_I[axis] = 0.0f;
			// clears all I on all axes
			// quad becomes unresponsive, except for throttle, until yaw < 1000
		}

```

ctzsnooze, 2017-12-09:
[test_props.zip](https://github.com/betaflight/betaflight/files/1544363/test_props.zip)

STL files for the test props used in the video, I made mine from TPU for toughness.  Should enable testing of overflow states on FC's with logging.

ctzsnooze, 2017-12-09:
Here is a test with ICM20689 with stock 3.2; I cannot reproduce the unstoppable flyaway that I had with this exact same quad last weekend.  But we can hear the motors spinning for quite some time in an attempt to resolve a fast spin.

https://www.youtube.com/watch?v=_0Gbdy7Rag4

Here is the same quad, but with 3.3 and my anti-YSTTM code above in place.  The video shows normal motor responses to low rate external yaw influences, but if the spin exceeds 1900, the motors do nothing until it drops below 1900.  Note throttle control remains normal at all times.

https://www.youtube.com/watch?v=pf5y54bkEzY

ctzsnooze, 2017-12-09:
Yaw spin during a race with stock 3.2 and an ICM20689:
https://www.youtube.com/watch?v=TjtPiUq5y4s

Here are some deliberately induced spins (deliberate angled ground strikes), all with MPU6000 gyro, new code as above, all self-limiting:
https://www.youtube.com/watch?v=p6ylT_zK7g0

It seems like the motors pushed the quad up during these spins,  but I know for sure the PIDs would not have been active if it was spinning more than 1900.  I suspect it actually wasn't spinning all that fast, but still fast enough to climb a lot.  Perhaps I didn't cut throttle hard enough / quick enough?.  Definitely shows that the MPU6000 is not immune, and that the spin happens very fast.

I'll get some logs from a Bluejay once I move the FCs around.

etracer65, 2017-12-09:
Ok, there's a problem with the gyro overflow slew limiter.  Basically it's not doing anything - and I can't figure out why.  I set my spin rig back up to test with 3.2.2 and was surprised to find the gyro overflow happening again with the ICM20608G gyro (MPU6000 is still fine).  I assumed another regression in the gyro detection so I modded gyro.c to force the slew filter to be enabled.  Tested again and found it still wasn't working and I'm seeing gyro overflow/inversion.

So I started debugging and first confirmed that the `gyroSlewLimiter()` function was in fact being called by modding it to return a hard-coded value.  This behaved as expected - confirming that the function was actually running.

So next I thought maybe the threshold wasn't being consistently reached and the logic not being triggered.  From the blackbox that didn't appear to be the case.  Looking at the raw gyro data I was seeing > abs(16384) between readings which should trigger the code.  Even so, I modded the code to put the unfiltered yaw gyro data into the roll axis, and the slew filtered yaw gyro data into the yaw axis.  Then comparing them in the log we could see any effect of the slew filter.  No affect at all.  For some reason the filter is not triggering.  Here's the code change I used for testing:

`//        gyroSensor->gyroDev.gyroADC[Y] = gyroSensor->gyroDev.gyroADCRaw[Y] - gyroSensor->gyroDev.gyroZero[Y];`
` `
`// DEBUG! put the non slew filtered yaw gyro into the roll gyro`
`// Then the roll and yaw axis can be compared to see the effect of the slew filter`
`  gyroSensor->gyroDev.gyroADC[Y] = -(gyroSensor->gyroDev.gyroADCRaw[Z] - gyroSensor->gyroDev.gyroZero[Z]);`
`//`
`// FORCE the slew filter`
`//#if defined(USE_GYRO_SLEW_LIMITER)`
`        gyroSensor->gyroDev.gyroADC[Z] = gyroSlewLimiter(gyroSensor, Z) - gyroSensor->gyroDev.gyroZero[Z];`
`//#else`
`//        gyroSensor->gyroDev.gyroADC[Z] = gyroSensor->gyroDev.gyroADCRaw[Z] - gyroSensor->gyroDev.gyroZero[Z];`
`//#endif`

The screenshot is showing both the roll and yaw axis but they overlap exactly so it looks like only one trace.  The log is attached.

Target is AIRBOTF4, gyro is ICM20608G and the build is from the current commit in the 3.2.x maintenance branch.  I don't see any changes that would have affected this and it definitely was working before so I'm confused.

[image]

[debug y and z axis.BBL.zip](https://github.com/betaflight/betaflight/files/1545043/debug.y.and.z.axis.BBL.zip)

etracer65, 2017-12-09:
Actually looking deeper into the log I do see a difference between the slew filtered yaw axis and the unfiltered roll axis (that has yaw data).  At the end of the spin when things are re-reversing at 5.437 the slew filter seems to kick in and the traces diverge.  Not really sure I understand what's going on here as it doesn't seem to match the logic.  Not sure why the slew-filtered yaw axis stays "stuck" at -1740 dps even when the unfiltered data appears to settle down at the end around 5.727 seconds.

etracer65, 2017-12-09:
@ctzsnooze In the video of you clipping a gate and then spinning, it's hard to tell for sure but it looks to me like the spin might be on the roll axis and not yaw.  The slew filter (when working!) is only on the yaw axis.  There was at least one blackbox log I saw that appeared to demonstrate the ICM206xx gyro overflow on the roll and/or pitch axis so that's a possibility.  The monolithic construction of the ICM-series gyros would suggest that they'd have the same "flaws" for each axis but that's only a guess.  Assuming it's true we might not see it manifest as much as yaw because it would be harder to induce.  Also roll/pitch have so much more authority they would be able to recover quicker.  I'm working on making a new mount for my spin rig to mount the FC vertically to test the roll/pitch axis.

ctzsnooze, 2017-12-10:
Hi etracer - thanks for your input.  Yes the videos don't show a clean yaw spin, there's significant movement on the other axes.

Today I logged the 'bathroom floor test' rig with a BluejayF4 ICM 20608 with an SD card for logging.

First tests, logs here, are in release 3.2.  The quad behaved perfectly, just like the others.  All induced spins self-terminated, including spins >2000 degrees/sec. 

Standard logs show no inversion of the gyro signal as far as the PIDs are concerned.

But the logs were interesting... they show massive slow oscillation on pitch and roll during the spin.  Yaw signal is constant, but the quad is sort of precessing with sinusoidal coupled pitch/roll oscillation.  It's obvious that the pitch/roll PIDs can't control it - maybe due to inertial issues while spinning, the pitch and roll PIDs somehow augment and maintain that oscillation?  The motors are varying up to 70% throttle so this factor alone is driving average throttle up a lot.  It looks like this (note time scale is compressed.

[image]

This fits with what we see in the videos and also with the characteristic 'fluttering' noise these quads make when they take off upwards.

Here are the blackbox logs.  Both start off with some stick inputs, then a couple of small scale forced yaw movements, then several big spins.  In every big spin the gyro goes to 1997 and does not reverse.  To my way of thinking, the combination of massive pitch and roll variation on top of attempting to deal with a yaw spin definitely could cause what we see in the videos.  Take a look:

[YawSpinTest1and2.zip](https://github.com/betaflight/betaflight/files/1545323/YawSpinTest1and2.zip)

ctzsnooze, 2017-12-10:
As far as code not triggering, I had trouble seeing how line 73 would ever activate slew limiting for a 20608:

```
#if ((FLASH_SIZE > 128) && (defined(USE_GYRO_SPI_ICM20601) || defined(USE_GYRO_SPI_ICM20689) || defined(USE_GYRO_SPI_MPU6500)))
#define USE_GYRO_SLEW_LIMITER
#endif
```

I also wondered about some FC's being excluded by the flash size requirement?

ctzsnooze, 2017-12-10:
OK here is a log with my code from above, with trigger for return of normal PIDs at 1500.  Much less motor response to the oscillation, but still some - because, it's now obvious, I forgot to zero D, so D is still active.  Big spins last longer, but the big motor oscillations are of much shorter duration since the yaw rate they are trying to control is never greater than 1500.  Not an awesome solution, but sort of OK... I need to modify the code to cut D as well and see what happens to that oscillation without D.

[image]

Here's that log

[YawSpinCode1500V1.txt.zip](https://github.com/betaflight/betaflight/files/1545337/YawSpinCode1500V1.txt.zip)

ctzsnooze, 2017-12-10:
And finally I modified the code to knock out D as well on pitch and roll whenever yaw is spinning > 1500 deg/s.

This image shows zero PID responses during the spin, constant motor signal, the oscillations in pitch and roll clearly are just the way I spin it, no different with D on or off.

[image]

Here's another spin but this time I vary throttle while PIDs are off, kind of cool to hear the motors responding smoothly while the quad was spinning:

[image]

Here's the log of that experiment:

[YawSpinCodeV2.BFL.zip](https://github.com/betaflight/betaflight/files/1545347/YawSpinCodeV2.BFL.zip)

I'll go fly this tomorrow and make some videos with logging.  Any ideas before I do?

ctzsnooze, 2017-12-10:
PS - compare the motor uptime on these recent logs to the stock code logs about 5 posts back.  Stock code has motor pairs near full on for much, much more time, at least a whole second.  Enough to gain a lot of altitude (40m at 4 times gravity, easily done).  With this code, motors are on full for only about a quarter of a second, and late (after a period of time where the operator has full control over throttle and can cut to idle if they wish.

etracer65, 2017-12-10:
I determined that the gyro overflow slew filter is not working *unless you are running in 32KHz mode*.  So the overflow YSTTM protection for ICM206xx gyros isn't working for the majority of the users as most people don't fly in 32K mode with BF.

https://github.com/betaflight/betaflight/issues/4736

ctzsnooze, 2017-12-11:
Ha ha yes like me I'm not in 32k mode.  Fantastic work, thank you!

But I have finally logs which show the cause of YSTTM!  - and it's not inversion, it's something else, it suggests some other coding error in yaw P limit.  

Background - today I took my BluejayF4 ICM20608 2206/2600 175mm 40404 quad to the park.  Well tuned, very fast.  I generated multiple yaw spins (nearly lost the quad twice, broke three sets of props and 6 straps.  The code was BF3.3 built with my absolute PID limiter so if yaw gyro > 1500 all PIDs (now including D) go to zero.  I had checked it worked beforehand.

Anyway I was able to record two very long yaw spins, one that persisted indefinitely that I had to disarm out of, the other which I think resolved by itself but took a very long time to do so.  I also recorded shorter yaw spins which did not persist.  My code worked fine but did not prevent ongoing severe yaw spins because too high absolute yaw is *not* the problem. 

Here are images from the logs, both scaled inwards in time.

This is from the Event1 log, it shows a brief self-terminating yaw spin.  Gyro goes monotonically below -2000 (no reversal), my code sets PIDs to zero, quad yaw spins with some slow pitch/roll wobble.  I held throttle so it's at about 60%.  No PIDs mean no motor reaction during this time.  When yaw gets below 1500, there's a bit of a jiggle, all three axes quickly sort themselves out, I keep flying.

[image]

I think this is what would happen every time if the quad did not have yaw spin reversal issues.

Here is a similar self-terminating yaw spin in the same log where my code activates, the gyro does not reverse, and I quickly regain control:

[image]

But then later in the same flight I got a 'classic' irretrievable yaw spin.  The graphic below shows it compressed in time 30%.  Whole event about 2.5s from hit to disarm.  I had to disarm, it wasn't gong to stop.  

At 1:10.145 I hit the ground and get a near maximal positive yaw.  Very quickly the yaw spin sign reverses to maximal negative yaw.  My code stops PIDs completely for a short time.  But then the PIDs start to oscillate crazy on yaw.  It seems to me that as the spin fell below 1500, my code re-enabled yaw PID drive, but the direction of the PIDs was wrong, *SPEEDING UP THE SPIN*.

I think there has been a reversal.  Further, even though the absolute values of the signals were in the accceptable range, the sign reversal has persisted, causing the yaw spin to continue indefinitely.

If this is the case, slew code as it currently stands would not work.  It is based on the premise that once gyro signals come back below abs 2000 they will have the correct sign. It appears, at least with this gyro, that this assumption is false.  The sign reversal appears to persist, causing the infinite yaw spin.

It also seems like the yaw PID sum limit just isn't working, I'm seeing PIDsum on yaw of 1908 during the correctible yaw spin.  This code has set pidsum_limit_yaw = 400 which seems to be ignored.  If the PIDsum limit for yaw was not ignored, at least the rate of climb would not be so high.

It seems likely that after the reversal, gyro data that we would assume was 'normal' can remain reversed indefinitely.  This would not necessarily be detected by simple slew code as we have now.   

[image]

Finally I have a log of a reversal that (eventually) did spontaneously reverse.  This was almost a 4s event.  It is at 34s in the Event 2 log.

Here the initial yaw spin at 32:407 is negative, but the sign quickly reverses positive, and after a brief zero PID period exceeding 1500, the quad gets stuck in that reversed vibration effect from 32:491 to 33.065.  It gains a lot of altitude and is spinning fast.  But then at 33.065 the gyro sign suddenly reverses to negative max, and my code cuts the motors.  It is now passively spinning, and continues to do that until 35.313 - almost 2.3s passive spinning in the air!  But the gyro sign is now the same as the initial spin.  You can see I cut throttle early in the passive spin period, amazing it would hover so long.  Just before crashing, at 35.313, the spin rate falls below 1500, in the correct direction, and the PIDs regain control within 0.3s.

[image]

So...

To summarise:

Gyro reversal happens in bizarre ways, it's not just simple sign reversal with normal data on recovery of non-maximal values, on at least the ICM20608.
Simple slew rate filtering therefore won't solve the problem, low slew rate data can be inverted
PID sum limits appear not to be working
My yaw spin PID limit code is an improvement but is useless if the gyro reverses.

When the gyro reverses, it seems all subsequent data should be considered suspect until the sign/direction of the gyro data returns to the original sign/direction that existed immediately before the gyro reversed.  Programmatically this would be difficult I guess.  

The two logs in full are attached here:

[YSTTM_Events.zip](https://github.com/betaflight/betaflight/files/1546369/YSTTM_Events.zip)

Matthias7Song, 2017-12-11:
A guy I know of said that he never had YSTTM with BF3.2.1 for icm20602 when he set 32k gyro sampling.

ctzsnooze, 2017-12-11:
It's quite possible Matthias, thanks for that input.  

But I think the gyro inversion detection code needs to be somewhat more complex than before to be reliable enough.   How stupid could those gyro manufacturers be!  Sign reversal!  WTF were they thinking!!!!

etracer65, 2017-12-11:
@ctzsnooze  Read through https://github.com/betaflight/betaflight/issues/4736

When not in 32k mode the gyro overflow on ICM206xx is being filtered through a hardware LPF in the chip and this causes the extreme value flips to be smoothed and the slew filter (or any other gyro rate limiting) to fail.  We need to figure out some other solution to the overflow detection.  I proposed a rate over time algorithm as a starting point.  Alternately you can enable 32k mode but still use a gyro/pid rate of 8K/8K and then the slew filter will work.

Secondarily we need to get a blackbox log of the event happening on a MPU6000 gyro.  We know this gyro doesn't overflow so we need to see what's happening with the PID controller.  All my quads have MPU6000's and I've hit lots of gates at speed (trust me!) and never get more than a quick (<1 sec) spin and then can recover.

ctzsnooze, 2017-12-11:
Awesome.  Thank you.  I hope we can make a workaround.  Hardware filtering would explain the soft reversal appearance in those logs.  

Rate over time should be good enough.  But also, absolute magnitudes matter.  In the bad ones, it goes from one value >say 1500 to the opposite >1500 value within a certain time, say 20ms.  Both abs threshold and time are needed.  

I'll try 32k mode on the Bluejay to see if that prevents the ongoing ones.  It is very scary each ongoing one I stand a real chance of losing the quad forever, I don't like doing this at all.

I think the true unstoppable spin won't happen on the MPU6000.  I think they just get normal flyaways for a second or so, like the self-limiting spins in those logs.  With my code they are easy to control and only gain a little altitude.  

Maybe with an MPU6000, if the props got so bent and the spin was so bad on multiple axes that D feedback could cause multi-axis high amplitude wobbles, maybe that might cause infinite climbing behaviour.  I will put my latest code on the MPU6000 machines and just fly them until I get some spins.  

As you can see from my videos, it's easy to induce these spins.  Just fly close to the ground in an arcing turn, 50% throttle, with enough roll angle on the frame that a prop clips the ground.  When the prop strikes the ground, the spin starts.  Or you just gouge into the ground!  But more than 50% of the time, you get a yaw spin.  You can do this yourself if you want to risk smashing everything.  My quads are pretty remarkably tough as well as fast and agile  :-)

ctzsnooze, 2017-12-11:
Note also this reversal did not happen with simple fast spins on my bathroom floor.  

They were initiated by spinning with my hand, ie gradual onset.  

Perhaps getting the reversal requires a certain speed of onset of the gyro data change, and I didn't achieve that in the at-home testing.

ledvinap, 2017-12-11:
@ctzsnooze : Some remarks;.

1:10.145 event - from this it seems that direction reversal is happening at relatively constant yaw rate - you code is stabilizing rotation at this speed (ratio correct and reversed readings at 32k is balanced so that filtered value is close to 1500). This is also supported by ground spinning tests

Using 32k gyro and downsampling will probably introduce considerable aliasing noise - 32k is close to gyro vibration frequency (25, 27, 29 kHz). Low-pass filtering at 32k will handle this, filtering on gyro and fc will provide similar results, but will have some performance penalty

Even with YAW PID disabled and no gyro reversal, it is possible that gyroscopic forces/precession of fast spin will couple ROLL and PITCH PID in a way that causes major oscillation and flyaway. It may even saturate mixer and prevent (or severely limit) YAW slowdown with YAW pid enabled.

Yaw authority is quite low, it may make sense to apply quite radical low-pass (tens of Hz) filter to yaw PID. It may improve flight performance under normal conditions - less noise coupling into roll/pitch, not wasting power on motor inertia

etracer65, 2017-12-11:
The problem is that any logic that relies on the gyro data will fail.  When the overflow occurs on the ICM206xx gyros the data is invalid and unpredictable.  There isn't simply a clean inversion.  For example, here's a yaw overflow event captured showing the raw yaw gyro values (unfiltered and unscaled - the raw ADC values from the gyro).  At the onset there is a unpredictable sequence of values.  

[image]

@ctzsnooze Your log looks like a "smoother" inversion (seemingly never getting to > 2000 dps before onset) because you're looking at the gyro data after it's been through lowpass and notch filters.  Also you're only logging at 1KHz and as you can see the "noise" during the inversion is extremely high frequency.  Lastly the slew filter may have partially affected.

> Our current slew limiter is based on the premise that once gyro signals come back below abs 2000 they will have the correct sign.

This is not really correct.  The slew filter works by looking for an "unreasonably large change from the previous sample".  If the current sample is more than abs(16384) from the previous, keep the previous (last known good).  This seems to work well for the example shown above.  The slew filter prevents the inversion from getting to the PID loop.  Here's the same graph with the gyro that the PID loop sees (slew, lpf and notches applied).

[image]

The problem is that when the gyro is not running in 32KHz mode it appears to enable a hardware lowpass filter in the chip and this "smooths" the output before it gets to the flight controller.  As a result the slew filter can't reliably detect the event by looking at extreme changes from one sample to the next.  For example, here's a raw gyro capture of the event when not in 32K mode and the gyro sampling at 8KHz (logging also at 8K).  The first apparent transition from positive to negative is actually about 8 sample of "slope" with intermediate values coming from the gyro.  The "steps" are too small for the slew filter concept to work reliably.

[image]

I started another issue to examine and look for improvements to the slew filter (or come up with some other solution to dealing with overflow) here.  https://github.com/betaflight/betaflight/issues/4736

As I've said multiple times: The YSTTM events are known to be caused by the gyro overflow defect on the ICM206xx devices.  While MPU6000 devices don't suffer this overflow/inversion problem, some are reporting runaway events but I don't think we've ever gotten true blackbox logs of it occurring.

@ctzsnooze If you feel like there's a problem with PID_sum limits then I would recommend opening a separate issue for analysis and tracking.  This issue is getting quite long and was effectively closed long ago (simple gyro detection bug).

I would caution applying too much analysis in cases where an inversion happened with an ICM206xx gyro.  In these cases the PID controller is responding to garbage data resulting in a positive reinforcement of the undesired behavior.  Trying to apply logical limits based on the garbage data simply won't work.  It might be useful if we can identify an occurrence with a MPU6000 gyro.

ledvinap, 2017-12-11:
@etracer65: The 'signal' corrupted with 2^14 jumps in 32k sampling mode is quite small (and slowly changing), detection at this speed should be quite reliable ...

ledvinap, 2017-12-11:
One interesting thing in @ctzsnooze logs is that actual yaw speed is visible in roll/pitch axes - sine wave with phase shift dependent on direction. Maybe this can be used to damp yaw rotation in first phase (but extraction may be complicated) ...

etracer65, 2017-12-11:
@ledvinap  Yes, if the gyro is running in 32k mode the slew filter works fine.  But very few people run in 32k mode as it’s still “experimental” and much noisier.  Also trying to run 32/8k on a F4 is borderline.  So most people are running 8/8k with these with 32k disabled (the default).  In this config the slew filter doesn’t work due to the lowpass app in the gyro.

ctzsnooze, 2017-12-11:
Very complex.

Wow that gyro sure puts out a lot of garbage in 32k mode!  

I've set my quads with ICM206xx's to 32k gyro sampling mode and 8k8k looptimes.  Had to increase moron threshold from 16 to 64 to arm.  Will test and log maybe next weekend.  Looks like the slew filter etc should solve it.

As for the other issues, by trapping high yaw rates >1500 and disabling PIDs, the overall user experience after generating a big spin is, for non-inverting gyros, much easier on the pilot.  Controllable spin without climb.

I'll open an issue for pid_sum_limit not working.

ledvinap, 2017-12-12:
@etracer65: Yes ... 
One possibility is to implement optimized downsampler that will run at 32k. It's nontrivial, it will have performance implications and only gain will be fixing Invensense bug ...

ledvinap, 2017-12-12:
Does anyone know if Invensense confirmed the bug / posted some reply? Someone was contacting them when this issue was discovered...

etracer65, 2017-12-12:
@ledvinap It seems like the original contact attempt died out.  Thinking the same I actually sent an email to their technical support address yesterday morning  Haven't received any response yet.

brycedjohnson, 2017-12-12:
A decimation FIR filter is pretty cheap filter (in cycles) if you wanted to bring 32Khz down to less then < 8kHz and not have any aliasing.  It is probably what the chip is doing internally when you have it in 8kHz mode.  You only need to run the filter at the decimation rate and not at the 32Khz rate.  (Every 4 samples in the 32kHz to 8kHz example).

ctzsnooze, 2017-12-13:
If we could do the slew detection and then a simple smoothing filter, both in the 32k datastream - that would be awesome!

ledvinap, 2017-12-13:
From performance standpoint, it may be best to enable FIFO in MPU and decimate all samples available when running GYRO task ... This will lump reads into single transaction, removing some overhead.
But FIFO will introduce some complexity, it will take some time to hunt down all bugs ...

closed 2018-05-05 by etracer65