betaflight/betaflight #4409: Camera control - pin incorrectly attempts to send UP command

pitts-mo, 2017-10-20:
Camera control - pin incorrectly attempts to send UP command

While disarmed, zero throttle, and full upward throw on elevator stick camera control pin sends a signal that is somewhat compatible with a cameras up command. 

Testing other stick deflections while throttle stick is zero yields no change to camera control output. 

I realized this only after hours of troubleshooting camera control when I finally realized I had not noticed the manual indicates arrow presses for this feature were intended to be performed while the throttle stick was around 50%.

major setup info (symptom is still seen on different pins and even when this configuration is reduced to defaults with just SBUS, and Camera Control pin related changes):

-3.2.1 on HGLRC SPRacing F3 compatible with integrated MWOSD and current sensor
-Foxeer XAT600M (hs1177)
-FrSky 9xd+
-XSR sbus + telemetry
-tramp telemetry
-LED strip

```
# diff
# Betaflight / SPRACINGF3 (SRF3) 3.2.1 Oct 15 2017 / 19:48:10 (434db9a36) MSP API: 1.36
name QAV210
resource PWM 2 NONE
resource CAMERA_CONTROL 1 A01

feature -RX_PPM
feature -RSSI_ADC
feature -TRANSPONDER
feature RX_SERIAL
feature SOFTSERIAL
feature LED_STRIP
map TAER1234
serial 1 32 115200 57600 0 115200
serial 2 64 115200 57600 0 115200
serial 30 8192 115200 57600 0 115200
led 0 8,8::AT:0
mode_color 7 0 4
aux 0 0 1 1400 2100
aux 1 4 0 1400 2100
aux 2 26 0 1900 2100
aux 3 28 0 1400 2100
aux 4 35 3 1900 2100
set rssi_channel = 7
set max_aux_channels = 4
set serialrx_provider = SBUS
set min_throttle = 1015
set motor_pwm_protocol = DSHOT600
set small_angle = 180
profile 1

set d_pitch = 27
set d_roll = 25
rateprofile 0

set roll_srate = 85
set pitch_srate = 85
set yaw_srate = 85
```

jflyper, 2017-10-21:
@DieHertz I think this is causing this behavior
https://github.com/betaflight/betaflight/blob/v3.2.1/src/main/fc/rc_controls.c#L337-L339

Is this an intensional?

DieHertz, 2017-10-21:
That is intentional and I don't see how that it's a problem :)

pitts-mo, 2017-10-21:
Oh, Sorry, What is this for? Can you provide description of this feature segment for manual? 

Not understanding this output was intentional was a distraction that prevented me from catching that I overlooked center throttle and rudder is required for normal directional camera control.  Damn nasty habit of returning throttle to zero :-)

DieHertz, 2017-10-21:
Many cameras have a secondary menu accessible by holding up button. Back when it was not possible to register long presses with camera control someone added this command to access voltage OSD menu on RunCam Swift 2.
It's still not possible to register long clicks with software PWM, at least I don't have a solution for it yet, so we'll leave it alone for now.

closed 2017-10-21 by DieHertz