betaflight/betaflight #4410: OmnibusF4 Current Meter showing wrong values (3.2.1)

ps915, 2017-10-20:
I just flashed from 3.1.7 to 3.2.1 and realised that the current meter does not work correct.
It shows on the desk about 30Amps at a scale of 400 and disarmed!
[image]

[image]

mikeller, 2017-10-21:
Ok, this happened to me as well. Some findings:

- it is intermittent, the current meter that is reported is off by a fixed number for the duration that the craft is powered up, or it isn't;
- I have verified with an external current meter that it is the reported current that is off, and not some component having an overly high consumption;
- I have taken blackbox logs, and the current reported by blackbox is off too. This is significant, since blackbox uses unfiltered `amperageLatest` for reporting, which means that it is the ADC reading that is off, and not the LPF filtering.

Jugsy, 2017-10-21:
I've had this problem in the other direction, i.e. current reporting too low. Total usage showed as 1A after approx 30 seconds of aggressive flight.
3.2.1 on DYSF4PRO

adamtfc87, 2017-10-21:
Also have had this problem on 3.2.1 and on 3.2 (rarely shows up on 3.2 but it does) with the Betaflight F4

proximac2, 2017-10-22:
Having same problem with amp meter showing way to high value, 32A, without arming. Do not change with higer thrust, static at 32A. Voltage meter is spot on! Using Omibus F4 V3 on BF 3.2.1

Tbeal79, 2017-10-25:
i am having this kinda problem also, updated to the 3.2.1 on the bf3 and now the each cell volts and the main battery volts are blinking witch means its to low, and the low voltage display inside just says land now and its on all the time just like the other 2 are..... thing is my battery is fresh off the charger and i have tested it and its not wrong...... so there has to be something up with this. it didnt do that in the 3.2 update to me.
im not sure if i have the same issue with the omibus f4sd i have, thinking of it ill check now lol.
ok so just checked and the omibus f4sd dont have them problems its just the bf3 fc. so i guess ill have to use my osd on my run cam and just watch my volts till it is fixed once again, or wait till my alarm goes off on my remote

cptquad, 2017-10-27:
I have the same problem, but the displayed current is consistently low, for me its almost half of the actual current usage.

e.g. My quad hovers at around 8 AMPS it shows around 3 in the current readout.

AirBreak69, 2017-11-01:
@Jugsy 
You might have a look #3934 where i mentioned scale/offset for my DSYF4PRO that lead to realistic drawn capacity values during flight.

RipperDrone, 2017-11-01:
On both my latest Omnibus F4 builds, amps are reported consistently low (about 1/3 of former / actual values). Has been ok until BF 3.2, conversion seems flawed since.

FasterRider, 2017-11-02:
I'm having a similar problem on Omnibus F3: 
Voltage warning kicks in immediately after connecting the battery. Continuous beeping and "Land Now". The 1S battery is full, shows 4.2V. (There is no current meter.) I thought the issue was gone when I upgraded from 3.2 to 3.2.1, but it's not. Also sometimes the warning settings reset themselves. For now I can only fly with constant beeping and watch the voltage manually. Well, I might just disable the warnings completely.

mikeller, 2017-11-02:
@FasterRider: This sounds more like your `max_cell_voltage` setting is too low, and this is causing your 1S to be recognised as 2S. Try increasing it.

FasterRider, 2017-11-02:
@mikeller: This is how it is set:

set bat_capacity = 700
set vbat_max_cell_voltage = 43
set vbat_full_cell_voltage = 41
set vbat_min_cell_voltage = 32
set vbat_warning_cell_voltage = 35

Yet the warning goes off as soon as I connect the battery, at 4.2V. Sometimes it doesn't happen, but most of the time it does.

mikeller, 2017-11-02:
@FasterRider: Try `set vbat_max_cell_voltage = 44` then.

FasterRider, 2017-11-03:
@mikeller That really seems to have fixed it, at least it didn't occur when I connected the battery 5 times in a row - great, thanks!

mcgarrr, 2017-11-21:
Having the same problem on the Hglrc F428. Showing 29A just plugging in the battery.

mikeller, 2017-11-21:
@mcgarrr: I haven't had this problem in any recent build in about 2 weeks. What version are you using?

mcgarrr, 2017-11-21:
@mikeller using 3.2.1

mikeller, 2017-11-21:
Try 3.2.2.

mcgarrr, 2017-11-22:
@mikeller. Flashed 3.2.2, same results.

kicekilla, 2017-11-27:
@mcgarrr I didn't think the hglrc f428 stack had a current sensor?...

mikeller, 2017-11-27:
@kicekilla: Ok, odd. I haven't had this on my gear in a while, but seems that the problem is still there. Do you have it every time, or just sporadically?

Mbertu, 2017-11-28:
I've a problem on DYS F4. On the osd the current draw max is 85 amps (does not matter, props, motors and escs), the total mah consumed are consistent with the data of the battery charger. This problem was reported by many people on an Italian Facebook group. The same max value is shown by an Hglrc F4 v5, on Omnibus F4 too.

jflyper, 2017-11-28:
@Mbertu If you (and other Italians) are experiencing current meter saturation, then it is the hardware specification.

Mbertu, 2017-11-28:
@jflyper what do you mean with "hardware specification" ? The board can't handle more than 85 amps or the current sensor cant handle that?

jflyper, 2017-11-28:
@Mbertu 
It's a current sensor specification (characteristic).

Ref:
https://www.rcgroups.com/forums/showthread.php?2798055-Understanding-Current-Meters

You can stack your shunt resistor to increase the range.
https://www.youtube.com/watch?v=mdH64v_ZYG8

Mbertu, 2017-11-28:
@jflyper thank you very much!

jupegarnica, 2017-12-15:
same problem here.  current sensor reading 25A when the copter is disarmed.  
any solution yet?

jflyper, 2017-12-16:
@jupegatnica what is your fc?

jupegarnica, 2017-12-16:
Ómnibus f4 , the Wizard x220s stock

jflyper, 2017-12-18:
Is x220s equipped with a current sensor???

sros25, 2018-03-03:
Same problem, 30A when disarmed on a Hobbywing F4 (OmnibusF4). Using last Betaflight 3.3.0 Final release.

Any solution?
Thanks.

jflyper, 2018-03-03:
@sros25 Does your craft have a current sensor? If not, current meter would not work.

sros25, 2018-03-03:
@jflyper yes it has a 120A current sensor. Any idea?
Thanks!

jeijefr, 2018-03-18:
I have same issue after an updated of my HGLRC F428 "omnibusF4"  3.1.5 to 3.3 and the current sensor doesn't work correctly (ADC). 
Have you found a solution ?

RipperDrone, 2018-03-18:
HGLRC F428/438/440 don't have current sensor, so you are presumably running Virtual Current Sensor setting? I also noticed on my F428/F438/F440 builds that the scaling is way off, could not yet determine a reasonable setting. Same happens on std Omnibus F4 FCs, shown Amps are very low, have been higher on older BF firmware versions....

RipperDrone, 2018-03-18:
On your F428, try setting current scaling LOWER to obtain higher mAh and Amps readings. At least that's what an old WiKi entry says... (unit seems to be 10mV/A, so setting a higher scaling factor means higher voltage readings required per Amp, or the other way round lower Amps for a given voltage reading...)

jeijefr, 2018-03-18:
thanks for your quick response i will try this solution.

brousselle, 2018-04-06:
Same problem here with OmnibusF4v5 and current sensor on ESC "airbot typhoon 32 4 in 1"
i only have the lipo connected to ESC connected to FC. nothing else connected.
The board is showing 33A with scale at 300 and 43A at 400.
Tried it with BF 3.2.4, 3.3.0, 3.3.1.
Will post a video soon to show it up. Tried it with an omnibus F7 and same ESC and it's working as expected.

jflyper, 2018-04-06:
There has been a change in how offset is treated. If you are using offset to calibrate the current meter, you have to re-calibrate it again. There will be a wiki write-up or video coming.

More specifically, the offset is applied AFTER the scale is applied. (It was BEFORE the scaling.)

closed 2018-04-06 by jflyper

mikeller, 2018-04-06:
@Dimand Said he'd do a how-to / video about the new current sensor calibration.

Dimand, 2018-04-07:
It's on its way. Did not get a chance in the week and I'm away this weekend but I know a few people need it so still going to try and get it up asap.

Dimand, 2018-04-07:
If you want something sooner have a look at this Google sheet from one of my last calibrations.

https://docs.google.com/spreadsheets/d/1lkL-X_FT9x2oqrwQEctDsEUhgdY19upNGc78M6FfJXY/edit?usp=drivesdk

brousselle, 2018-04-07:
@Dimand you shoud have received a request access from me to your Google sheet.
I still wonder how we can set an offset of -33000 mV where max value seems to be -1600 ?
But i will wait for the wiki.

brousselle, 2018-04-07:
Thx for the access. But the question is still there:
my offset is way bigger than yours (-33000), and max value is -1600.
Morevover i don't understand why it works with omnibus F7 and not with F4v5 (current sensor is on the ESC)

Dimand, 2018-04-07:
Is it a hall effect sensor? If so the offset will be much larger.

Seems I forgot to enable link sharing. Dam phone apps, should be ok for everyone now.

The offset value has been made larger with a bugfix recently in #5467. There is still a 32 amp limit there which is unfortunate.

brousselle, 2018-04-07:
Well i don't know if this is a hall effect sensor: https://store.myairbot.com/esc/typhoon324in1.html.
But from my understanding the current sensor from the ESC gives a voltage that the FC reads to display the Amp consumption. So why the F7 says 1.5 Amp (with default params) and the F4 33A (also with default params) ?

bulby97, 2018-10-09:
> Same problem, 30A when disarmed on a Hobbywing F4 (OmnibusF4). Using last Betaflight 3.3.0 Final release.
> 
> Any solution?
> Thanks.

@sros25 Did you find any solution ?