betaflight/betaflight #4415: [Feature Request] Bring the GPS support for F3 back!

MB3Drift, 2017-10-21:
Building my rig for mid/long range on a BFF3 on 3.2.1, I was unable to activate the GPS in the configuration tab.
I found the problem: I'm on betaflight F3, and it seems that GPS support has been removed from all the F3 board!
I'm am hugely disappointed, and even more because I bought the BFF3 board thinking: "this is a Betaflight FC it should run absolutely every feature of the BF firmware", and it is not.
So I ask a question: since the Long Range trend is booming right now (thanks to the TBS Mini Crossfire), and since Betafight offers so many GPS informations in the OSD (coordinates, altitude, gps speed, home arrow, distance from home, etc...), WHY remove this features from virtualy HALF of the FC's of the market?
I saw it was work in 3.1.7, I flashed immediatly the 3.1.7 and the GPS is supported, but I don't have all of the top cool features of the 3.2 like dynamic filter and ironically all the OSG GPS features?
So I can have GPS working but no GPS OSD indicator (or a really few useless ones), or I can have no GPS support but everything I want on the OSD! Nobody find this ironic?
I saw that it was removed because of chip capacity, but can you not remove really useless features that nobody use? Like servo output support, and support for all useless aircraft like octoX, Wings, Airplanes etc?
I saw many comments saying that people should go using Inav, but we want fly quads so we fly Betaflight. People flying airplanes, wings and OctoX already use Inav or other firwares instead of betaflight, so why not make room here instead of GPS?

I'm really not into programming and I have a huge respect for the developpers, but I think that this is a bad choice. GPS Wings and airplanes don't need turtle mode, Dshot 1200, my 5inch GPS long range quad does!

Please bring back the GPS support for F3! Thnaks for reading if you made it to the end!

mikeller, 2017-10-21:
The answer to that is simple: Flash space limitation. All of the optimisations and the addition of faster protocols like Dshot and things like onboard OSD have caused the size of the Betaflight firmware to grow, and now we are at a point where the 256kB flash on the F3 aren't enough any more to contain Betaflight with all of the features. At some point when the flash on F3 was overrunning and something had to be removed, GPS was the least useful for racing that was left, and so GPS was removed. Sorry to say, but if you want race performance **and** all of the other features, you're going to have to upgrade to an F4.

MB3Drift, 2017-10-21:
I understand the Flash space limitation, and the fact that GPS was removed because of that. But it can be reintroduced in the code, and an other feature that si REALLY useless, like servo support or wings, octoX etc? Because if the GPS is really the least usefull, why to developpe such OSD features.
There is really NO other useless feature to remove for F3?? Servo support is really useless I think, more than GPS.
I'm kinda angry right know, I bought the FC 2 weeks ago because I loved it on my last build, my long range build isn't even finished yet and already outdated, had to change the FC, and on my other quad too...
Running iNav like I said is not even a thought, not having the Dshot and the other coolest features of Betaflight is not a option! Stay in 3.1.7 is the least painfull option...

wind0r, 2017-10-21:
Here you have a bff3 with gps but without dshot

[betaflight_3.2.1_BETAFLIGHTF3.zip](https://github.com/betaflight/betaflight/files/1404408/betaflight_3.2.1_BETAFLIGHTF3.zip)

jflyper, 2017-10-22:
> other feature that si REALLY useless, like servo support or wings, octoX etc

FYI, there are many users of servos and plane-like mixers.

OCTOX has a user, too. #3139

So, they are not completely useless.

gregm412, 2017-10-23:
can u give me a colibri race with gps and no dshot for my vendetta? i use gps on my long range vendetta

mikeller, 2017-10-23:
[betaflight_3.2.1_COLIBRI_RACE.zip](https://github.com/betaflight/betaflight/files/1406343/betaflight_3.2.1_COLIBRI_RACE.zip)

fiveangle, 2017-10-23:
Perhaps pointing F1 and F3 users toward an easy build environment is the answer ?  A pre-configured VirtualBox VM for example would be ideal.  Anyone know if something like that exists already ?

wind0r, 2017-10-23:
There allready is a preconfigured VirtualBox VM(vagrent) and a Docker Container for a long long time.

AndersHoglund, 2017-10-23:
We should have a binary plug-in system, and a Tetris game in BFC where the user can mix and match feature-plugs until it all fits on the FC.

mikeller, 2017-10-23:
I think if we went that way, we should look into a server based build system like OpenTX and nodeMCU are using. But I don't even want to think into the bug reporting / debugging aspect of this ('I've got a build for F3, with options X, Y, Z, but not A, B, and now option Y does not work as it should...').

MikeyZee, 2017-10-30:
I agree with bringing back this feature.
I am currently in the process of creating a HSC major project which will be a simple an easy to use flight log interpreter. Obviously black box render and things like that exist but I am planning on creating something a little different. Is there a way I can have the f3 omnibus (board I purchased for this project) activated with a GPS only for logging purposes on the black box? 
Obviously the flashing of INAV is an option but with flashing there is always a relatively high chance of bricking the board which is the last thing I would want to do at this point. 
It sounds like the whole feature on f3 boards has been disabled which is a shame and I understand that it is flash storage restraints but what I don't understand is the NAZE32 runs an f1 chip, which is extremely slow and nearly clocks out the CPU during flight with a few settings enabled in beta flight although, correct me if I am wrong, but GPS can be used on this? 
I don't understand that, or is there more flash storage on a really really old board?

Thank you, 
I look forward to the replies and I am currently gathering parts for my build and will need a solution soon, otherwise I may need to buy an F4/F7 board...

jflyper, 2017-10-30:
I believe there's no global resurrection of GPS feature on F3 with BF, at least until features become selectable by a user.

You can always build your custom firmware. Read few comments back.

DanNixon, 2017-10-30:
> with flashing there is always a relatively high chance of bricking the board

There is absolutely zero chance of bricking the board.

mikeller, 2017-10-30:
@MikeyZee: First of all, don't worry about bricking your board, this is almost impossible with STM32 based boards, since their boot loader is stored in ROM and therefore indestructible.
There is little to no chance of GPS coming back to F3. To the contrary, at the moment more non-essential features have to be removed to make space for features that are required for some users to make it work at all, like new and improved RC protocols.
What you could do is set up a build environment, and build the firmware from a custom target. If you remove all of the RC / telemetry protocols that you don't use, you should be able to get enough space to fit GPS. Or simply go for iNav.

MikeyZee, 2017-10-30:
@jflyper: ~ Ok that's fair enough, it's not on the high priority list at the moment which makes sense as betwflight is more tailored towards performance flight. I will have a look at the past comments and try some of them with the GPS enabled and potentially D Shot disabled. Ok that's good to know I guess, just trying to remove anything that can go wrong because I've been having waaaay to many instances of Murphy's Law recently. Thankyou for the insanely quick response! 

@DanNixon: ~ Awesome, thanks for letting me know, maybe I will experiment with iNav once my board comes in.

@Mikeller: ~ Thanks for the information, it seems like the hobby/sport is growing rapidly and betaflight is having to expand at the same rate and it's doing exactly that. Kudos to everyone keeping it updated and spotting bugs and performing fixes, it's what it's all about!
I'm learning a few languages at the moment and C is the next one on my list which would be very handy for this project. 

Thanks all for the clarification

arrowcircle, 2017-11-15:
@mikeller Could You please make a build of 3.2.2 for colibri race with gps code?

mikeller, 2017-11-15:
@arrowcircle: How about you try your hand at it yourself? There is lots of documentation for it around, like this: https://github.com/cleanflight/cleanflight/blob/master/docs/Customized%20Version.md

As for the build itself, you can use a self contained docker image for that: https://hub.docker.com/r/betaflight/betaflight-build/

martinbudden, 2017-11-15:
@arrowcircle , yes why not try yourself. It's not that hard and once you know how to do it, you'll be able to do it for 3.3 and later releases...

Werks89, 2017-11-16:
Is it not possible to just have a side note where the GPS slider is saying "Not supported on F3 boards"? I realize I'm whining and should have Googled sooner but man thats annoying after spending quite a few hours.

mikeller, 2017-11-16:
@Werks89: Happy to add one, can you please point out where you came across this statement?

arrowcircle, 2017-11-16:
@martinbudden @mikeller Thanks for the links.
Its very easy to compile custom firmware with docker.
I made simple bash script to apply changes to COLIBRI TARGET https://gist.github.com/arrowcircle/88945fbc0b723268eed5c7990371741a
Is it ok to disable so much features?

BTW, do I need to enable nav?

Werks89, 2017-11-16:
@mikeller What do you mean? I was just saying it would save a lot of frustration for others trying to add GPS on F3 boards running Betaflight after version 3.1.7.

richard-scott, 2017-11-16:
I appreciate why the code was removed, but maybe this sort of thing could have been managed much better.

From an end user perspective there has been no announcement of the removal of the GPS code from F3 boards in **ANY** of the 3.2.x [Release Notes](https://github.com/betaflight/betaflight/releases) only [this](https://github.com/betaflight/betaflight/pull/3280) can be found if you hunt about. The removal of **ANY** feature should be announced like the demise of F1 support, and scheduled in for a future date not just removed without notice. 

Also, it would be really great if the project adopted a [Semantic](http://semver.org/) versioning scheme. This way a release that is not 100% backwardly compatible - such as the removal of GPS code - can be given a totally new version number to avoid confustion.

This new _limited edition_ firmware should have been released as v4.0.0 and by doing that end users would know to look at the Release Notes to see what has changed. That's how the rest of the Internet does it :-)

wind0r, 2017-11-16:
GPS-Code never was removed it is still there :)
I agree with that such a change should be in the Release Notes.
Even better Configurator should warn Users when GPS is enabled but not compiled in, but this needs time and most developers spend there free time in more features and not user interface stuff.
Everyone can fell free to help making the User Experience better.

This Project allready uses Semantic Versioning for a long time based on Interface and Flight Code.
Based on Targets simply isnt possible with over 100 different Targets.
Also saying that 3.2 would be 4.0 because gps got removed isnt correct since 3.2.0 never removed gps for f3 it is just not compiled in by default.

mikeller, 2017-11-17:
Good work, @arrowcircle. You can disable any features that you don't need or want. Some features might be dependent on others, so disabling the underlying feature will only work if the feature that requires it is disabled as well. You don't need NAV, this code has not been maintained in a while, so it's probably not working well anyway.

@Werks89: I was just asking you to point out where you would like to have the "Not supported on F3 boards" added, so that I can look into adding it.

@richard-scott: Totally agreed that the GPS removal was not handled in the best possible way. What you have to understand is that there is a fundamental difference between the end of F1, which was a decision that was made by the core developers, and then announced, and the removal of GPS support from F3, which was a solution to the problem of the firmware not fitting into flash on F3. This problem was not planned for, it cropped up in the middle of the release process, and a solution had to be found, without time to make an announcement ahead of time. Re semantic versioning, since Betaflight is not a library or an API (but includes a number of APIs), the distinction between major and minor releases in SemVer is not as clear cut for Betaflight as it is for libraries or APIs. If we were to use 'backwards compatibility' as the deciding measure for major or minor releases, we'd end up with only major and bugfix releases, since no two non-bugfix releases are backwards compatible when it comes to the configuration API and / or MSP.

arrowcircle, 2017-11-18:
@mikeller Hi! I tried to flash built image, but cant activate GPS in the configuration. What I doing wrong? Maybe I need to enable/disable additional flags? Gist with path is in prev comment

mikeller, 2017-11-18:
@arrowcircle: Are you trying to build this from master (3.3.0) or v3.2.x-maintenance (3.2.3)? In 3.2.3, the define is called `GPS`, not `USE_GPS`. And did you assign a port to GPS before trying to activate the feature?

arrowcircle, 2017-11-18:
@mikeller I checked out v3.2.2 tag. I copied my diff from your image. I also tried it with hands.

mikeller, 2017-11-18:
@arrowcircle: You need to change `#define USE_GPS` to `#define GPS` in this case - a lot of the defines got renamed between 3.2 and 3.3.

arrowcircle, 2017-11-18:
@mikeller Thank You,  I will try this. What tag is better to get? v3.2.x-maintenance or 3.2.2?

mikeller, 2017-11-18:
`v3.2.x-maintenance` is the branch that is used for all 3.2.x releases, and it is constantly updated with the latest bugfixes (but no new features), so I'd go for that.

bavariangerd, 2018-01-20:
I bought an omnibus F3 because of it`s 3 UART`s to connect GPS, Bluetooth and telemetry and 
can`t get GPS working till i found, after many hours seaching, here the reason why.
Because I`m not as familiar as i should in programming, please can you give me an actual version
for the Omnibus F3 for Betaflight with GPS - i do not need Dshot.
Thank you

geokos77, 2018-04-23:
Hi folks, I have the same problem, I have an OMNIBUS F3 and wanted to make GPS work on latest betaflight versions. I tried to make a custom firmware but i'm still unable to activate GPS in the configuration tab. I've build with the default firmware choosen by docker which is 3.4.0, i've removed uselless receivers support I don't need and added **#define USE_DSHOT_DMAR** and **#define GPS**
Did I wrote something wrong ? (I checked that the size of the custom firmware is less than original). I've attached my target.h file so you can check it
thank you

[target.zip](https://github.com/betaflight/betaflight/files/1937059/target.zip)

mikeller, 2018-04-24:
@geokos77: No, that should be fine. Did you assign GPS to an UART on the 'Ports' tab and save **before** you tried to enable it on the 'Configuration' tab? If not it won't work.

Also, enabling `USE_DSHOT_DMAR` on OMNIBUS  is pointless (and uses up valuable space), since this board has a hardware configuration that does not get benefits out of using DMAR.

geokos77, 2018-05-24:
No one can help me with my build :( ?

Jakala91, 2018-06-21:
Hey guys,
could someone create a build of Betaflight 3.4 RC2 for the SPF3 with GPS?
Thanks in advance! :D

strepto42, 2018-06-24:
Hmmm, I'm guessing this issue will see a few more hits in coming weeks due to the introduction of GPS Rescue mode with 3.4.0...

GPS might well become a more common request for the old F3 boards. I guess I'm gonna have to set up a build environment at some stage and play code tetris... :/

(I note also that the configurator still doesn't provide clear feedback on this missing feature)

closed 2019-05-14 by mikeller