betaflight/betaflight #4420: Throttle Sticks arm/disarm delay

sedwo, 2017-10-22:
I've upgraded to BTFL v3.2.1 and am now experiencing an extended throttle stick arming delay of ~2secs. Previously when arming/disarming with the sticks it would be <1sec.

Anybody know what might affect this and if there are any parameters I can adjust for it?

Using FrSKY x9D+ with Rotorgeeks SSDF3 FC.

richard-scott, 2017-11-07:
What CPU% does your board run at?

I had similar slowness to usage when I had my Gyro and Pid Loops above 4K/4K on an F3 board.

sedwo, 2017-11-07:
F3 CPU seems to hover ~35%.  Still plenty of response time.  And it worked fine on BTFL 3.1.7.  Once I upgraded to 3.2.1 this delay issue arose.

richard-scott, 2017-11-09:
How do you know there is a delay in arming? What are you using to judge this?

sedwo, 2017-11-09:
After two (2) years of flying my quad the experience becomes very familiar and sensitive to its performance.  So while I don't have accurate timing tests, it's quite evident to me when I expect the props to stop spinning based on a DISARM signal and the results display a lag.

richard-scott, 2017-11-15:
I've just had it happen on my SP Racing F3 Evo. I am digging out black box data.

I was hovering at waist height and did a throttle punchout to see how the power was and it when skywards. I flipped the disarm switch, but it took a while to stop turing props. luckily it went straight up and came straight back down, but got to over the height of a 2 floor house before it desended.

I'm not liking BF 3.2.2 right now as that scared the shite out of me.

mikeller, 2017-11-15:
@richard-scott: Blackbox and `diff` will be helpful, thanks.

richard-scott, 2017-11-16:
Here you go:

[LOG00395.zip](https://github.com/betaflight/betaflight/files/1480581/LOG00395.zip)
[spracingf3evo-diff.txt](https://github.com/betaflight/betaflight/files/1480614/spracingf3evo-diff.txt)

Once the craft had been collected I noticed that the RX had disconnected so that was my initial thought about loss of control... There is a loss of RX signal indicated half a second or so after the suspect increase in throttle, but there is no log of me lowering the throttle. 0.5 sec is what the Stage 2 failsafe timeout is configured at so I can only assume that on the throttle increase the RX disconnected from its JST cable.

It is odd that the loss of RX signal isn't indicated earlier as soon as the Stage 2 timer starts counting down and it is also odd that there is no record of the failsafe being activated... or have I miss read things?

richard-scott, 2017-12-07:
@mikeller I had slow disarming in the BF Configurator (BFC) with an SP Racing F3 EVO board. I've since got an ANYFCF7 board and the issue is not there anymore. I am using the same TX and RX and the new board disarms in the BFC instantly, even before I've even taken my finger off the arm-stick on the TX. With the Evo board I could flick the switch and remove my finger from the switch before the icon went out in the BFC

mikeller, 2017-12-07:
@richard-scott: I would not take the time it takes for the disarming to show up in configurator as any indication of how long the disarm actually takes - the way the MSP protocol is handled is not designed to ensure updates happen in real time.

richard-scott, 2017-12-08:
@mikeller fair enough, but I used that as my basis for time as it was the only thing i could see. When disarming out in the field the same thing happens, there is a small gap (less than half a second) between landing and disarming the switch and the props stopping. F7 board is ok tho.

alwaystinkeringcom, 2018-02-10:
Just to chime in. I noticed this when I just updated to 3.2.3 from 3.1.7. Stick arm/disarm used to be almost instantaneous. Now it takes a second or two. Is there a new setting for arming sequence delay or something?

balrog-kun, 2018-07-06:
Yep, I imagine this is considered a feature but I'm noticing I'm breaking a lot of props because my disarms are not instantaneous anymore.  It's still present in latest code.  There's no setting to go back to the earlier behaviour judging from searching setting names in cli.

haslinghuis, 2022-05-08:
Closing as inactive

closed 2022-05-08 by haslinghuis