betaflight/betaflight #4421: Voltage reading faulty

alphat5, 2017-10-22:
Hi, 

I just updated to betaflight 3.2.1, everything works finy expect that the voltage reading is absolutly not precise. In Betaflight 3.1.7, the voltage reading was correct. I updated with the full chip erase option enabled.

Here is my dump code and a DVR for showing my error. I use a 4S LiPo fully charged (checked before flying).

Betaflight / BETAFLIGHTF3 (BFF3) 3.2.1 Oct 15 2017 / 19:28:25 (434db9a36) MSP API: 1.36
name MellowFPV
resource BEEPER 1 C15
resource MOTOR 1 A06
resource MOTOR 2 A07
resource MOTOR 3 B08
resource MOTOR 4 B09
resource MOTOR 5 B00
resource MOTOR 6 B01
resource MOTOR 7 A03
resource MOTOR 8 A02
resource SERVO 1 NONE
resource SERVO 2 NONE
resource SERVO 3 NONE
resource SERVO 4 NONE
resource SERVO 5 NONE
resource SERVO 6 NONE
resource SERVO 7 NONE
resource SERVO 8 NONE
resource PPM 1 B07
resource PWM 1 NONE
resource PWM 2 NONE
resource PWM 3 NONE
resource PWM 4 NONE
resource PWM 5 NONE
resource PWM 6 NONE
resource PWM 7 NONE
resource PWM 8 NONE
resource LED_STRIP 1 A08
resource SERIAL_TX 1 A09
resource SERIAL_TX 2 A02
resource SERIAL_TX 3 B10
resource SERIAL_TX 4 NONE
resource SERIAL_TX 5 NONE
resource SERIAL_TX 6 NONE
resource SERIAL_TX 7 NONE
resource SERIAL_TX 8 NONE
resource SERIAL_TX 9 NONE
resource SERIAL_TX 10 NONE
resource SERIAL_TX 11 B01
resource SERIAL_TX 12 NONE
resource SERIAL_RX 1 A10
resource SERIAL_RX 2 A03
resource SERIAL_RX 3 B11
resource SERIAL_RX 4 NONE
resource SERIAL_RX 5 NONE
resource SERIAL_RX 6 NONE
resource SERIAL_RX 7 NONE
resource SERIAL_RX 8 NONE
resource SERIAL_RX 9 NONE
resource SERIAL_RX 10 NONE
resource SERIAL_RX 11 B00
resource SERIAL_RX 12 NONE
resource LED 1 NONE
resource LED 2 NONE
resource LED 3 NONE
resource RX_BIND 1 NONE
resource RX_BIND_PLUG 1 NONE
resource SPI_SCK 1 B03
resource SPI_SCK 2 B13
resource SPI_SCK 3 NONE
resource SPI_MISO 1 B04
resource SPI_MISO 2 B14
resource SPI_MISO 3 NONE
[... 480 lines cut ...]
set d_level = 75
set p_vel = 55
set i_vel = 55
set d_vel = 75
set level_limit = 55
set horizon_tilt_effect = 75
set horizon_tilt_expert_mode = OFF
rateprofile 0

set rc_rate = 175
set rc_rate_yaw = 165
set rc_expo = 0
set rc_expo_yaw = 0
set thr_mid = 50
set thr_expo = 0
set roll_srate = 70
set pitch_srate = 70
set yaw_srate = 65
set tpa_rate = 10
set tpa_breakpoint = 1650 

The DVR Record: https://youtu.be/XmTz7vsr1VI.

Thanks,
Kind regards,

Arakon, 2017-10-22:
Go back to 3.1.7, see if it still happens. If yes (which is likely), your battery died or you have i.e. a bad connection. 
Voltage reading is working and accurate here on 3.2.1 and 3.3.0 just as before.

alphat5, 2017-10-22:
@Arakon 
Hi, thanks for you're reply. My batteries are new and after the fly I checked them with my battery checker. It gave the correct voltage and showed 4S and each cell was at the correct voltage. Normaly I fly with "mah draw" and stop @ 1100mah drawn for a 1500mah. Each cells was at 3.8V.
I mentioned that at 3.1.7 it worked fine. For me this is a FW issue. I can go back to 3.1.7 but I all ready know that it will show the correct voltage...

Arakon, 2017-10-22:
Considering noone else reported this issue so far, it's not likely to be a FW issue. The only thing your video really shows is that the voltage is sagging fast and hard, which can be just as easily weak batteries. It's pretty normal for even a bad lipo to recover quickly to a decent resting voltage once powered off.
What batteries are you using?

bigb2000, 2017-10-22:
It seems that the FC detects 5s battery and divides the battery voltage by 5 (instead of 4). Anyone knows how Betaflight detects cell count?

bigb2000, 2017-10-22:
Does anyone know what vbat_detect_cell_voltage parameter does?

mikeller, 2017-10-22:
@bigb2000: `vbat_max_cell_voltage` ('Maximum Cell Voltage' in the configurator) is used to detect the cell count. You need to increase this if the detected cell count is too high with your batteries.

mikeller, 2017-10-22:
@alphat5: After looking at the video, not sure if there is anything wrong. The filtering frequency of the  battery voltage that is displayed was increased in 3.2, which means that now sagging of the battery is less hidden than it was before, which looks like what you are experiencing.

alphat5, 2017-10-22:
@Arakon I use Tattu Racing Series with an IR of 7 and Acehe Green with an IR under 7. My LiPo are not defective ;)

---
@mikeller I don't understand realy understand what you're saying? But otherwise, gona check vbat_max_cell_voltage ;).

alphat5, 2017-10-22:
Issue seems to be fixed. 

I chaged the `vbat_max_cell_voltage = 42` to `vbat_max_cell_voltage = 43`. Without arming my quad, it correctly shown the total voltage and the cell voltage  . I will do an test, but for the time beign I close the issue.

Thanks for the help given.

closed 2017-10-22 by alphat5