betaflight/betaflight #4422: how to map port H?

johnty, 2017-10-22:
I'm using [this](http://michaelwhalen.co.uk/gadgets/quadcopter-parts/betaflight-3-1-0-f4-flight-controller-integrated-compatible-flysky-receiver-20x20mm-built-5v-bec/) FC. it was bought from banggood and now only the FrSky version is available.

The LED_STRIP pads on this guy, after some poking around, seems to be on port *H* 0. How would I access this resource? Would a custom compile be required to specifically address that port, as CLI doesn't seem to parse H00 for the resource command.

mikeller, 2017-10-22:
Pretty sure port H is not (yet) supported in Betaflight per se, even with a custom build. Ask the supplier for the source for the firmware for this build, to get clarification for what pin is used.

jflyper, 2017-10-22:
@johnty You clearly have mis-read the pin or mis-understood the orientation of the chip, because PH0 and PH1 are used for oscillator input/output.

[image]

johnty, 2017-10-23:
@jflyper AH, indeed! looks like the pin in question is actually PC12. Thanks!

Arakon, 2017-10-23:
D02 doesn't have a valid timer, I believe. There are a few omnibus F4 clones around that have the LED pad mapped to the wrong pin. You could probably assign TX1 if available, or PA15 to it, and solder directly to the MCU.

johnty, 2017-10-24:
@Arakon, thanks! That could possibly explain why the board doesn't seem to boot properly if I try to assign LED_STRIP to D02. I get no response on boot, and then after a while  I hear the USB plugged-in sound in Windows,  but the COM port doesn't appear and I get a USB device malfunction error. On OSX it intermittently lags the USB bus (the MacBook's kbd/mouse hangs intermittently). Will try to direct solder to the MCU if I really want LED functionality with this board, and in the mean time will try another one.

Thanks everyone for the comments!

closed 2017-10-24 by johnty