betaflight/betaflight #4423: Ability to disable the Summary Screen on Betaflight OSD

AndrewFarley, 2017-10-23:
## Problem
The summary screen is front-and-center in the screen and can be obnoxious, especially for simpler mini-quads who you really only rely on 1-3 pieces of information which are already on your OSD.  In this case, the summary screen just re-shows this same information that is already on the OSD menu, but annoyingly in the middle of your viewport, making the summary screen quite useless for a percentage of us.

## Desired Functionality
Some way to disable the summary screen, and simply keep showing the regular in-flight OSD.

## Ways to accomplish this
I propose one of two ways to accomplish this... 
1. The most obvious to me, is if all items are un-checked in the summary area, then it does not show the summary screen.  Currently (in 3.2.0) if you un-check all summary items, it just shows the summary screen with no data (hilariously so).
1. A secondary way would simply be a "toggle" of some sort (which is accompanied by a flag in the cli) which disables the Summary screen altogether.

The first approach above is the least-impact and would require almost no code changes, no new interface elements in the configurator, and no new CLI key to set/unset, and would probably make for a more fluid user-experience.  Because, if you un-check every summary item, it should just not show the summary instead of an empty screen eh?  😄

mikeller, 2017-10-23:
Yeah, 1. sounds like the obvious choice. If you don't want any information on the stats screen, it should not be displayed. :grin:

McGiverGim, 2017-10-24:
Done, using the method 1 ;)

AndrewFarley, 2017-10-24:
Haha, thank you sir!  :)  +1 @McGiverGim

etracer65, 2018-05-05:
Note that you can also set up the "OSD DISABLE SW" mode and when that's active (OSD disabled) it will also hide the post-flight stats screen (addresses point 2).

closed 2018-05-05 by etracer65