betaflight/betaflight #4425: reciever issue.

JasonLavender, 2017-10-23:
Hello,
I recently purchased the R-XSR and the Taranis X9D from aloft hobbies, and I am setting up betaflight for my quad. However, when I click on the receiver tab, the bars for the inputs do not move when I use the sticks. Under configuration, I made sure that I put in a serial based receiver and SBUS. I am using the R-XSR. I have bound the receiver with the transmitter (X9D) and the tx shows that it is connected with the receiver. Do you have any ideas on how to fix this? Also, I am using the Femto F3 by Emax. 
Thank you,
Jason

mikeller, 2017-10-23:
@JasonLavender: GitHub is for questions and discussion around the **development** of Betaflight. For questions around the configuration and use of Betaflight, please visit [RCGroups](https://www.rcgroups.com/forums/search.php?searchid=62637789&query=betaflight%20configuring%20r-xsr).

closed 2017-10-23 by mikeller

stk21, 2017-11-25:
I am having the same issue.  Was there a solution to this?  The like to RCGroups has nothing.

stk21, 2017-11-25:
Actually.  This fixed it.  https://www.youtube.com/watch?v=d9cIg5RZrWo

Provided 3-wire harness from FrSky has the pins wrong.