betaflight/betaflight #4426: Vtx Power levels reset back to 25mW at boot

HFMan, 2017-10-24:
When using the OSD to change VTx settings, the power setting is not being saved to the VTx configuration settings.

How to reproduce:

1)  Go into OSD, change any VTx settings.
2)  Doesn't matter if you Save, Save & Exit, or Save & Reboot.
3)  Reboot FC.
4)  Power is changed to 25mW on next boot.
5)  This can be verified via CLI and checking vtx_power - it will still be what it was before you changed it in OSD.
6)  vtx_power can be set in CLI to any valid setting.  This will always be the setting when the FC powers up or reboots.

User configuration:  OMNIBUSF4, IRC Tramp HV, Betaflight build 97c3e4b.
Reference https://github.com/betaflight/betaflight/pull/4265 for when this was pulled into Betaflight.

codecae, 2017-10-24:
Trying to understand what's happening between steps 4 & 5... when you say 

> it will still be what it was before you changed it in OSD

What was it "before" in this scenario?  If it was 200mW "before" and you changed the channel/band as described in step 1, does the CLI still show 200mW and the OSD shows 25mW?

codecae, 2017-10-24:
Have you tried setting the power in the CLI and see if the OSD is correct up on reboot?

HFMan, 2017-10-24:
@codecae - Basically the VTX power doesn't get saved in configuration- only the band and channel.  I have verified this by looking through the vtxsettings code (which is a mess IMHO).  No matter what you do in the OSD, the vtx_power SETTING never gets updated.  If you set vtx_power in the CLI, upon boot the power is restored to that setting correctly.

codecae, 2017-10-24:
Does the OSD show the value you specified in the CLI after you change it and reboot?  I'm also suspecting that there's potentially an init issue in CMS.

HFMan, 2017-10-25:
Yes, after reboot the OSD shows the value set in the CLI.  Like I said, it is not STORING the value, recalling the value and setting the power at boot is working fine.

raphaelcoeffic, 2017-10-25:
@null in the first version we had, the current power setting was pulled from the VTX itself, rather than from global VTX settings. Has it changed since then?

codecae, 2017-10-25:
@raphaelcoeffic #4265 created a bit of a chicken/egg situation with respect to vtx settings persistence.  The persistent settings on the VTX are now slave to a parameter group.

PR #4455 (2017-11-09): Move vtxSettingsConfig to vtx_common.  Rework MSP_SET_VTX_CONFIG and CMS.
PR Status:  In Development and Testing

Fixes: #4437 
Fixes: #4426
Fixes: #4454

Retired vtx_settings_config in favor of vtx_common.  Removed implicit EEPROM writes for MSP_SET_VTX_CONFIG updates, making it more like it's MSP_SET counterparts.  This requires subsequent MSP_EEPROM_WRITE request to perform EEPROM updates.  Lua scripts will need to send this request, like they currently do for all other MSP setters.

CMS performs EEPROM writes explicitly upon user election.  Nothing had to change here.

Fixed issue where CM [... cut]
closed 2017-11-09 by mikeller