betaflight/betaflight #4429: LED Strip broken on Omnibus F7

SS396, 2017-10-24:
Problem:
LED Strip is not functioning correctly.  (verified LED Strip working correctly on other targets using 3.2.1)

Testing method:
I set all LEDs in the strip (0-7) to color mode with the color of black (off), then selected one LED (position 0) to be white and saved the settings.  I observed the LED in position 0 turning white, however another LED in position 1 also turned on as green.

Flight controller info, identifier: BTFL, version: 3.2.1
Running firmware released on: Oct 15 2017 19:42:49
Board: OBF7, version: 0

mikeller, 2017-10-24:
Ok, 2 possible causes:
- voltage issues;
- timing issues.

Have you added a diode as described in https://github.com/cleanflight/cleanflight/blob/master/docs/LedStrip.md#connections (and no, the fact that this is not needed for the same LED strip on other flight controllers does not mean that it's not needed on the OBF7, the voltages are right on edge as it is, and minor differences in the on-board voltage regulators can make all the difference.

If that does not help, do you have access to a logic analyser that you could use to visualise the output signal on the LED_STRIP pin?

jflyper, 2017-10-24:
Omnibus F7s actually do have higher 5V than other Omnibus family variants. Try an inline diode.

SS396, 2017-10-25:
Installed a diode on the voltage in to the LED Strip as suggested by mikeller.

Voltage before the diode is 5.13, after the diode to the LED Strip it is 4.37 when its loaded.

Now the LED Strip doesn't light up at all with or without the diode and I thought I might of damaged something, but I tested the strip on an arduino and it does function correctly.

I have access to an oscilloscope so I can hook up to the LED data pin but I'm going to need a bit of instruction on timings and what a proper signal is supposed to look like as I've never done this sort of hardware diagnostics, but I'm willing to learn and experiment.

jflyper, 2017-10-25:
@SS396 I will run a test myself when I go back to my bench this evening.

In a mean time, can we see your CLI command "diff" and "resource list" outputs?

mikeller, 2017-10-25:
@SS396: There is a good writeup on the timing requirements for WS2811 LEDs: https://wp.josh.com/2014/05/13/ws2812-neopixels-are-not-so-finicky-once-you-get-to-know-them/

The symptoms that you described first (second LED white instead of dark) would suggest that bits after the first 24 (used for the first LED) were recognised as 1 instead of 0.

SS396, 2017-10-25:
@jflyper: here ya go, I look forward to your test results:

# diff

# version
# Betaflight / OMNIBUSF7 (OBF7) 3.2.1 Oct 15 2017 / 19:42:49 (434db9a36) MSP API: 1.36

# name
name Omnibus F7

# resources

# mixer

# servo

# servo mix

# feature
feature MOTOR_STOP
feature TELEMETRY
feature LED_STRIP
feature DYNAMIC_FILTER

# beeper

# map
map TAER1234

# serial
serial 5 32 115200 57600 0 115200

# led
led 0 4,0:N:C:1
led 1 5,0:N:C:0
led 2 6,0:N:C:0
led 3 7,0:N:C:0
led 4 8,0:N:C:0
led 5 9,0:N:C:0
led 6 10,0:N:C:0
led 7 11,0:N:C:0

# color

# mode_color
mode_color 6 0 2
mode_color 6 1 6

# aux
aux 0 0 0 1300 2100
aux 1 2 1 1300 1700
aux 2 13 2 1300 2100
aux 3 28 1 1700 2100

# adjrange

# rxrange

# vtx

# rxfail
rxfail 5 s 1500
rxfail 6 s 2000
rxfail 7 s 1000
rxfail 15 s 1000

# master
set mag_declination = 12000
set rssi_channel = 16
set use_unsynced_pwm = ON
set failsafe_throttle = 1250
set failsafe_procedure = AUTO-LAND
set bat_capacity = 2200
set gps_provider = UBLOX
set gps_auto_baud = ON
set pid_process_denom = 1
set osd_units = IMPERIAL
set osd_vbat_pos = 2497
set osd_rssi_pos = 2465
set osd_tim_1_pos = 2487
set osd_tim_2_pos = 2519
set osd_flymode_pos = 2082
set osd_throttle_pos = 225
set osd_vtx_channel_pos = 377
set osd_current_pos = 385
set osd_mah_drawn_pos = 353
set osd_craft_name_pos = 106
set osd_gps_speed_pos = 218
set osd_gps_lon_pos = 82
set osd_gps_lat_pos = 65
set osd_gps_sats_pos = 51
set osd_home_dir_pos = 302
set osd_home_dist_pos = 303
set osd_compass_bar_pos = 2089
set osd_pid_roll_pos = 423
set osd_pid_pitch_pos = 455
set osd_pid_yaw_pos = 487
set osd_debug_pos = 1
set osd_power_pos = 321
set osd_pidrate_profile_pos = 345
set osd_warnings_pos = 2473
set osd_avg_cell_voltage_pos = 76
set osd_pit_ang_pos = 257
set osd_rol_ang_pos = 289
set osd_battery_usage_pos = 2504
set osd_disarmed_pos = 2442
set osd_nheading_pos = 2104
set osd_esc_tmp_pos = 82
set osd_esc_rpm_pos = 83
set osd_stat_max_alt = ON
set osd_stat_endbatt = ON
set osd_stat_tim_1 = ON

# profile
profile 0

# rateprofile
rateprofile 0

==================================

# resource list
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: FREE
A01: BARO_CS
A02: FREE
A03: SERIAL_RX 2
A04: SPI_PREINIT
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: FREE
A09: FREE
A10: FREE
A11: USB
A12: USB
A13: FREE
A14: FREE
A15: MPU_CS
B00: MOTOR 1
B01: MOTOR 2
B02: FREE
B03: FREE
B04: FREE
B05: FREE
B06: FREE
B07: FREE
B08: FREE
B09: FREE
B10: FREE
B11: FREE
B12: OSD_CS
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: FREE
C01: FREE
C02: ADC_CURR
C03: ADC_BATT
C04: FREE
C05: FREE
C06: SERIAL_TX 6
C07: FREE
C08: FREE
C09: FREE
C10: SPI_SCK 3
C11: SPI_MISO 3
C12: SPI_MOSI 3
C13: FREE
C14: FREE
C15: FREE
D00: FREE
D01: FREE
D02: FREE
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: FREE
D09: FREE
D10: FREE
D11: FREE
D12: LED_STRIP
D13: FREE
D14: FREE
D15: BEEPER
E00: LED 1
E01: FREE
E02: SPI_SCK 4
E03: SDCARD_DETECT
E04: SDCARD_CS
E05: SPI_MISO 4
E06: SPI_MOSI 4
E07: FREE
E08: FREE
E09: MOTOR 3
E10: FREE
E11: MOTOR 4
E12: FREE
E13: FREE
E14: FREE
E15: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: LED_STRIP
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: FREE
DMA1 Stream 4: FREE
DMA1 Stream 5: FREE
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: FREE
DMA2 Stream 1: FREE
DMA2 Stream 2: FREE
DMA2 Stream 3: FREE
DMA2 Stream 4: ADC
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE

@mikeller: Well, what you have noted isn't what I was experiencing though.   I set position zero to white and it displayed correctly, but position 1 was never changed so it should of remained off, but it was displaying green (not white).  So based on that I conclud that the first 24 bits were sent correctly, but the next 8 bits were detected as 1's where they should of been 0's.

This is what I'm seeing on the LED data pin, I'm pretty sure it isn't supposed to have that increasing curve in the beginning.  

[image]

SS396, 2017-10-25:
This is the LED Strip data signal on a SP Racing F3 OSD.

[image]

So after comparing that and the referenced webpage, the signal isn't correct at all (theres no length at the peaks to indicate a 1).  It could very well be an issue with the board.  Not sure how to determine that in order to proceed further.  I look forward to any suggestions.

mikeller, 2017-10-25:
@SS396: Yeah, that signal from your OMNIBUSF7 does not look correct at all. Just out of interest, would you be able to use resource remapping to map the LED_STRIP output to a motor pin on the OMNIBUSF7 and scope the output there?

SS396, 2017-10-25:
@mikeller: K, that outputs a signal that looks identical to the SP Racing F3 OSD, I'll spare the capture image unless you really need it, so thats some welcome news.  However I noticed that LED_STRIP has moved to DMA1 Stream 7 after I remapped LED_STRIP to B00.  Is that DMA move normal and any way to move it back but still output on B00?

mikeller, 2017-10-25:
@SS396: Thanks for testing, that supports the theory that something with D12 is not in order. From looking at the signal, I would guess it's either the board design, or your board is faulty.

The DMA streams that can be used are restricted by the timer channels that are available for the individual pins, with each pin supporting only a few (if any) timer channels, and each timer channel only supporting a few DMA streams. The way target definitions in Betaflight are done is that for every pin that needs it, a timer channel and DMA stream (out of the available ones) are determined at compile time. To change the DMA assignment for your pins (if possible at all), you'd have to change the target definition and recompile.

SS396, 2017-10-25:
@mileller: How does one determine that D12 is the correct pin in the first place?  Any idea where D12 is at on the physical chip (didn't really have any luck searching for a STM32F745 pinout map), just wondering if I can successfully probe there and get a valid signal.  I assume it is just a straight through unmodified signal directly to the jumper.   I guess I really need @jflyper to post on whether or not he has the same results on his board.  That would at least tell me to quit wasting everyones time with a bad board.  :(

jflyper, 2017-10-25:
I'm still on the way back to my bench. 700km to go.

Don't rush, let me do a bench test.

Oh, and is this a V1 board? @SS396 
I recalled that V2 has a MOSFETs based signal booster between PD12 and LED pad that converts 3V3 to 5V. (Wondering if V1 has them, too ... I have to poke the board.)

mikeller, 2017-10-25:
@SS396: One looks it up in the target definition that the board's designer has provided: https://github.com/betaflight/betaflight/blob/master/src/main/target/OMNIBUSF7/target.c#L33

http://www.st.com/resource/en/datasheet/stm32f745ie.pdf is the datasheet you want to consult.

SS396, 2017-10-25:
@jflyper:  I'm not trying to rush anyone, so please forgive me if you read it that way, it was not my intention. I don't know how to identify the board version, I see no identifying markings or revision numbers on the silkscreen.  It was just purchased last week.  It arrived with 3.2 betaflight installed but the gyro icon wouldn't illuminate in the configurator and no matter what I did I couldn't get the quad x  setting to a standard non reversed propeller rotation, so thats when I noticed a new release and flashed it instead. 

@mikeller Thanks for that PDF, thats what I was unsuccessful at finding.  After browsing it I still can't help wondering how possible it would be to have a simple typo error in the target.h like what if it is supposed to be pin D13?   They seem almost identical and very close to each other.

Also why does the omnibus F7 target have LED_STRIP defined and something like an omnibus F4 doesn't even though they both have a dedicated LED pin jumper?

jflyper, 2017-10-25:
Does it have a clear plastic cover over the MCU and gyro chip?

mikeller, 2017-10-25:
@SS396: A typo like you describe would become immediately obvious during testing, since it would cause the LED_STRIP output to not have any signal at all.
OMNIBUSF4 **does** have an LED_STRIP definition: https://github.com/betaflight/betaflight/blob/master/src/main/target/OMNIBUSF4/target.c#L51

jflyper, 2017-10-25:
I confirmed LED strip has a problem on Omnibus F7 V2. I'll look into this.
Btw, remapping `MOTOR 4` to `LED_STRIP` works.

mikeller, 2017-10-25:
Is this caused by the output driver that you mentioned, @jflyper? If so, that is quite sad. Tried to do the right thing, and fell so short.

SS396, 2017-10-25:
@jflyper: no cover on gyro.

@mikeller hah, that assumes someone actually took the time and tested it during development.  Also I meant #DEFINE LED_STRIP in target.h.

@jflyper Thanks for the confirmation.  I'm curious do you get the same odd signal as I did?  And position +1 = green when set to white?  

At least now I know my board isn't the only faulty one.  :P

mikeller, 2017-10-25:
@SS396: Ah that one. A lot of the defines that enable features are done these days by target MCU type / flash size in `common_fc_pre.h`: https://github.com/betaflight/betaflight/blob/master/src/main/target/common_fc_pre.h#L98

The fact that the OMNIBUSF4 target still has a (now redundant) define of its own is just a leftover of the target being around before the common defines were introduced.

slawekgajzner, 2017-10-26:
Unfortunately, the level converter is probably not working well.
After removing the red part and after connecting the blue pin, the Led Strip works fine.

[image]

jflyper, 2017-10-26:
Yes. I can confirm that mod works.

SS396, 2017-10-26:
@slawekgajzner:  The instructions are not detailed enough for me to be willing to risk modification.  So remove both parts you have marked in red? (6 pin chip AND the resistor, I assume)  Then bridge the two pads (middle right and top right) you have marked in blue?  Is a solder bridge enough or does it need a tiny bodge wire.

Also is this mod only valid for banggood OmnibusF7 clone boards?  Are the parts in a similar location on the non clones?

I wish this was a software issue and not a hardware one, kind of really disappointing to have to take a soldering iron and perform surgery on a brand new board just for some silly blinking lights.

mikeller, 2017-10-26:
@SS396: Yeah, makes you wonder how these boards ever made it through testing...

slawekgajzner, 2017-10-26:
@SS396 : You have described exactly that.
I have two Omnibus F7 bangood board, I do not know if other non-clones are the same problem, probably not. 

I do not think anyone checked in the tests or the LEDs are working :)

jflyper, 2017-10-26:
I think I've tested the LED multiple times in the past, especially when I was working with N-Channel woes... Now I'm not sure how or if I did it 😢

mikeller, 2017-10-26:
Could it be that the level converter circuit is badly designed and quits it sooner or later when powered up?

jflyper, 2017-10-26:
I'm not sure... The circuit is somewhat over complex. It could be made with single mosfet, but uses two.

SS396, 2017-10-26:
To me (and I'm not a hardware engineer, I'm a software one) it looks like the signal is inverted.  If you look at the images I posted, one starts high and drops low, and the correct one starts low and goes high.  Shrug.  

Then theres that whole ramping up issue where its not returning to a steady starting point, not sure whats causing that.

jflyper, 2017-10-27:
No, it's not inverted.

I believe it's caused by slow switching characteristics of the mosfet used. The part under question is NTJD4001N dual N-channel mosfet, with following switching chracteristics.
(https://www.onsemi.jp/pub/Collateral/NTJD4001N-D.PDF)

[image]

Delays sum up to 216 ns, which is considerably slow against high time of 350ns for '0' signal.

jflyper, 2017-10-27:
I talked with myairbot, and they said they will fix the issue. They also said they are planning, for users of genuine boards, free replacement with the new board when it is available, or voucher toward future shopping at their store.

SS396, 2017-10-27:
@jflyper Again the capture clearly shows the signal starting and moving in the wrong direction.  That alone proves it sinks instead of rises.  Not sure how you can disagree with the scope.

So all Omnibus F7's are affected, even the non clones?

Tell myairbot they should send me a V3 board as a bounty payment for doing their development "testing" of the documented features.  Grin.

jflyper, 2017-10-27:
This is what I got.

[image]

And with more bandwidth
[image]

jflyper, 2017-10-27:
> So all Omnibus F7's are affected, even the non clones?

Yeap 😩 

Clones cloned bad circuit 😬

SS396, 2017-10-27:
> This is what I got.

@jflyper : On the board you've tested.  What I got on my board the signal sinks as documented.  Obviously the difference is we are not testing the same boards.

So on that MOSFET they are using pin 5 as an input from D12, and 6 is output, so pin 3 is connected to pin 2, and 1 and 4 get what 5 volts?  Whats the purpose of chaining the two together in series?

jflyper, 2017-10-27:
> So on that MOSFET they are using pin 5 as an input from D12, and 6 is output, so pin 3 is connected to pin 2, and 1 and 4 get what 5 volts?

I think the circuit is the same.

> Whats the purpose of chaining the two together in series?

Probably combined two current drivers. I've sent them a single mosfet solution and suggested faster ones.

SS396, 2017-10-27:
> I think the circuit is the same.

> Probably combined two current drivers. I've sent them a single mosfet solution and suggested faster ones.

Didn't really answer the questions I asked.  SMH.

ServusServorumDei, 2017-10-27:
So, @slawekgajzner, is this a known safe LED issue fix?  I just contacted
Airbot last night after I tested 1 of 2 OBF7 FC, LED is effed on mine as
well.  I was hoping this was isolated to a specific batch but as I read on
the thread, it seems like everyone is having the same problem.  At sixty
bucks a pop, it sucks that we have to be the ones to engage in fault
analysis.

Philip Monserrat
Data Engineer
Weatherford International
11943 Spencer Road FM 529
Houston, Texas 77041
Mobile: (281)714-0898
email: philip.monserrat@weatherford.com

ServusServorumDei, 2017-10-27:
> Mobile: (281)714-0898 <(281)%20714-0898>

@Kernel Panic:  You must chill.  We're all pi$$3d, grinding on @jflyper is
not going to help.....unless @jflyper developed the board, then yeah...JK.

Philip Monserrat
Data Engineer
Weatherford International
11943 Spencer Road FM 529
Houston, Texas 77041
Mobile: (281)714-0898
email: philip.monserrat@weatherford.com

slawekgajzner, 2017-10-27:
@ServusServorumDei In other flight controllers that I have, the processor pin is output directly to the LED output without any level converter, so I think it is safe.

ServusServorumDei, 2017-10-27:
@jflyper what type of MOSFET would be needed to replace this?

Philip Monserrat
Data Engineer
Weatherford International
11943 Spencer Road FM 529
Houston, Texas 77041
Mobile: (281)714-0898
email: philip.monserrat@weatherford.com

slawekgajzner, 2017-10-27:
This should be ok. PMGD780SN.

http://pl.farnell.com/nxp/pmgd780sn-115/mosfet-n-ch-trench-dl-60v-sot363/dp/1758102

ServusServorumDei, 2017-10-27:
Thanks, @slawekgajner, this would be very helpful. Just to make sure, you
removed the resistor and the MOSFET completely, is that right?

Philip Monserrat
Data Engineer
Weatherford International
11943 Spencer Road FM 529
Houston, Texas 77041
Mobile: (281)714-0898
email: philip.monserrat@weatherford.com

slawekgajzner, 2017-10-27:
yes

ServusServorumDei, 2017-10-27:
Kewl, I'll try that.

Philip Monserrat
Data Engineer
Weatherford International
11943 Spencer Road FM 529
Houston, Texas 77041
Mobile: (281)714-0898
email: philip.monserrat@weatherford.com

SS396, 2017-10-27:
@ServusServorumDei: I am very chill, thank you.

>@jflyper what type of MOSFET would be needed to replace this?

I don't believe replacing the MOSFET will magically fix the issue, along with the timing delay issue its also a board layout one.  From my own analysis I believe they were attempting to use it as a cascode amplifier.  And if that was actually the case, the point where they failed was when they connected pins 2 and 3 together, it should of been pins 3 and 1, and pin 2 to ground.  Thanks google.

lexusburn, 2017-11-06:
Airbot told me to use the left middle pin of the mosfet (next to the edge, not MCU) instead of the LED Output, because there's a bug in the board layout of v1 and v2 board.

When v3 board is released, they give discount for it to owners of v1/v2 after sending them the bill.

lexusburn, 2017-11-17:
Really simple solution if you don't need all Pinouts: remap LED strip.
Open betaflight CLI and copy the lines below two swap RX6 with LED_STRIP.
I use TX6 for SmartAudio, and RX6 was unused, because of these i've decided to use RX6.

# remap led strip: swap RX6 with LED_STRIP
resource SERIAL_RX 6 E01
resource LED_STRIP 1 C07
resource SERIAL_RX 6 D12
save

nk86, 2017-12-11:
Dear lexusburn,
I tried your solution by remapping RX6 and the Led strip. Unfortunately, in my case motor 4 stops working. If I remap the Led strip to TX6 motor 2 stops working. The Betaflight version is 3.2.2 and the board OB72 version 0. Is there a version difference or did I make a mistake?

casinick, 2018-04-01:
Can someone tell me if it's possible to remap the TX3 or RX3  to use it to camera control, because it does not work on the LED PIN...

Thanks.

closed 2018-04-02 by jflyper

ALopes90, 2018-05-22:
Tried in 2 Omnibus F7 v2 with Betaflight 3.3.0 and I could not fix it with any remaping.
And the hardware I can't try them because the layout is not the same as the pictures in this post.
Anyone can help? Thank you

nk86, 2018-05-22:
I also could not remap it. With remapping one motor stopped working or the whole board crashed and I had to unbrick it by the bootloader button. This happened with two different board revisions.

But I found out that the actual flickering problem on my F7v2s is not the broken circuit. It was because of the voltage of 5.3 V. The WS2812B strips start to flicker and show wrong colors when the voltage is higher than 5 V. So I put a small buck converter in between and set it to 4.8 V. That works pretty well.

mikeller, 2018-05-22:
@nk86: Even just a diode normally works for that purpose.

ALopes90, 2018-05-22:
@nk86 As I said I already tested 2 of my 3 Omnibus F7 v2, I after testing every software possibility never could make it light up in any color, no flicker no nothing. Tried to change between many versions of Betaflight, I'm going to try putting 5v from another source into the led strip.

nk86, 2018-05-22:
I'm not sure if the broken circuit was fixed but on my 2 F7v2s only the voltage was the problem. So try a diode or buck converter.

You don't need another power source. If you use one then make sure that the ground of the Omnibus and the ground of your separate power source is connected. Otherwise you may destroy it.

ALopes90, 2018-05-22:
What diode should I use? I don't know anything about diodes =S

mikeller, 2018-05-22:
@ALopes90: Pretty much any diode will do, so if you've got one lying around or can scavenge one from some broken electronics then just give it a go. All it needs to do is to lower the level of Vin by a small amount to make sure the level of Din is detected correctly.

ALopes90, 2018-05-22:
What do you mean by the level of Din?

ALopes90, 2018-05-22:
Testing now

mikeller, 2018-05-22:
@ALopes90:

> when I use it the voltage drops to 0.18v 

That suggests that you have mounted it in the wrong direction. It has to be connected so that its anode is towards the supply voltage.

There is a lot of variance for different WS2811 type LEDs in what levels they accept as logical 1. The supported input voltage range is large for most models, but since the spec defines the logical 1 level relative to the input voltage, that won't necessarily help you if your input voltage is higher than 4.7 V and the signal level is 3.3 V. There's also differences in what voltage the different GPIO pins can drive on an STM32, so changing pins might have improved the situation enough, considering that it's only marginally not working to start with.