betaflight/betaflight #4430: Betaflight 3.2.1 gps arrow issue

mattyfpv, 2017-10-24:
This video show the problem :https://www.youtube.com/watch?v=p-ULdKjkqgk

Arakon, 2017-10-24:
Did you update the font at some point?

mattyfpv, 2017-10-24:
No ,Its all stock

mikeller, 2017-10-24:
@mattyfpv: Try uploading the font again.

mattyfpv, 2017-10-24:
Ok, thanks

McGiverGim, 2017-10-24:
I had the same problem. My first idea was to update the font too, but I couldn't test it again because my GPS was suddenly dead. I'm waiting for a replacement. So very interested in your tests.

mattyfpv, 2017-10-25:
i updated the font... but the problem persists

McGiverGim, 2017-10-25:
The "correct" symbols go from 0x60 to 0x6F, and the incorrect symbols of your video go from 0x51 to 0x5F.

The function that calculates the symbol is:
https://github.com/betaflight/betaflight/blob/6626995047dff8293894b39eec3c2134e9809644/src/main/io/osd.c#L234-L245

@mikeller maybe the pass from the calculate value `int` to the result value `uint_8`  is transforming some values into negatives one? I don't have the GPS to test it and my C is not too high ;)

mikeller, 2017-10-25:
Good find, @McGiverGim!

From looking at it, this should be changed to be always positive:

https://github.com/betaflight/betaflight/blob/6626995047dff8293894b39eec3c2134e9809644/src/main/io/osd.c#L390

McGiverGim, 2017-10-25:
In the original code, the first thing done after the subs was to sum 360 to the value and make a % 360, in this way this value will always be positive:
```
  int16_t h = GPS_directionToHome - DECIDEGREES_TO_DEGREES(attitude.values.yaw);

   h = (h + 360) % 360;
   h = h * 2 / 45;
```
The code has been refactored since then, and now this is done in the `osdGetHeadingIntoDiscreteDirections()` and some calculations were changed in a posterior PR (to make a offset by half sector). I think that here is when the new operation can be negative:
https://github.com/betaflight/betaflight/blob/6626995047dff8293894b39eec3c2134e9809644/src/main/io/osd.c#L224-L232
Now multiplying by 2 and adding 360 maybe is not always positive. Without GPS I can't make tests, but I will do an excel with all possibilities to be sure.

McGiverGim, 2017-10-25:
@mattyfpv If I give you a new firm for your target coud you test it?

mattyfpv, 2017-10-25:
For sure, i m using kakuteF4

McGiverGim, 2017-10-25:
Here you have:
https://drive.google.com/open?id=0B9bVROK8o4e-SjJMMnFnYV93aUE
Is the version 3.2.1 with a patch added.

McGiverGim, 2017-10-26:
@mattyfpv Did you see the link? ;)

mikeller, 2017-10-26:
@McGiverGim: Can you open a pull request for this please?

mattyfpv, 2017-10-26:
Yes, i have already updated :), i Will try ASAP

McGiverGim, 2017-10-26:
@mikeller I was waiting for the test, but opened ;)

mattyfpv, 2017-10-26:
@McGiverGim 
Now works well , thanks 👍

mattyfpv, 2017-10-26:
Now the Arrow point in the wrong direction.

McGiverGim, 2017-10-26:
☹️ I will study the code again. Thanks for your test.

McGiverGim, 2017-10-27:
@mattyfpv yes, my earlier patch was wrong, sorry. Here you have another to test:

https://drive.google.com/open?id=0B9bVROK8o4e-cEVIekZfUzd6eGc

I'm pretty sure that this is the correct code. Please test it. Thanks!

mattyfpv, 2017-10-28:
Now Its point in the opposit direction

McGiverGim, 2017-10-28:
:( Always in the opposite direction? Or only sometimes?

McGiverGim, 2017-10-28:
Do you have the MAG configured correclty? Is the north pointing to the north in the compass that appears in your OSD?

McGiverGim, 2017-10-28:
@mattyfpv It does not seem to point it the opposite direction. It seems it points 90 degrees to the right of the "real" symbol. Have you verified your MAG? The North is in the North? I will take a look to the code again, but please verify your MAG.

McGiverGim, 2017-10-28:
I've done another hex, with the latest code of the PR, but in principle, this hex will have the same result as the latest one you tested. If you can test it please do it, but as I said I don't think that the result will be different (but maybe I make a mistake with my earlier code).

https://drive.google.com/open?id=0B9bVROK8o4e-d2E4SHRNU0s3eW8

But verify your MAG before testing this. If this does not work, I think we will need to wait until my GPS arrive to make the tests by my own changing pieces of code at the same time.

mattyfpv, 2017-11-01:
Now works perfect!!! Thanks man!

jaisor, 2017-11-08:
Any ETA on when this will be merged in a release build? I'm having the same issue on an Omnibus F4 board.

DanNixon, 2017-11-08:
@jaisor See #4445.

It is already merged and is available in nightly builds. It will be part of the 3.2.2 maintenance release which will be released soon.

jaisor, 2017-11-08:
Awesome, thanks!

richard-scott, 2017-11-09:
@jaisor try a build from here: https://ci.betaflight.tech/job/Betaflight%20Maintenance%203.2/lastSuccessfulBuild/artifact/obj/

Abancalari, 2017-11-13:
@McGiverGim Suggesting here. Would it be possible to supplement the heading typically provided by MAG with a  vector calculated from 2 GPS coordinates? And then have a CLI command to provide how many frames to look back when generating that vector? I made an excel sheet plotting some points and vectors and calculating a 0-360 degree value for the RTH arrow.

McGiverGim, 2017-11-13:
Please make a feature request in the issues page, to see if someone more experimented than me can gave some idea of how to do it. The heading is affected by accelerometer so I don't know how it will work.
If it is easy, I can try to implement it but I'm newbie in C code.

Abancalari, 2017-11-13:
@McGiverGim It seems the IO from the GPS antenna already pulls the approximate heading in degrees. Maybe it's just a simple swap of the MAG or ACC supplied heading with the one from GPS?

mikeller, 2017-11-13:
@Abancalari: Using GPS to get a heading information will only work reliably if it is used on a fixed wing aircraft, since it is based on the assumption that the vehicle only ever moves in the same direction relative to its frame. For a multirotor craft that can go into any direction, this will provide bogus information a lot of the time. We could probably add this, but only enable it if a fixed wing mixer is selected.

Abancalari, 2017-11-13:
@mikeller I understand that for shorter flights, a multirotor's flight path is too sporadic for this to make sense seeing as it can move in any given direction. But for longer flights where you're moving in a pretty straight path over longer distances (similar to a wing), wouldn't it work the same way? Assume the from of the quad is always facing the direction it's moving. Even just a CLI toggle to select where the heading is pulled from would be fine.

mikeller, 2017-11-13:
@Abancalari: Yeah, having it as a CLI option and disabled by default would work as well - I just would not want to have to field support requests from users not understanding the way this works for 'please fix GPS heading for flying acro'.

Abancalari, 2017-11-13:
Agreed. But as it is in 3.2.2, you're going to run into that issue anyways. Maybe "home arrow" in OSD should be disabled unless MAG is present or GPS heading is selected. Or maybe just add a note next to it in the OSD config page.

arrowcircle, 2017-11-15:
@mikeller Could You please make a build of 3.2.2 for colibri race with gps code?

malarcky, 2017-11-18:
In 3.2.2, the problem with the strange character is solved, but my fc is rotated 270 from yaw, and my arrow shows an offset of 270 degrees too, in 3.2.1 I did not have this problem.

McGiverGim, 2017-11-18:
In betaflight configurator the compass shows the North correctly?

malarcky, 2017-11-18:
Yes, as soon as I connect, it start with north position on the front.
[image]

McGiverGim, 2017-11-18:
And your quad is facing to the North?

malarcky, 2017-11-18:
I have no compass in this quad, only gps, but in 3.2.1 this feature works okay without compass. It show pointing north as soon as I connect the quad no matter if its facing north or not.

McGiverGim, 2017-11-18:
As far as I know, this can not work right without compass. Are you sure that in 3.2.1 this worked? What fc are you using?

malarcky, 2017-11-18:
Yes, it worked perfectly without compass, only with gps and accelerometer, Asgard v2. I have dvr of the problem if its neccesary.

mikeller, 2017-11-18:
Yes. The workaround is to make sure you are pointing the quad straight north at the time you are plugging in the battery. It 'works' without a compass, but obviously in this case the flight controller cannot get an absolute reading for its orientation, only one relative to its initial position.

malarcky, 2017-11-18:
But why in 3.2.1 the arrow point to my start position but not in 3.2.2?

mikeller, 2017-11-18:
Could be coincidence, since it is dependent on what direction your craft pointed when you plugged it in. In order to verify if this works correctly or not, you'd have to use a compass, and not rely on an arbitrary direction being north.

malarcky, 2017-11-18:
So if I plug the quad pointing fordward (and as I dont have compass, that is the position of its north) it will be "work" like the 3.2.1 in the video? I will try tomorrow.

mikeller, 2017-11-18:
You'd have to point it straight north when plugging in, since in the absence of a compass the firmware initialises the inertial model for direction to be 'north'. The home arrow is calculated from taking the vector from the GPS coordinates where the craft was booted to the current GPS coordinates, and then subtracting the current direction that the craft is pointing. Since the GPS vector's direction is an absolute direction, this will only work if the inertial model has an absolute direction as well.

beeb, 2017-12-26:
Sorry to open this again, but regarding this "absolute direction", couldn't BF also get the current direction from the last x GPS points? It would in fact be the velocity vector that would give you an absolute idea of where the craft is headed. This would remove the need for manual calibration (i.e. pointing north while plugging in) or compass.

beeb, 2018-01-01:
PR #4817 has changes that go in the direction mentioned above, nice!

McGiverGim, 2018-07-17:
It seems this has been fixed. Closing it.

closed 2018-07-17 by McGiverGim