betaflight/betaflight #4431: Notch filter will not save

LeinesFPV, 2017-10-24:
Hello,
if I set my Notch filter to 0 and then save, the values after a reboot are again on default.

If I set this in the CLI to 0 and then save, these are in the CLI to 0 and the filter tab is still default. Same game with my LUA script ...

I use the newst BF 3.2 on an Hobbywing FC + 4in1 combination.

mikeller, 2017-10-24:
@LeinesFPV: What version of the configurator are you using? There was an older version that had this bug.

LeinesFPV, 2017-10-24:
Hey,
I found the mistake. With version 3.2.1 of the configurator it goes. The error is probably due to the new version 3.2.2!

LeinesFPV, 2017-10-25:
Here is the attempt of a friend,

• from floor arriving at the CLI, the notch1_hz set to 0 -> Configurator continues to display stock value, but CLI is 0

• as soon as you set a higher value as Stock (400) by CLI, this is then also displayed on the tab. Put it under 400 again, regardless of whether it is set to 0 or 350, the Configurator then always shows the stock value of 400.

• If you set _cutoff to 0 instead of _hz, it works and the GUI shows both values to 0 and the switch is disabled. If you set _cutoff to greater than 0, then in the GUI, _hz is displayed again

• If you use the GUI to switch off, both values are set to 0, which can then be verified by CLI

wind0r, 2017-10-25:
Remember if you want to disable a notch filter you need to set both values to 0. 
This has nothing todo with the firmware, thats just the configurator being "strange".
For the new notch disable toggles to work both values need to be 0 .. if just one is 0 default values will be shown.

etracer65, 2017-10-25:
> Remember if you want to disable a notch filter you need to set both values to 0.
This has nothing todo with the firmware, thats just the configurator being "strange".
For the new notch disable toggles to work both values need to be 0 .. if just one is 0 default values will be shown.

This is not actually true - it is only necessary to set the notch center frequency to 0 to disable the notch filter.  It has been this way since the implementation of notch filters.

I'm not sure what the actual issue is.  Turning off the switch sets the center and cutoff to zero which disables the filter (because of the zero center frequency).  Turning the switch back on has to populate with some non-zero values to actually enable the notch.  In the firmware there is no place to store the "previously entered" values so the only reasonable option is to reset the values to the defaults.

Woody9999, 2017-10-26:
I'm not sure if this is related or not, but something I ran across in the last couple of days.

https://www.rcgroups.com/forums/showpost.php?p=38518617&postcount=52539

Tbeal79, 2017-11-12:
so has this been solved? bc i just seen there was an update and seen that there was this 3.2.2 firmwear. i also saved all my stuff before switching over. and i have got the same thing as you guys did with the filters.
]my question is if i turn the notch slider to off and it turns of the 2 filters for each setting is it the same as just setting the one like 400 to 0 and saving?
im trying not to burn up my motors here lol.
i much rather have had it the old way and set them all to the way i want to them, bc as i see it now i either have everything turned on, or have all three of them sliders turned off.
so im not really seeing why this would have ever been changed from the old way at all?
im running the omnibusf4sd pro version
and im getting that low battery warning now also, so i guess i have to also go in and change the values like i had to for my bf3 fc.
im not going to flash my other 2 fc's to the 3.2.2 till i have heard back from one of you guys about filter settings, bc the 3.2.1 version i can change them my self and it will save what i want it to be set at.
thanks to how ever can get back to me on this

mikeller, 2017-11-12:
@Tbeal79: See this comment: https://github.com/betaflight/betaflight/issues/4431#issuecomment-339273212

closed 2017-11-12 by mikeller

Tbeal79, 2017-11-12:
so mikeller what im gathering is that you just turn the toggle off in order for it to really work right? 
bc then i have been doing it wrong the hole time and setting only the 400-0 and the other 2 as well to 0, when in fact i should have been setting both of them to 0 on each one of them? 
is this what you are telling me? 
bc thats what i gather from reading wind0r post.
thanks again