betaflight/betaflight #4432: ESCs wont start up all the way with matek F405 fc and matek pdb

LoganFPV, 2017-10-24:
I was out flying the other day and I got snaged by a ghost branch. I disarmed my quad and rearmed it to try to get it out of the tree. The quad fell to the ground which is good of course but the grass was still wet from the rain earlier in the day. i unplugged the battery so i could dry it off and when i pluged it back in it did the tones but then when i armed it and gave it throttle the quad freaked out and started flipping everywhere. I disarmed it and unplugged the battery once again and the ESCs only did 2 beeps. 

The OSD on the FC wouldn't tell me the voltage of the battery. I plugged it into my computer and re-flashed the board to see if that would work. but it didn't. So then i tried to flash the ESCs but it still wouldn't arm. even in betaflight the voltage wouldnt read. 

I tried spooling up the motors in the motor tab but it wouldn't work

please help

Arakon, 2017-10-24:
What exactly does this have to do with the BF software? Please post things like this on i.e. rcgroups. This is not the place.

mikeller, 2017-10-24:
@LoganFPV: GitHub is for questions and discussion around the **development** of Betaflight. For questions around the configuration and use of Betaflight, please visit [RCGroups](https://www.rcgroups.com/forums/search.php?searchid=62637789&query=betaflight%20fixing%20a%20quad).

closed 2017-10-24 by mikeller