betaflight/betaflight #4437: Setting a combination of VTX Power and Band/Channel via MSP triggers reboot

codecae, 2017-10-25:
When using MSP to set the VTX band and/or channel while also adjusting the output power, a reboot occurs.  Setting these values in separate MSP requests does not cause a reboot.  I have triggered this on SmartAudio and si618 has triggered this using a Tramp.

Video of issue: https://youtu.be/xDU1wXmaMns

codecae, 2017-10-25:
One thing that is standing out is that ```saveAndNotify()``` is called twice in this scenario.

HFMan, 2017-10-25:
I too have seen this occur, both when using the OSD and when using the new Spektrum api.   It doesn't appear to be a full reboot, as the Betaflight splash does not occur on the OSD.

si618, 2017-10-25:
Tested on revised lua scripts and confirm saving two or more changes to vtx trigger beeps. Doesn't appear to be any particular setting or value.

ßF: DYSF4PRO (DYS4) 3.3.0 Oct 23 2017 / 12:20:58 (e6e0898) MSP API: 1.37 (master)
VTX: Tramp HV V2
[lua scripts](https://github.com/codecae/betaflight-tx-lua-revised/commit/57453f97647949b461b5815595902081a4ae54ea).

codecae, 2017-10-25:
beginning to wonder if the following is manifesting itself in unexpected ways
https://github.com/betaflight/betaflight/blob/master/src/main/fc/config.c#L651

ethomas997, 2017-10-25:
What's happening is that the VTX configuration values are being written to the new "vtx_" settings, and a settings "save" is invoked (that's where the beep happens).  The "save" is necessary so the VTX configuration is persistent.  (The "vtx_" settings are written out to the VTX at startup).  See PR #4265 for more info on the VTX settings.

codecae, 2017-10-25:
Testing has shown that it only seems to beep when a second saveConfigAndNotify is triggered

ethomas997, 2017-10-26:
@codecae:  Sorry, at first I thought you were referring to the buzzer beep that happens in response to a settings save.  Can you tell for sure that a flight-controller reboot is happening?  (Do you get a repeat of the startup LED flashes and buzzer beeps?)  If you do a 'Save' (without reboot) in the configurator, do similar motor beeps happen?

codecae, 2017-10-26:
@ethomas997 It is not a reboot.  ESCs are beeping the motors with the protocol pattern (second set of beeps).  Wondering if it's some sort of condition caused by rapid eeprom writes and reads.

https://www.youtube.com/watch?v=xDU1wXmaMns

ethomas997, 2017-10-26:
Are the motor beeps happening via this code in "io/beeper.c" (line 382)?:

```
#ifdef USE_DSHOT
        if (!areMotorsRunning() && beeperConfig()->dshotBeaconTone && (beeperConfig()->dshotBeaconTone <= DSHOT_CMD_BEACON5) && (currentBeeperEntry->mode == BEEPER_RX_SET || currentBeeperEntry->mode == BEEPER_RX_LOST)) {
            pwmDisableMotors();
            delay(1);

            pwmWriteDshotCommand(ALL_MOTORS, getMotorCount(), beeperConfig()->dshotBeaconTone);

            pwmEnableMotors();
        }
#endif
```

If so, the next step would be to trace why it's being triggered...

PR #4455 (2017-11-09): Move vtxSettingsConfig to vtx_common.  Rework MSP_SET_VTX_CONFIG and CMS.
PR Status:  In Development and Testing

Fixes: #4437 
Fixes: #4426
Fixes: #4454

Retired vtx_settings_config in favor of vtx_common.  Removed implicit EEPROM writes for MSP_SET_VTX_CONFIG updates, making it more like it's MSP_SET counterparts.  This requires subsequent MSP_EEPROM_WRITE request to perform EEPROM updates.  Lua scripts will need to send this request, like they currently do for all other MSP setters.

CMS performs EEPROM writes explicitly upon user election.  Nothing had to change here.

Fixed issue where CM [... cut]
closed 2017-11-09 by mikeller