betaflight/betaflight #4440: Can't use ESC passthrough with BF 3.2.X

GilDev, 2017-10-25:
Hi,
I flashed my Afro 12A ESCs with BLHeli 14.9 the other day.
On a FLIP32+ (basically a Naze32), passthrough configuration works on BF ≤3.1.7 and doesn't work on BF ≥3.2.0. ESC configuration softwares says the FC firmware is not compatible and that I should upgrade it.

AndersHoglund, 2017-10-25:
As the ESC Config sw can talk to the ESC and see that they are not compatible, the passthrough is working fine. Not much more betaflight firmware can do about that.

closed 2017-10-25 by AndersHoglund

GilDev, 2017-10-25:
I don't understand. The ESC Config SW can talk to the ESC **only** on BF ≤3.1.7. When only changing the FC firmware, upgrading it to BF 3.2.X, it stops working.
The ESCs are compatible in all cases, the config SW says the FC firmware doesn't support passthrough.

DieHertz, 2017-10-25:
Passthrough has been disabled F1 targets in 3.2

GilDev, 2017-10-25:
Oh alright, thank you for the heads up.
I searched online but couldn't find any information. It seems to often be the case with Betaflight, the only solution seems to be to read every changelog…

wind0r, 2017-10-25:
There is a reason why Betaflight 3.2 is the last release for F1 Boards.
Nearly every feature is disabled to make the code fit.

Just searching for Naze shows the same "issue"
https://github.com/betaflight/betaflight/issues/4184
https://github.com/betaflight/betaflight/issues/3701

GilDev, 2017-10-25:
Sure, thanks for the clarifications and sorry if I didn't search deep enough.