betaflight/betaflight #4443: No RSSI Channel 16

Chachipower, 2017-10-26:
I am trying to set up RSSI on my Frsky XM+ with BF 3.2.2 but under "RSSI Channel" dropdown menu, I am only able to see 14 Channels. I need to select Channel 16.

# version
# Betaflight / KAKUTEF4 (KTV1) 3.2.1 Oct 15 2017 / 19:36:58 (434db9a36) MSP API: 1.36

# name
name Furibee_X215

# resources

# mixer

# servo

# servo mix

# feature
feature -TELEMETRY
feature LED_STRIP

# beeper

# map

# serial

# led
led 0 9,15::A:13
led 1 8,15::A:13
led 2 7,15::A:13
led 3 6,15::A:13

# color

# mode_color
mode_color 6 0 13
mode_color 6 1 5

# aux
aux 0 0 0 1925 2100
aux 1 1 1 900 1100
aux 2 2 1 1300 1700
aux 3 4 11 1300 1700
aux 4 13 2 1875 2100
aux 5 28 0 1875 2100

# adjrange

# rxrange

# vtx

# rxfail

# master
set acc_trim_pitch = -2
set acc_trim_roll = -2
set baro_hardware = NONE
set rssi_channel = 12
set dshot_idle_value = 250
set motor_pwm_protocol = DSHOT600
set vbat_min_cell_voltage = 32
set current_meter = ADC
set vbat_scale = 111
set ibata_scale = 275
set pid_process_denom = 1

# profile
profile 0

# rateprofile
rateprofile 0

mikeller, 2017-10-26:
@Chachipower: GitHub is for questions and discussion around the **development** of the firmware and tools. For questions around the configuration and use of Betaflight, please visit [RCGroups](https://www.rcgroups.com/forums/search.php?searchid=62637789&query=betaflight%20cleanflight%20configuration%20use).

[image]

closed 2017-10-26 by mikeller

jflyper, 2017-10-26:
@Chachipower CH16 is AUX 12.

Chachipower, 2017-10-26:
That I understand, I just dont see channel 16 on the drop down menu for RSSI channel. My understanding is that it is Aux 12 but I am supposed to select channel 16 on drop down menu. However, the drop down stops at 14.  Thanks

wind0r, 2017-10-26:
You understand that it is AUX12 is CH16.
Why you still want to select the AUX16 then?

mikeller, 2017-10-26:
@Chachipower: I think your configurator is outdated, and is showing channel numbers without the 'AUX' in front of them. If you look at the screenshot that I posted you can see that the numbers in the drop down are actually AUX numbers, and this has been made more obvious in newer configurator versions.

Chachipower, 2017-10-26:
I believe I have outdated configurator as someone stated. In other words. In the RSSI channel drop down, it shows channels. On a video I have been using to help configure my RSSIn it shows them selecting channel 16. Mine only goes to 14. On the receiver inputs I know that aux 12 would be 16. I was just told the drop down should show aux not channels. I will try to update and see if that works. Thanks