betaflight/betaflight #4449: Anti burn motor mode-Blheli 32

Bugg69s, 2017-10-27:
I have idea to make people not burn motors because prop is stuck. It will stop pushing current to faze after one second. It knows that motor is not spinning. It will be good at new turtle mode in DShot commands.
Most motors burn out because ESC is just pumping power to the ESC and it burns isolation.

It will be cool to have something like that.
I think it could be possible with not bl heli 32 ESC's too

DieHertz, 2017-10-27:
ESC is pumping power to the ESC? You may want to double-check your information :-)

wind0r, 2017-10-27:
With that description I do not understand what betaflight should do. Could you please explain further?
This sounds like something esc firmware needs to implement and not betaflight.

Telemetry could be used to detect not spinning motors, but thats only a kiss and blheli32 feature.

Bugg69s, 2017-10-27:
@wind0r in BLheli 32 FC can get information of rpm. 
It will stop trying to spin motor after one (or less) second.

It will look at stick position and RPM.
If will be still 0 RPM after one second (or less) it will stop trying to spin the motor.

I think it will be possible to get it in not blheli 32 ESC's. It will need to be in ESC and 
need to have something to do with timing

Bugg69s, 2017-10-27:
@DieHertz idk if it is pumping or not, but I know that if burns the copper windings.
That means that it pushed too much current for long time

DieHertz, 2017-10-27:
Where do you think FC gets this information about eRPM from? ESC uses this info and tries to cut power upon stalling, but if that happens at full throttle energy has to go somewhere, and it goes into burning your equipment.

mikeller, 2017-10-28:
I think using the flight controller and ESC telemetry data to effect a motor shutdown as a failsafe would be pretty poor design. If anything, this should be done in the ESC firmware: bitdump/BLHeli#266

What should be looked at if this is implemented in the ESC firmware is to make the flight controller use the ESC telemetry information to detect if eRPM of a motor has dropped to 0 and act accordingly (potentially stop the other motors too, or at least keep them from spinning up in a futile attempt to compensate).

Bugg69s, 2017-10-28:
@DieHertz in BL heli 32 ESC is should be able to get pretty good rpm readings

wind0r, 2017-10-28:
@Bugg69s how you know they are pretty good?
Remember exactly those RPM readings are used by the ESC to detect a stalled motor.

Also I still wonder what Betaflights Job is in this case. As you said your Idea is possible without telemetry (e.g blheli) What should betafligh do?

I still think getting ESC Telemetry into the loop is usefull e.g to emergency failsafe / crash recovery the quad if a motor failes but i think that isnt the Idea here.

fiveangle, 2017-10-28:
> I still think getting ESC Telemetry into the loop...

It _**might**_ allow faster response to detecting when motor isn't responding to RPM change commands, but would probably still be limited by the required hysteresis, which might be close to a zero-sum game.  In the end, the gyro is the boss.

Jahnkeanater, 2017-11-02:
Blheli_32 already has current protection, low RPM power protection, and temperature protection. There is no reason for betaflight to do something the ESC's already do.

By default, current protection is turned off you need to enable it through BLHeliSuite.

fiveangle, 2017-11-02:
> Blheli_32 already has current protection, low RPM power protection, and temperature protection. There is no reason for betaflight to do something the ESC's already do.

ORLY ? :)  Obviously there **_are_** reasons: like when the ESC reports that the RPM commanded could not be attained, the FC can either ignore it and begin a wonderful Flip Of Death like it does now, or choose to do soemthing else...

Does something like that belong in BF ?  Probably not as it isn't really a performance thing and BF focuses on the gyro for that data as much as it can now.  But could something like that be useful for more autonomous platforms ?  Quite possibly.

But yeah, love the conviction ! 👅

DieHertz, 2017-11-02:
Dave Johnson ni need to be that sarcastic and remove messages afterwards. @Bugg69s talked about anti-burn protection, not what you're talking about.
When ESC telemetry will become more available and some dev has interest in working on it, eRPM and other data will let Incorporated into BetaFlight in-flight code for feedback.

mikeller, 2017-11-02:
> When ESC telemetry will become more available and some dev has interest in working on it, eRPM and other data will let Incorporated into BetaFlight in-flight code for feedback.

Too late, that's already happened: :wink: 

[image]

But agreed, ESC telemetry is only a niche for now, most ESCs do not yet (fully) support it. (Who wants to solder wires to tiny pads in the middle of their ESC?)

fiveangle, 2017-11-02:
> @Bugg69s talked about anti-burn protection, not what you're talking about.

It was precisely this realization that made me want to rescind.  As I was incorrect.  I noticed it immediatley afterward so felt better to not leave it up there.  I do appoligize @Jahnkeanater.  I was conflating two issues unnecessarily.

-=dave