betaflight/betaflight #4450: SP Rating F3

Mmarsian, 2017-10-27:
# If your issue looks like a hardware fault or a configuration problem please don't raise an issue here.

## Please consider using other user support options such as asking the manufacturer of the hardware you are using, RCGroups: https://rcgroups.com/forums/showthread.php?t=2464844, or Slack: https://slack.betaflight.tech or other user support forums & groups (e.g. facebook).

## If you believe there is an issue with the firmware itself please follow these steps:
1.) Describe your problem;
2.) Include ways to reproduce the problem;
3.) Provide information about your flightcontroller and other components including how they are connected/wired;
4.) Add the used configuration and firmware version;
5.) Create a `diff` and post it here in a code block. Put ``` (three backticks) at the start and end of the `diff` block (instructions  on how to do a diff: https://oscarliang.com/use-diff-not-dump-betaflight/); and
6.) Remove this Text :).

Arakon, 2017-10-27:
Uh.. yeah?
SPRF3 doesn't use BFOSD, so there's nothing to configure that way. This is not a BF issue.

Mmarsian, 2017-10-27:
So sorry. I'm writing on the phone. I  now corrected the first post.
Station, please look in the Betaflight configurator or under Betaflight downloads. Sp racings f3 evo, f4 evolution, and osd/pdb are now supported thanks to hydra. So if it's part of Betaflight how can it not be an Betaflight issue?

mikeller, 2017-10-28:
@Mmarsian: There is no benefit to using Betaflight on the SPRACINGF3OSD over using cleanflight, and Betaflight on the flight controller will work with cleanflight on the OSD, so just install cleanflight on the SPRACINGF3OSD, and use the cleanflight configurator.

Mmarsian, 2017-10-28:
@mikeller Thank you. I did as you recommended and it works 👍  ...... almost.
Since Betaflight Configurator has no settings like in Cleanflight Configurator to set the current meter to "MSP - OSD/Slave" I lost the current meter functions. Everything related is 0. Are there CLI settings to use the current meter on the SPRF3 osd/pdb?

I still wonder what the file "betaflight_3.2.1_SPRACINGF3OSD.hex" is used for.  Maybe @hydra can enter this discussion?

mikeller, 2017-10-28:
:-(

Sorry for that. Looks like you are stuck with using cleanflight for the moment. I thought @hydra had pulled these changes over into Betaflight.

The SPRACINGF3OSD does not have much use really, we should probably stop building it since there does not seem to be a useful way to configure it.

mikeller, 2017-10-29:
@Mmarsian: Good news: I've got a fix for the configurator that allows you to select MSP/OSD as a current meter source. But you'll have to wait for the next release of the configurator (or download it from https://github.com/betaflight/betaflight-configurator/pull/702 and install as source).
Even better, I've also added MSP/OSD as a CLI option in #4457. :-)

PR #4457 (2017-10-29): Added missing 'MSP' entry to current meter source list for CLI.
Fixes #4450.
closed 2017-10-29 by mikeller