betaflight/betaflight #4452: Simulation

aloniko84, 2017-10-27:
Hello,
As I trying to use simulation with SITL, i understand that SITL its only software simulation.
Is there any way we can make a simulation based hardware, that means, an app like gazebo will receive data from flight controller that received RC's sticks from a real RC and the simulated drone will fly in simulation screen. 
I think its can be done because betaflgiht configurator already receiving data about RC.
Maybe there is already some simulation that I can use?
thank you

mikeller, 2017-10-28:
@aloniko84: What is the purpose of the flight controller in the setup you propose? Just connect your TX to the PC and use the RC commands in your simulation.

jflyper, 2017-10-28:
I think he is thinking of HITL.

cs8425, 2017-10-28:
HITL +1
If sensor data send via serial, I think the bottleneck may be the serial baud rate.
Real loop time will be limited by sensor data input speed.

ledvinap, 2017-10-28:
Isochronous USB transfer would be very interesting option. It would allow reasonable speed and may help with PC side latency. It would also eventually allow attaching FC to some ARM SBC (Pi zero etc) for high-level functions ...

aloniko84, 2017-10-29:
So how I should do it?
HITL will work i think

mikeller, 2017-10-29:
I guess at it's simplest, you can just use `MSP_MOTOR` (https://github.com/betaflight/betaflight-configurator/blob/c2850e432386cacccb62ee8cd371c0497a48b7fc/js/msp/MSPHelper.js#L130-L134) to read the craft's motor output and use them in your simulated model. (This will obviously leave out any feedback from the gyro, but feeding simulated acceleration back into the flight controller would first have to be implemented.

mikeller, 2017-10-29:
Put on your Betaflight goggles, man. :sunglasses:

cs8425, 2017-10-29:
I had wrote a simple node.js program to grab values from joystick ('USB Simulator Cable' act as joystick).
not so clean but still can work.
https://gist.github.com/cs8425/51893e2f90812aa3831558503597fa1a

mikeller, 2017-10-29:
That looks like it should work to get the data to the flight controller. I would probably rather use serial for this use case instead of TCP, it is tested a lot more.

aloniko84, 2017-10-30:
@cs8425 

> I had wrote a simple node.js program to grab values from joystick ('USB Simulator Cable' act as joystick).
> not so clean but still can work.
> https://gist.github.com/cs8425/51893e2f90812aa3831558503597fa1a

1.What is the hardware configuration ? How to connect all?
2.What should i do? (I'm new fot Linux)
Thank you.

cs8425, 2017-11-01:
@aloniko84 
'USB Simulator Cable' act as joystick, will have a file at `/dev/input/jsX` (X = 0, 1, 2 ....)
line 144 @ emu-dx6-msp.js: `var js = new Joystick(0);`, change `0` to match your joystick number.
line 127-142 @ emu-dx6-msp.js: Change `joy2RC`, `joyBtn2RC` to match your transmitter channel.
line 5-6 @ emu-dx6-msp.js: change to match your SITL IP and port.
and then just run it.

@mikeller 
Yes, just replace `net.connect()` with some serialport package still can work.
But we need to add some custom MSP command to set sensor data for HITL.

@ledvinap 
Isochronous USB transfer would be great but we may need to write our driver...
That's a big issue.

tigert, 2017-11-17:
This is a bit related and might or might not be useful, but this issue made me remember seeing this:
http://ardupilot.org/dev/docs/flightgear-hardware-in-the-loop-simulation.html

s-ol, 2020-01-11:
I would be interested in picking this back up. A standard 8KHz Pidloop should be possible at the default 152KHz baud if my napkin calculations aren't wrong.