betaflight/betaflight #4458: Horizon and Angle Reversed

parildsen, 2017-10-29:
Lizard 95 running  Betaflight / OMNIBUS (OMNI) 3.2.1 Oct 15 2017 / 19:42:04 (434db9a36) MSP API: 1.36
BF 3.3.2

My angle and Horizon are reversed.  in BF i have my standard set up, aux 4 switch set, down (2000) is Horizon) and up (1000) is Angle.  Airmode set at 1000 on aux 4.   when i fly its opposite.  No sure if I am missing any other setting but this is my standard config.

mikeller, 2017-10-29:
Have you verified with your TX in the 'Modes' tab in configurator that the configuration is still the way it should be? How's your `diff` look like?

parildsen, 2017-10-29:
wow, thanks for the quick reply Michael
 
below is a screenshot.  When in Horizon mode the quad will flip/roll 360, in
Angle it is restricted
 
also I noticed highlighted in blue Board Omni , version )     is that
normal?
 
 

  _____  

<https://github.com/betaflight/betaflight/issues/4458#issuecomment-340256180
SmvcxB_ks5sxGYHgaJpZM4QKSCN> the thread.
<https://github.com/notifications/beacon/AfqLZazgyUpPKYC5_emWBTCLP6ZMwM7uks5
sxGYHgaJpZM4QKSCN.gif>

jflyper, 2017-10-30:
> When in Horizon mode the quad will flip/roll 360, in Angle it is restricted

That is absolutely normal.

closed 2017-10-30 by jflyper