betaflight/betaflight #4460: Omnibus F3/F4.. Mag on by default?

Arakon, 2017-10-29:
Current build (last commit fb3bc456e5fa1c3c92552359488d61321fe3b6d0), both on Omnibus F3 and F4.
When pasting back a CLI diff from a previous version, Mag now defaults to on on these targets. It used to be off by default (especially since the Omnibus boards don't have an onboard mag usually).
Intentional or bug?

AndersHoglund, 2017-10-29:
This is one of the reasons you should not copy & paste settings across versions.
What is the MAG settings without pasting anything, only doing "Full chip Erase", or "Reset Settings" or the "defaults" command ? That is when you have the default setting. After pasting a diff, you are no longer in a default state.

mikeller, 2017-10-29:
@Arakon: Yes, the default for this was indeed changed in:
https://github.com/betaflight/betaflight/commit/60b8e0f05ea295056b823d7d23bcb363c9f4f8a3#diff-cb07ec757bf8d1c9a158ffe6e94ecfc7R67

Not sure what the motivation behind this is, @MJ666 can you please give us some insight?

Arakon, 2017-10-29:
@AndersHoglund: Well, yeah.. but I'm not about to re-set up everything from scratch just because I flashed a revision with 2 merges in between. I know my stuff enough that I verify whether something errored out or got weird. The default value is on, fresh defaults or no.

AndersHoglund, 2017-10-29:
I did not make my self clear. Was there an incorrect default setting of MAG? Or was the incorrect setting of MAG as a result of pasting a diff from another version? Two different things you mixed up there. @mikeller already answered the question.

jflyper, 2017-10-29:
I think it was my work that made them default on.

IIRC, as baro and mag were to be made configurable, it was necessary to generate a reasonable default device with preprocessor macros for backward compatibility for `AUTO`. I should have introduced some `{baro,mag}_default_device` for this at this point, but it was left untouched and the generated default device was used as active device.

I think we will be looking into this when we go fully configurable.

mikeller, 2017-11-01:
@Arakon: Thanks for bringing this up, I think the answer is somewhere in between. :-P Can you please close this?

closed 2017-11-01 by Arakon