betaflight/betaflight #4462: In - flight memory corruption on CLRACINGF4

mikeller, 2017-10-30:
Placeholder issue for discussions and resources pertaining to this issue, as reported by multiple users.

@bnn1044, @codecae, @AndersHoglund.

mikeller, 2017-10-30:
[LOG00003.BFL.zip](https://github.com/betaflight/betaflight/files/1427085/LOG00003.BFL.zip)

AndersHoglund, 2017-10-30:
Here are  3.2.1 built in CygWIN, to test if we have a build env problem similar to the IMPULSERCF3 problem. There CygWin builds worked, Ubuntu builds failed.

[betaflight_3.2.1_CLRACINGF4_cygwin_build.zip](https://github.com/betaflight/betaflight/files/1427119/betaflight_3.2.1_CLRACINGF4_cygwin_build.zip)

jflyper, 2017-10-30:
Which build does this happen?

bnn1044, 2017-10-30:
@jflyper. some user reported on the lastest release 3.2.1

codecae, 2017-10-30:
I'm seeing reports on the Facebook group about the issue occuring with an XSR-M.  Given that the diffs show no changes in the rx code between "working" and "not-working" builds, I think it's fair to rule out any code in the rx tree.

codecae, 2017-10-30:
Just food for thought -- the same build is in use on the original clracingf4.  I haven't heard of any issues with anyone using it on the old hardware.  If it were a build problem, wouldn't the issue be more widespread among all hardware using this target?

AndersHoglund, 2017-10-30:
OK, makes sense. Seems more like a pure hardware problem then. Maybe not our issue at all.
Unless there are hw changes needing fw support/changes. Do we know what the hw diffs are?

bnn1044, 2017-10-30:
I don’t think it is betaflight’s problem. I will get the board has issue
back and check with the hardware.
The hardware is indentical to the old board just pcb layout different.

mikeller, 2017-11-07:
@bnn1044: I think I read somewhere that this has been identified as a problem with the voltage regulators. Can this issue be closed?

bnn1044, 2017-11-07:
Yes.

closed 2017-11-07 by mikeller