betaflight/betaflight #4464: PID modification by Spektrum telemetry

mangoldc, 2017-10-30:
There are intentions to implement PID modification by telemetry screen for Spektrum ( e.g. something like there is existing for RF)?

AndersHoglund, 2017-10-30:
To my knowledge, there are no public info or specs on it from Spektrum. Propriatory RF- Spektrum feature, I think. Will check some more though....

mangoldc, 2017-10-30:
As far as I understud it should be possible to build a screen in the TX by read out of telemetry data from FC....!?

AndersHoglund, 2017-10-30:
Sure, easy if you know how and have the required details on what data, and how, the Tx expects. 
I did ask, got a quick reply. 

> I’m sorry, we can’t talk about our future plans for that with anyone outside the company at this time.

AndersHoglund, 2017-10-30:
It could of course be done using stick gestures and the general "Text Gen" telemetry sensor. 
The public spec is here: https://github.com/SpektrumFPV/SpektrumDocumentation/blob/master/Telemetry/spektrumTelemetrySensors.h

I did ask for a plans on a specific PID ctrl menu and a PID tm sensor which is a different thing.

AndersHoglund, 2017-10-30:
Maybe we could run CMS on the Spektrum Tx display ? Then you would have PIDs and everything else too.

mangoldc, 2017-10-30:
Sounds like a plan, no idea if that works and how big the effort is. Would be like OSD with stick commands but display on the TX instead,

jflyper, 2017-10-30:
How big is the bandwidth of downlink?

AndersHoglund, 2017-10-30:
I really do not know about bandwidth, as of now the big problem is the small 9x13 text box. I have a SRXL textgenerator displayport up and running, registered to CMS. But only truncated menu pieces on Tx screen. Not looking too good or promising, I'm afraid.

mangoldc, 2017-10-31:
Agree, pity is Spektrum not more cooperative :-(

mangoldc, 2017-10-31:
Look that - toughts!?
https://youtu.be/cJth9s4lxN0

AndersHoglund, 2017-10-31:
Yeah, I've seen that. And it is the TM Textgenerator 9x13 display in action, same as I use. Always possible to do a PIF adjustment and/or settings menu specifically adapted to that small textbox.
I am not sure we want to do that, rather see if it is possible to make CMS run on a small device. 
Anyone know the CMS minimum screen requirement?

DanNixon, 2017-10-31:
@AndersHoglund You can adjust the spacing of the rightmost column in the CMS. Doing this you might *just* be able to squeeze the PID/rate options into 13 columns (most of the CMS would not fit though).

AndersHoglund, 2017-10-31:
Yeah I have seen there are cli settings for that. I'll do some more experiments, project not entirely  cancelled, yet.... Maybe a sliding window on a larger virtual textbox? Not sure we have the bandwifth for that. 
Or, I could make the "displayport_srxl" etc ready to use for someone knowing CMS better that me. Divide the project in two parts, CMS and displayport.

DanNixon, 2017-10-31:
I remember the same discussion with respect to FrSky gear and it turned out to be no, not without some kind of string caching on the radio (which on the Taranis is not overly complicated, keeping them in sync with the firmware probably more so).

With a slimmed down CMS (maybe 3 lines: menu title, setting, value) it may well be possible if you can choose to send a subset of the lines rather then the entire text area on each update.

Is the sensor priority fixed by the protocol? If not then maybe text could be given the highest priority when disarmed and other sensors having higher priority when armed (just thinking out loud, I'm not really familiar with Spektrum).

mikeller, 2017-10-31:
Yeah, a slimmed down version (essentially the good old ms-dos type 'config entry by prompt' :-)) should be possible. But unfortunately this will probably only be possible for F4 (or better), since we're acutely running out of space on F3. :-(

AndersHoglund, 2017-10-31:
I will experiment with TM frame scheduling and prio. There are are two mandatory hi prio frames, QOS and RPM, and then a number of user frames. We have CURRENT and TEXTGENERATOR now. All shall be sent in sets of 3. I.e. "QOS", "RPM", "TEXT".  "QOS","RPM", "CUR".  "QOS", "RPM", "TEXT".  "QOS","RPM", "CUR", etc etc I will play a bit with it ..... 
If you watch the Spektrum Video, it is slow line rate there too.

AndersHoglund, 2017-10-31:
My prototype is here: https://github.com/AndersHoglund/betaflight/tree/spektrum_cms_telemetry
I have to take a break from this now, at least for a day or two. If someone wants to play with it, please do. Fork and pull as needed. Warning, kinda nasty prototype coding still. 
Running at 50Hz TM frame rate and a 9x30 text buffer for CMS, upper left 9x13 corner shows on Tx screen. CMS on OSD still there, the semi-secret way to change display applies ( Do the stick gesture to enter CMS multiple times, each gesture switches device).

AndersHoglund, 2017-10-31:
Prototype updated. If you pulled, pull again.

AndersHoglund, 2017-10-31:
I could not give this a rest, had to try a few things.... This now seems totaly doable. 
There are a lot of wasted screen estate in CMS that can be removed for small screens. A quick hack in CMS for now, needs a proper solution. And a lot of testing to see that eacjh abd every CMS menu page fits.  See attached screen shots, actual live CMS on DX18g1 telemetry screen.

[image]
[image]
[image]
[image]

mikeller, 2017-11-01:
I can't test since I don't have the hardware, but the screenshots look useful.

AndersHoglund, 2017-11-01:
Basic support for breaking long lines into multiple lines added, for numeric fields only so far. Problems with page swap also. 
[image]

HFMan, 2017-11-02:
@AndersHoglund - I did a pull on this and loaded it up.  I don't understand how to "activate" the CMS screens you built.  I did an auto-config of Telemetry from my DX9, and it found and added the "Text Gen" screen.  I go to that screen, and it has a title of "CMS".  Plug in the Quad, but nothing shows up on this screen.  What am I doing wrong?

AndersHoglund, 2017-11-02:
Yes, I know there are long string and right column problems. Handle numeric values so far. Very early prototype still. VTX power is not defined as a number, it is a string. Or OME_TAB actually.
  
Not sure what to do about the standard semi-secret CMS way if switching screens. Multiple activation stick gestures until it shows up, as you did discover.

AndersHoglund, 2017-11-02:
Menu paging fixed and OME_TAB multi line support added.

Abancalari, 2017-11-04:
@AndersHoglund If it's not too much trouble and you can set up builds the same way as you did for VTX testing, I can test this with my setup as well.

AndersHoglund, 2017-11-08:
An new, slightly different, prototype is available here: https://github.com/AndersHoglund/betaflight/tree/spektrum_cms_tm_proto2
Now every menu item takes up two rows on small narrow display.  First line the item text, second line the item value, right aligned. 

Previous prototype turned out to be a bit too big project, not really possible to adapt current CMS to some "dynamic page size" concept. Put on hold, will probably be scraped eventually....
This new prototype is much simpler, done in an evening. Also still very prototype-ish, will need a lot of tidy-ing and refactoring. WIP and PoC.

AndersHoglund, 2017-11-08:
@Abancalari 
http://andwho.sytes.net:8080/job/AndWho_Betaflight_Spektrum_CMS_over_TM_proto2/
But unfortunately the build fails on FURYF3OSD target, flash overflows. Will look into that later.

Abancalari, 2017-11-08:
No worries. My rig is ready for testing whenever. Another thought, what does the "DISPLAY" function on the config do?

AndersHoglund, 2017-11-08:
What DISPLAY function ? Where exactly??
Anyway, updates and re-bases pushed. Almost ready for a PR...
I have also found a few legacy problems with CMS on OSD, top and right hand column alignments. Changed CMS on OSD to also have value items aligned to right hand side of screen, same as CMS over TM. One special case less, saves flash space. And looks much better, imho.

Jenkins should be ready and green shortly. Please test on both TM/Tx and OSD.

Abancalari, 2017-11-08:
Seems it was carried over from CF and has just been left alone all this time. Not very useful though: https://github.com/cleanflight/cleanflight/blob/master/docs/Display.md

Anyways, will test tonight!

AndersHoglund, 2017-11-08:
Aha, that OLED display. Never tried it. Not sure if it can be used as a `displayport `device and run CMS.

jumatic, 2017-11-08:
Hello together, tested it with Omnibus F4 Pro V3 and DX6 G2. It works. Good thing for adjusting pids etc. on the field without the need of googles and notebook. 
Very useful.

Abancalari, 2017-11-09:
Ok. Testing on  DX9 and ClRacing F4. Menu works fine. It does seem like the values are being cut off on the right side though. For example on the PID PROF value (1,2,3) I see no number indicating which one I'm on. Same thing inside the PID menu or frequency changing for Smart Audio. It's just one digit every time.

AndersHoglund, 2017-11-09:
OK, great. Thanks for testing. What values are cut ? On OSD or on Tx/TM ?? 
I know some headings are longer than 13 chars, and will of course be truncated on the small 9x13 Tx screen. Also some long item name text lines are cut. But no item values I hope. 

I have been playing with TM transmissions, trying to send changed frames only, to get a bit better performance. But I run into task sync issues all the time. Loosing chars and whole lines randomly. Seems CMS task running at 60Hz and SRXL at 50Hz (whole TM task runs at 250Hz) are interfering, even though they should be non pre-emptive. Have not found any good "atomic test-and-set" ways yet. Current method or strcmp() and strncpy() is very crude.... Hints are welcome.

Abancalari, 2017-11-09:
Values were what was cut off on Tx/TM. Seems like they just need to be shifted one space to the left. Didn't take a look at the OSD. Will do that tonight.

AndersHoglund, 2017-11-09:
Hmm, what values? Example or screen shot please.

AndersHoglund, 2017-11-10:
OK, now I understand your problem with truncated lines. Your Tx with airware 1.20 does that, only 12 chars wide text box. Problem will be fixed in 1.21, 13 chars. And I am running 1.21beta, so do not see the problem. A fix is pushed, will be available on Jenkins soon. Also rebased to latest master HEAD.

AndersHoglund, 2017-11-10:
A few updates pushed, small performance increase and more stable text transmission. Was running too fast for the Tx or radio link to keep up, 50Hz lowered to 30Hz. Please reflash and retest.
Almost ready for a rebase, squash and a PR now, unless some blockers are found.

AndersHoglund, 2017-11-11:
PR #4545 created. We can continue discussions there. Closing this issue

closed 2017-11-11 by AndersHoglund