betaflight/betaflight #4466: Disarm switch is ignored

ijustwannafly, 2017-10-31:
Getting a wonky disarm situation recently.

Testing betaflight 3.2.1 latest release for friends builds.

Pikoblx board
Spracingf3 board

Frsky xm+ rx on both boards.

On cleanflight 2.1 both boards disarm correctly with a disarm switch.  

On betaflight 3.2.1 arm is correct but if the quad tumbles out of the sky for whatever reason disarming in midair is not recognized and instead the motors continue to spin while grounded and disarm switch is active.  The only way to stop the motors is to shut off tx to force fail safe.

The pre-arm feature is not activated/used/setup in these scenarios.

Hopefully I am missing something.

Both boards are not being restored with any dumps.  Simple flash, reboot, configure and test fly.

DanNixon, 2017-10-31:
What do you mean by "disarm switch"?

Arakon, 2017-10-31:
Check if "disarm regardless of throttle value" is off. 
I did have one situation last week where my copter crashed into tall grass and the motors kept spinning despite disarming, but I had put that down to possibly losing contact to the transmitter due to the grass.. but there may be a bug there.

mikeller, 2017-10-31:
@Arakon:

```
set disarm_kill_switch = OFF
set min_check = 1015
```

Could it have been that your throttle wasn't exactly at 0?

@ijustwannafly: A diff would help.

Arakon, 2017-10-31:
Nope. I use this exact setup on all my copters. Always have. Never an issue. Throttle was definitely at zero.
When I saw the props were spinning, I did check the throttle of course too, but they just kept going.

ijustwannafly, 2017-10-31:
@mikeller 

Definitely throttle was at 0.

Scenario:

Cleanflight 2.1 on pikoblx and spracingf3 - fly as normal and on a "crash" or other unexpected scenario disarm via switch immediately kills the motors regardless of throttle value/position.  I have had my quads hit a branch in mid-air and rather than wait for the quad to hit the ground I disarm quickly and it has not failed.

Betaflight 3.2.1 (official) on pikobls and spracingf3 - fly as normal and on a "crash" or other unexpected scenario disarm via switch does not immediately kill the motors.  When the quad is downed with throttle zero and the disarm command (via switch) sent the motors are still spinning.  Again, the only way to disarm motors in this scenario is to shut off the tx to force failsafe.  

I thought it was the receiver initially but back and forth between cf2.1 and bf3.2.1 it persists.

There could be a setting on bf I do not know about as I have been using cf2.1 for a good while now due to all my f4 boards just work on that build.

I can get a diff later after work.  On cf2.1 I usually go into the cli to to enable the dynamic filter option but on bf3.2.1 the configurator has those available through the gui.  Beyond that, I usually just change things that are accessible through the gui to keep issues at a minimum.

ijustwannafly, 2017-11-01:
Here's the diff from the pikoblx but after reviewing, it would seem this situation is an option/flag that didn't exist before (afaik).  

Disarm motors regardless of throttle value.   Default is on.  When turned off it causes the scenarios that I experienced.  However, it still doesn't explain why, when downed and throttle is at zero, the quad still refuses to disarm.  It would seem the throttle value is a first requirement for disarm in this situation rather than an "and" situation.  

One of the quads I was testing was having esc desyncs.  When the quad would desync I would disarm immediately in air as soon as it desynced so that it would not continue spinning the motors when down.  However, when the quad hits the ground the motors are still spinning even when throttle has reached 0.  Toggle the arm switch to arm then disarm with throttle 0 does not disarm.  Fail safe needs to be forced.
[image]

# diff
# Betaflight / FF_PIKOBLX (PIKO) 3.2.1 Oct 15 2017 / 19:33:27 (434db9a36) MSP API: 1.36
name JB_ATOM3
resource MOTOR 1 A07
resource MOTOR 5 NONE
resource MOTOR 6 NONE
resource PPM 1 NONE

feature AIRMODE
feature ANTI_GRAVITY
feature DYNAMIC_FILTER
beeper -GYRO_CALIBRATED
beeper -RX_LOST
beeper -RX_LOST_LANDING
beeper -DISARMING
beeper -ARMING
beeper -ARMING_GPS_FIX
beeper -BAT_CRIT_LOW
beeper -BAT_LOW
beeper -RX_SET
beeper -ACC_CALIBRATION
beeper -ACC_CALIBRATION_FAIL
beeper -READY_BEEP
beeper -MULTI_BEEPS
beeper -ARMED
beeper -SYSTEM_INIT
beeper -ON_USB
beeper -BLACKBOX_ERASE
aux 0 0 0 1800 2100
adjrange 0 0 3 950 1025 18 2
adjrange 1 0 3 1350 1425 15 2
adjrange 2 0 3 1450 1525 20 2
adjrange 3 0 3 1525 1575 17 2
adjrange 4 0 3 1625 1700 19 2
adjrange 5 0 3 1975 2050 16 2
adjrange 6 0 3 1775 1875 0 2
set gyro_sync_denom = 2
set gyro_notch1_hz = 0
set gyro_notch1_cutoff = 0
set gyro_notch2_hz = 0
set gyro_notch2_cutoff = 0
set acc_hardware = NONE
set blackbox_device = NONE
set dshot_idle_value = 600
set motor_pwm_protocol = DSHOT600
set current_meter = NONE
set battery_meter = NONE
set disarm_kill_switch = OFF
set deadband = 2
set yaw_deadband = 2
set pid_process_denom = 1
profile 0

set dterm_lowpass_type = PT1
set dterm_notch_hz = 0
set dterm_notch_cutoff = 0
rateprofile 0

set roll_srate = 63
set pitch_srate = 63
set yaw_srate = 63
set tpa_rate = 25

mikeller, 2017-11-01:
`disarm_kill_switch` has been around for a long time. But since you always used the default you probably never noticed it, until now that you accidentally switched it on (potentially through a config that wasn't reset properly during an update.

Looking at the rest of your config, I also notice `set min_check = 1015`, which is quite low, so that if (for example) there was mechanical play or drift, or you'd accidentally hit the trim switch on your TX, you could easily have been in a situation where your stick was at the low end, but the signal was still above 1015.

However, will try to recreate the situation that you are describing, and test to make sure that disarming works in a safe fashion.

ijustwannafly, 2017-11-01:
@mikeller the min check 1015 is arakons diff.  My pikoblx diff is shorter but does show the disarm kill switch off option.  I have switched it back on and issues are resolved but I still think there has to be a way to disarm when the switch is used while flying and then see an arm disarm sequence with 0 throttle to completely disarm.  That way it won't completely disarm while flying but you can still disarm via the switch when 0 throttle is seen.  This of course is aimed at the setting of disarm motors regardless turned off.

It's amazing how defaults work until they accidently break and then it's a facepalm moment when the cause is simple complacency.

mikeller, 2017-11-01:
Sorry, I got that mixed up.

> but I still think there has to be a way to disarm when the switch is used while flying and then see an arm disarm sequence with 0 throttle to completely disarm.

I don't quite understand what you mean by this. Arming is binary, your craft is either armed or not, and there is no 'half-armed'. The correct solution in almost all cases is to leave `disarm_kill_switch` on, unless you've got a very good reason to not do so, and in this case toggling the disarm switch will disarm under all circumstances.

ijustwannafly, 2017-11-01:
> I don't quite understand what you mean by this. Arming is binary, your craft is either armed or not, and there is no 'half-armed'. The correct solution in almost all cases is to leave [...]

In the scenarios I encountered, I disarmed mid flight with throttle above 0 (stick 0) but the motors did not disarm because of the disarm kill switch option turned off.  Even when throttle was returned to 0 the switch disarm was still not able to disarm the motors.  

So disarm mid flight throttle non-zero but motors don't disarm.  Now the switch is in the disarm position but when going to 0 throttle the motors will still not disarm.  This is fine.  The issue I think is when you really do want to disarm the disarm switch is not recognized after 0 throttle has been reached.  So a seco dary disarm sequence based solely for the disarm kill switch option off scenario may be needed.

I will try to reproduce this scenario later.

closed 2017-11-01 by ijustwannafly

mikeller, 2017-11-01:
@Arakon: I tested this with settings from your diff. I was *not* able to reproducing the craft *not* disarming when the arming switch was switched off while the throttle was low.

Arakon, 2017-11-01:
Strange. Throttle was definitely at 0, I am using the hall gimbals, there's zero jitter or drift.
It may still be down to the receiver somehow then. I updated to a current 3.3.0 build for the time being and will see if it happens again.