betaflight/betaflight #4470: [Feature Request] FPV Camera Control BF3.2.1

Poitivin, 2017-10-31:
FPV Camera Control is a great new feature in BG 3.2.1 but for those using minimOSD there seems to be a "collision" on the way the sticks are used

BF FPV Camera Control needs both sticks centered and then yaw full right to get enabled.
MinimOSD needs both sticks centered and then yaw full right and full pitch up to get enabled

Is there a way to have the FPV Camera Control having a different stick control to avoid the minimOSD and FPV Camera Control colliding on the way these are activated?

Thanks
Fernando

DieHertz, 2017-10-31:
So what is the issue? They aren't the same and neither should trigger the other one.

Poitivin, 2017-10-31:
Sure they are not the same (as I tried to describe) but both sticks center and yaw right wouldn't give an option to reach the minimOSD menu as it will trigger the cam OSD first...or am I thinking wrong?

DieHertz, 2017-10-31:
Why wouldn't it if minimOSD is activated not by yaw right but by yaw right + pitch up?
Did you actually try it before creating an issue? :-)

Poitivin, 2017-10-31:
I know about the pitch up for the minimosd but if you yaw right, wouldn't that trigger the cam OSD immediately and prevents the pitch up? As far I could see, once you yaw right there is no latency that would allow other functions to trigger.

I haven´t tested yet and am not trying to raise an issue but actually a feature request in the subject. If I'm wrong I apologize as the intention is to give awareness and ask this great feature to be used with no impact to other stick commands

DieHertz, 2017-11-01:
You're supposed to pull yaw right and pitch up at the same time, this is a different stick command than just yaw right and it shouldn't trigger camera control, if you're moving sticks separately you may run into issues with other stick commands as well.

closed 2017-11-02 by jflyper