betaflight/betaflight #4472: Feature request BattGo

DeepnineRC, 2017-11-01:
See www.battgo.org 
Maybe usefull for single cell voltage warming and lipo temperature.

Ordered a charger and battery.

DieHertz, 2017-11-01:
No protocol description on this one is provided and I doubt Chinese will give it away.
Having no capable battery at hand I can't really say whether it's viable to reverse-engineer it
or not, I only have the battgo checker but not willing to invest into charsoon batteries at the moment.

digitalentity, 2017-11-01:
I pinged BattGo about a week ago about protocol and hardware description. No answer so far.

mikeller, 2017-11-01:
I wonder if they have just stuck a battery management / control chip that is designed for laptop batteries into an RC LiPo? These often use encryption for ~money making~ counterfeit protection purposes...

DieHertz, 2017-11-01:
I'm certain they want to license the technology to batt manufacturers in for of a "blackbox" chip and closed-source software to manipulate it, otherwise they would have posted a technical description on their page.
Chances are it won't make it into betaflight then. On the other hand, when I took my BattGo BG-8S apart I found a counterfeit F103 typical for BLHeli32 ESCs, no special chips, therefore it may be a small matter of programming.

digitalentity, 2017-11-01:
I think the protocol is fairly simple. Most likely 1-Wire or UART. Data format could probably be reverse-engineered. The only problem is that on-wire data might be encrypted.

DieHertz, 2017-11-01:
I have the checker, send me a battery and we'll find out :-D

pawelsky, 2017-11-03:
Single wire UART @ 9600 baud. Unfortunately looks encrypted.
https://www.rcgroups.com/forums/showthread.php?2981195-ISTD-BattGO-BG-Linker

digitalentity, 2017-11-04:
Not sure it's encrypted. Not ASCII - for sure, which is understandable. Ordered a linker and a 1300 charsoon battery.

pawelsky, 2017-11-04:
I'm pretty sure it is, when you look at a longer recording of the communication even though the same data is being sent the data part in the messages is different every time.

jflyper, 2017-11-06:
What is the nature of the "_same message_"?

I think the encryption is pretty much weak, as there seems to be extremely short blocks (only 16-bits long) as far as the examples in rcg tell.

I'm interested in the initial few message exchanges upon connection from two or more different pack instances to find out if there is a key exchange protocol running.

mikeller, 2017-11-06:
@jflyper: The sad reality is that it does not matter how strong the encryption is - in the USA any device created to circumvent it is illegal, and its author and user subject to potential prosecution. (see https://en.wikipedia.org/wiki/Digital_Millennium_Copyright_Act) :cry:

ledvinap, 2017-11-06:
@mikeller : Some of us live in partially sane countries: https://en.wikipedia.org/wiki/Reverse_engineering#European_Union

digitalentity, 2017-11-06:
Even under DMCA a person who is in legal possession of a program, is permitted to reverse-engineer and circumvent its protection if this is necessary in order to achieve "interoperability" with other devices and programs. 

So a legal owner of BattGo-enabled battery is allowed to breat the on-the-wire ecryption to acheive interoperability with Betaflight.

mikeller, 2017-11-06:
@ledvinap: Even people living in partially sane countries have a history of getting arrested for DMCA violations when making the grave mistake of entering the USA: https://en.wikipedia.org/wiki/United_States_v._Elcom_Ltd.

@digitalentity: As far as I am aware, this exemption clause exists, but it can only be applied to a list of specific cases: https://en.wikipedia.org/wiki/Digital_Millennium_Copyright_Act#Anti-circumvention_exemptions I can't find anything in there that seems to be pertaining to Betaflight.

digitalentity, 2017-11-06:
@mikeller I'm not a DMCA lawyer, but the clauses you are referring to exist in addition to the explicit permission of a *lawful owner* to circumvent protections:

17 U.S.C. 1201(f)(1) and (2):
> (1) Notwithstanding the provisions of subsection (a)(1)(A), a person who has lawfully obtained the right to use a copy of a computer program may circumvent a technological measure that effectively controls access to a particular portion of that program for the sole purpose of identifying and analyzing those elements of the program that are necessary to achieve interoperability of an independently created computer program with other programs, and that have not previously been readily available to the person engaging in the circumvention, to the extent any such acts of identification and analysis do not constitute infringement under this title.
> (2) Notwithstanding the provisions of subsections (a)(2) and (b), a person may develop and employ technological means to circumvent a technological measure, or to circumvent protection afforded by a technological measure, in order to enable the identification and analysis under paragraph (1), or for the purpose of enabling interoperability of an independently created computer program with other programs, if such means are necessary to achieve such interoperability, to the extent that doing so does not constitute infringement under this title.

pawelsky, 2017-11-06:
@jflyper 
> What is the nature of the "same message"?

Messages come in sequence which is repeated over and over again, therefore one can assume then that these are messages carrying the same data, yet the payload is very different each time.

> I think the encryption is pretty much weak, as there seems to be extremely short blocks (only 16-bits long) as far as the examples in rcg tell.

I'm not an expert in encryption, so not sure if I understand why you would assume the encryption is week based on the fact that there are some messages which are 16 bits long. There are messages that are longer than 16 bits in the example on RCG.

> I'm interested in the initial few message exchanges upon connection to find out if there is a key exchange protocol running.

There is are initial messages with source 01 and target 00 (broadcast?) with 13 bytes of payload (different in every message) sent a couple of times to which the battery eventually responds with a message with 12 bytes of payload.

DieHertz, 2017-11-06:
Have that's some sort of DH protocol handshake?

jflyper, 2017-11-06:
@DieHertz 
> Have that's some sort of DH protocol handshake?

Don't know yet.

@pawelsky 
> why you would assume the encryption is week based on the fact that there are some messages which are 16 bits long

16-bits means there are only 65,536 different patterns of cleartext at the most for the message, which is an extremely small space. Usually, when a message is short, it is padded to some length to avoid the space being too small.

> sent a couple of times to which the battery eventually responds

If those message are the same, then the protocol (not the cipher) is weak. It will provide an crypt-analyzer  to build and test a protocol module that mimics the battery.

jflyper, 2017-11-06:
Would STM32-CRYP-LIB be the starting point?

ledvinap, 2017-11-06:
@jflyper: stream ciphers work quite fine with short messages - message context is encrypted and unreadable, but other checking mechanism is necessary to detect tampering (with XOR based stream cipher it is easy to selectively modify messages. If you can guess the plaintext, you can change it).

From provided trace on rcgroups, it doesn't look like strong encryption - too much repetition / low entropy. But that is just wild guess, longer trace would be necessary ...

jflyper, 2017-11-06:
@ledvinap Doesn't stream ciphers require reliable transport for synchronization?

pawelsky, 2017-11-06:
@jpflyer
> If those message are the same,
As I wrote in my previous post, they are not - payload is different in each of these messages.

pawelsky, 2017-11-06:
> I'll update the RCG thread with longer trace later tonight

Done (see first post)

ledvinap, 2017-11-06:
@jflyper : Yes. But self-synchronizing ciphers exits. It is also possible to re-negotiate on mismatch (but that may very easily lead to weakness).

It would be interesting to see what happens when communication error is introduced in BattGo protocol ...

ledvinap, 2017-11-06:
Hmm ... long sample looks totally random when taken globally (entropy 7.97 bits/byte)... 

but some sequences are interesting(from random places), some replies looks correlated (some sequence with offset)
```
81 0B 0B 0D 0F 11 13 15 17 19 1B 1D 1F
BD 18 FB A1 F6 A0 F6 A1 F6 F8 F6 39 F8 18 F8 18 F8 18
11 66 10 66 11 66 48 66 89 68 A8 68 A8 68 A8
D8 38 58 B8 D8 38 58 B8 D8 38 58
```

BTW: I wouldn't expect someone using 16bit sum of bytes as checksum to implement encryption correctly ...

ledvinap, 2017-11-06:
```
[     64061] AA 02 01 0C 77 B6 0D 0F 0E E0 1E 20 D1 3F C1 BF 14 05 
[     64140] AA 01 02 04 F7 37 8C 8F 50 02 
[     64149] AA 02 01 0C 76 B7 0F 0E 0D E1 1D 21 D2 3E C2 BE 15 05

ledvinap, 2017-11-06:
@pawelsky : Can you please try to sample large amount of similar data? Something where minimal changes between replies are expected - maybe repeatedly reading battery serial? 2byte request may be interesting (only 65k variants)

pawelsky, 2017-11-06:
One regularity easy to spot is also that the each message before the 0x19 long one has payload with 2 equal bytes or bytes which are exactly 0x10 apart :)
```
32 32
A1 A1
2F 3F
BD CD
55 55
E3 E3
72 72
01 01
8E 9E
1C 2C
AA BA
39 49
C7 C7
55 55
E3 E3
72 72
01 01
8E 9E
1C 2C
AA BA
39 49
etc.
```

Unfortunately I cannot select messages to be sent - the USB dongle does it by itself in a loop

If I had to guess I would say that the message of 0x13 length carries the cell voltages as it is sent most often. Also the request before it is 4 bytes long which would match the PC interface command where you specify the command (probably represented by first 2 bytes) and then range of cells (00 03) for which the data is to be sent (next 2 bytes) . 

One of the messages must also carry the easy to identify manufacturer name ("Charsoon"), but I'm not sure which one.

ledvinap, 2017-11-07:
@pawelsky : 
Phase with 4B command -> 12B reply,
command[0] bytes are mostly 44 (0x2c) apart.
- When 0xff would follow, 0x01 is sent instead and bits in next byte are modified

Reply to 2b command is 'encoded' depending only on  reply[0] (replies with same reply[0] are identical)

2B Cmd -> 13B reply: there is something weak about low-1-bit replies:
```
AA 02 01 0D 08 DB 90 90 90 90 90 90 90 90 90 90 90
AA 02 01 0D 18 EB A0 A0 A0 A0 A0 A0 A0 A0 A0 A0 A0
AA 02 01 0D 80 43 08 08 08 08 08 08 08 08 08 08 08
AA 01 00 0D 81 0B 0B 0D 0F 11 13 15 17 19 1B 1D 1F
AA 02 01 0D A1 62 6B AD EF 31 73 B5 F7 39 7B BD FF
AA 02 01 0D A2 61 6E B2 F6 3A 7E 82 86 8A 8E 92 96
AA 02 01 0D A3 60 6D B3 F5 3B 7D 83 85 8B 8D 93 95
AA 02 01 0D A4 67 74 BC C4 CC D4 DC 24 6C B4 FC 04
AA 02 01 0D C0 03 C8 48 C8 48 C8 48 C8 48 C8 48 C8
AA 02 01 0D E8 3B B0 70 B0 70 B0 70 B0 70 B0 70 B0

ledvinap, 2017-11-08:
@pawelsky : It looks like everything necessary is in data provided. But I'm not good at playing with numbers, it would be easier if larger sample is provided.
Can you please sample few transactions, recording each for about 10 minutes?

pawelsky, 2017-11-08:
Sure, send me you email via PM on RCG and I'll send you a longer log later tonight. As I mentioned earlier I cannot select a particular transaction, the USB dongle calls them continuously in a sequence, so the log will just contain 10 minute log of the whole communication.

ledvinap, 2017-11-08:
First byte of each message is used as encryption key for following bytes. Message is XORed with value dependent on key.
I have table for low 5 bits of key: https://docs.google.com/spreadsheets/d/1E3HZp7wlAWTsHH35tWmfdfOYFm3KP0cXykwrMA4J8xc/edit?usp=sharing
Each column corresponds to one byte in message, each row to one key
Decoding key repeats after 16 bytes(for low-order 5 bits), only lower 5 bits of message are decoded. 
Decoding is done as `plain[index] = cypher[index] ^ table[key&31][index&15];`

Bit plain.0 is dependent only on bits key.0, plain.1 is dependent on key.(1:0) and index.0, plain.2 is dependent on key.(2:0) and index.(1:0) etc ..

For each subtable (and topmost bit in that table) lower half is inversion of upper half(key.4 is marked in table).

It should be possible to extract whole table for 8bit key, but it is a bit tedious.

I'm missing something terribly simple, the table is probably build with repeated modulo addition/CRC or similar simple algorithm. I just don't see it ... Any ideas?

mikeller, 2017-11-08:
Nice work! There's probably a bit of money to be made from this for manufacturers of knock-off laptop batteries. :-D

jflyper, 2017-11-09:
@ledvinap Hats off!!!

ledvinap, 2017-11-09:
Full decoding table is in google sheet. (2 rows are missing, I'll fill it in eventually)

Table is derived directly from trace data (for longest reply, length 28):
- filter packets with longest length (28)
- sort them by `key` byte
- remove duplicated (reply is always the same - all packets with same key are identical)
- extract reply data - key `128` is weak(from previous analysis) , the key is 0x08 repeated
- zero out data in reply packet - XOR with known reply
- resulting values are key sequences (XOR-encoded zero packet)
- fill-in missing rows - `SEQ(key^0x80) == SEQ(key)^0x80` (this is valid for lower bits too, but you can't get higher bit this way)

Rows for key 74/202 are missing. Corresponding messages are missing in my trace, it's easy to extract it from shorter messages

ledvinap, 2017-11-09:
Added decoded packets to spreadsheet

ledvinap, 2017-11-09:
Encoding key used to sent data to battery seems to be derived from free-running counter. It is mostly dependent on time since last packet ...

pawelsky, 2017-11-09:
Nice :)

So, looks like my guess about message with 0x13 lenght was correct

44 00 03
45 00 03 B9 0E B8 0E B9 0E E0 0E 1C 00 00 00 00 00 00

Battery current state
-----------------------
Request: 
0x44 - ID
0x00 - start cell idx
0x03 - end cell idx

Response: 
0x45 - ID
0x00 - start cell idx
0x03 - end cell idx
0x0EB9 - 1st cell voltage (3769mV),
0x0EB0 - 2nd cell voltage (3768mV),
0x0EB9 - 3rd cell voltage (3769mV),
0x0EE0 - 4th cell voltage (3808mV),
0x1C - temp (28 degrees celsius)
0x000000000000 - ???

There is also another one that I recognize:

88
89 01 BE 0A E4 0C 68 10 D8 0E DC 05 00 00 32 00 20 03 EC 3C EC 3C 01 04

Battery factory data
-----------------------
Request: 
0x88 - ID

Response: 
0x89 - ID
0x01 - ???
0x0ABE - ??? default charge current (2750mA)
0x0CE4 - ??? low battery voltage (3300mV)
0x1068 - ??? full battery voltage (4200mv)
0x0ED8 - ??? default storage voltage (3800mV)
0x05DC - capacity (1500mAh)
0x0000 - ???
0x0032 - max charge rate (50 * 0.1 C)
0x0320 - max discharge rate (800 * 0.1 C)
0x3CEC - ??? (15596)
0x3CEC - ??? (15596)
0x01 - ??? battery type (LiPo)
0x04 - number of cells (4)

ledvinap, 2017-11-09:
The encoding table is killing me ... it's generation must be quite simple, but I don't see it ... 

It is probably some logical/arithmetic operation, must be injective (each column of table is transposition of 0..255).
Some rows are simple addition, but non-zero lower bits get in the way. 
Also each bit is affecting only higher ones + only inverts result at it's level.

ledvinap, 2017-11-10:
OK, decoding part is quite done.
Updated table(for up to 32 byte messages) and uploaded both logs I have.

I have working algorithm to generate key, but I am still missing something, it's slightly too complicated now ... I will post it eventually

It should be trivial to create replacement for BG-linker using FTDI serial + some javascript. Any volunteers? Or anyone donating BattGo battery?

CF/BF implementation should be quite easy, just implement initial negotiation (parallel battery communication is probably possible) and parse simple packet (@pawelsky provided enough info)

ledvinap, 2017-11-10:
One thing I don't understand is reason behind protocol obfuscation ... It's totally brain-damaged, it's trivial to break it (at least partially) and it has all telltale signs of poor security .. Also it seems that it's implementation is non-trivial, so it's wasting resources.

mikeller, 2017-11-10:
@ledvinap: See my comments above re DMCA - the encryption does not have to be effective for the manufacturer using it to be able to sue you for breaching the DMCA.

jflyper, 2017-11-11:
@pawelsky 
> Response:
>0x89 - ID
>0x01 - ???
>0x04 - number of cells (4)

May be:
0x01 - Parallel
0x04 - Serial

0x01 0x04 = 4S1P

pawelsky, 2017-11-11:
> May be:
> 0x01 - Parallel
> 0x04 - Serial
> 
> 0x01 0x04 = 4S1P
> 

I thought about that, but unfortunately I was not able to confirm that as the PC GUI does react to this byte being set to 2 or more, so I left it unknown for now.

ledvinap, 2017-11-11:
@mikeller : A bit off-topic, (and IANAL) but in this case even DMCA interoperability clause is not necessary - for DCMA you need something copyrightable to protect, and I can't imagine how battery voltage would apply (https://en.wikipedia.org/wiki/Lexmark_International,_Inc._v._Static_Control_Components,_Inc. was about cartridge loading program).

mikeller, 2017-11-11:
@ledvinap: The copyrightable part of it is the program that interfaces with the battery - just like in the Lexmark case it was found by the court to be the program that was embedded in the toner cartridges. (Yes, this was overthrown eventually, but only after the defendant spent the money required to appeal this up to the supreme court.)

ledvinap, 2017-11-11:
@mikeller : as I understand it, cartridge contained program that got copied (almost)  verbatim to unofficial cartridges.

ledvinap, 2017-11-11:
@mikeller In this  case only interface is affected, both battery and pc/linker side is left out

mikeller, 2017-11-11:
@ledvinap: Copying of the program was one of the things they got sued for and lost before they appealed. The second, only loosely related thing was breach of the DMCA: 

`On the DMCA claims, the court found that the SCC microchip circumvented Lexmark's authentication sequence,[19][20] and that the reverse engineering exception to the DMCA did not apply, because it only covers the independent creation of new programs that must interoperate with existing ones, and SCC did not create any new program.`

(from https://en.wikipedia.org/wiki/Lexmark_International,_Inc._v._Static_Control_Components,_Inc.)

jflyper, 2017-11-12:
@ledvinap 
> I have working algorithm to generate key, but I am still missing something

How about considering bits are lost during transposition due to short variable for intermediate values?

ledvinap, 2017-11-12:
@jflyper: the algorithm works, but it needs to iterate all 8 bits to get next key. I thought it may be optimized (getting previous key is much easier  operation, it could be used to get whole 128 bytes of key sequence fast).

But it's probably result of incompetent design, original algorithm may be even more  complicated

pawelsky, 2017-11-25:
Added some more messages to my [previous comment](https://github.com/betaflight/betaflight/issues/4472#issuecomment-343196748)

pawelsky, 2018-07-12:
Official list of some of the battery parameters
https://www.battgo.org/down/Intelligent%C2%A0battery%C2%A0parameter.zip

And possibility to purchase the BattGo modules (didn't try it yet, not sure if you need a developer account first)
https://www.aliexpress.com/store/product/BattGo-Best-choice-for-power-battery-Battery-Goes-Smarter/2857080_32884933978.html

BertoldVdb, 2020-08-24:
I needed to implement a piece of code to read the battery telemetry. This is the scrambling algorithm, it is super simple: 
https://gist.github.com/BertoldVdb/ff583068317b9ab947495f19efb2fc18

The main() function generates the table in the google doc, to verify that the algorithm is correct.

ledvinap, 2020-08-25:
Eh ... obviously I am very poor in cryptography ...

mikeller, 2020-10-24:
@ledvinap:

All your fault, they died because we never hacked it and added support for it to Betaflight. :stuck_out_tongue_closed_eyes: 

[image]

pawelsky, 2021-06-22:
BattGo is now exclusive to Horizon Hobby

[image]

BertoldVdb, 2021-07-03:
I open-sourced a Golang library I made to speak this protocol: https://github.com/BertoldVdb/go-battgo It has all the details needed to fully implement the interface and it supports multiple batteries attached on the same bus. Although this is probably not useful anymore seeing @pawelsky 's last comment.

I don't use these batteries anymore after having some problems with high self-discharge (to 0V). I found that TI makes many easy to use BMS chips that support a UART compatible* one-wire interface (HDQ) and have the added advantage of supporting current monitoring, protection and balancing.
There is no auto discharge function, although I assume you could configure balancing to 3.7V to make it discharge the cells slowly. You can only have one device per HDQ interface, but this is not important for me.

*) Not really UART like BattGO, but an UART can be used to emulate the waveform and talk to it.