betaflight/betaflight #4483: Buzzer type question

Bugg69s, 2017-11-04:
Hi, will active buzzer module work as buzzer ?
Or can i make it to beep with buzzer port or uart ?

Bugg69s, 2017-11-04:
Buzzer like this:
 http://s.aliexpress.com/EBjEzA7z

mikeller, 2017-11-04:
This is dependent on the board that you are using, but for most boards, connecting the 'I/O' pin (sic) to 'buz -' on the board will work.

Bugg69s, 2017-11-04:
Ok, thanks. I have omnibus f3

mikeller, 2017-11-04:
Just try it. Or ask the manufacturer of the board if it is supported. The schematics for boards are not usually publicly available, so nobody except for the manufacturer is going to be able to tell without testing.

Bugg69s, 2017-11-04:
Ok thanks :)

closed 2017-11-04 by Bugg69s

Bugg69s, 2017-11-05:
@Mikeller that buzzer is like a pwm buzzer

jflyper, 2017-11-05:
Omnibus F3 buzzer circuit is a low side switch. Active (self activated) buzzer works fine. Simple piezo + PWM does not work as the buzzer pin does not have a timer.

Bugg69s, 2017-11-05:
@jflyper can i remap buzer to like motor 5 and make it work ?

jflyper, 2017-11-05:
Yes, with the referenced module and Omnibus F3.

>  but it is pointless as you will waste the onboard switch circuit (unless you have borked the onboard buzzer circuit in some way or another.)

jflyper, 2017-11-05:
@Bugg69s 
I will not answer further questions.
You can redirect them to Betaflight thread or other appropriate forums on the net if there is any.