betaflight/betaflight #4486: Target FF_PICOBLX has crazy high PIDs

Arakon, 2017-11-04:
As defined by config.c, the PIDs for this (pretty average, hardware-wise) F3 board are:
```
            pidProfile->pid[PID_ROLL].P = 80;
            pidProfile->pid[PID_ROLL].I = 37;
            pidProfile->pid[PID_ROLL].D = 35;
            pidProfile->pid[PID_PITCH].P = 100;
            pidProfile->pid[PID_PITCH].I = 37;
            pidProfile->pid[PID_PITCH].D = 35;
            pidProfile->pid[PID_YAW].P = 180;
            pidProfile->pid[PID_YAW].D = 45;
```

Rates are also fairly high and not "safe defaults".
This causes issues when using the default values on copters such as the Kingkong Flyegg series.
Is there a specific reason for these high PIDs?

mikeller, 2017-11-04:
@MJ666 should be able to answer this.

MJ666, 2017-11-05:
This values only getting active for all of the brushed FuriousFPV FC's. This are the two Acrowhoop and the Nuke boards. This boards are used for whoop stye copters. I got this values from FuriousFPV and fly them myself with my two whoops without problems. For an standard 
PikoBLX FC this values should not get active and normal BF defaults should be set. If they get activated for some reason the brushed ESC detection is not working as expected. Likely the ESC drags the signal of motor port to ground. This is very unusual and I have not seen this before. A while ago I fixed an issue with the timing of the hardware detection of the AlienFlightNG F7. This caused the wrong version was detected. This problem could be related.

Arakon, 2017-11-05:
I believe it's the stock ESC/FC stack used in the Kingkong Flyegg copters.
https://www.banggood.com/Kingkong-FLY-EGG-Racing-Drone-Spare-Parts-2-3S-BLheli_S-10A-ESC-p-1172987.html
https://www.gearbest.com/flight-controller/pp_775587.html

Here you can see it defaulting to the high PIDs:
https://youtu.be/NA86jTYjdjA?t=3m24s

MJ666, 2017-11-05:
In the video you can see the PikoBLX board is detecting brushed ESC's this happens wen the motor1 signal is probed and has low potential. This is normally happen with the usual 10K pulldown resistor of the brushed ESC's. I have never seen this happen on brushless ESC's yet and also have a few builds with various brushless ESC (including one with an 10A 4in1 ESC 20x20mm). Attached you can find an test build of the 3.2.2 firmware with an extended delay to see if this is an timing problem as i have seen on the AFNG F7 controllers. Otherwise it is an specific issue of the ESC used for this builds. Normally brushless ESC should have high (inactive) potential at the motor inputs if nothing is connected. during probing the output of the motor signal is not enabled.

[betaflight_3.2.2_FF_PIKOBLX.zip](https://github.com/betaflight/betaflight/files/1443951/betaflight_3.2.2_FF_PIKOBLX.zip)

martinbudden, 2017-11-07:
@MJ666 , can we close this issue?

MJ666, 2017-11-07:
Have not gotten any feedback from a tester yet. Can not reproduce the issue on my copters here.

MJ666, 2017-11-13:
Looks to be this specific ESC is behaving special if not powered up and dragging the signal to low. Usualy I have never connected the battery during flashing the FC and have not seen this before. Will do some more testing to see if i can see this beheavior on any of my copters. This problem should affect a few more targets. The easy workaround  is to power the ESC an reset settings.

MJ666, 2017-12-07:
No more information. Looks to be related to the specific ESC. Closing this here.

closed 2017-12-07 by MJ666