betaflight/betaflight #4488: How to Arm upon Power-Up?  Or Enable Blackbox while disarmed

vikings4276, 2017-11-05:
Hello,
Is there any way I can configure my FC running Betaflight to arm upon power-up, or another way to enable blackbox recording while the FC is disarmed?
The application I'm using this for does not have a receiver, I am looking to log data from the FC to a serial OpenLog SD card while the FC is powered.

jflyper, 2017-11-05:
Assuming you are tethered and using motor tab sliders, you can use `blackbox_on_motor_test = ON`.

vikings4276, 2017-11-05:
I am not tethered, the FC is powered via a battery.  Is there any way to do this untethered?

jflyper, 2017-11-05:
I can't imagine your application, but try the command. If it doesn't fit your need, come back again.

vikings4276, 2017-11-05:
Ok, I'll be more specific.  I'm trying to use an old Naze32 FC to record gyro and accelerometer readings to an SD card for a high powered rocket application.  The FC is mounted in the nosecone and needs to log data for the duration of its flight.   Power is applied via a switch and the FC will be relatively inaccessible once powered.   How can I get the command you specified to "latch" even after power-off?

mikeller, 2017-11-05:
This application sounds like it will need a 'blacbox log always on' option (which could be useful for other situations as well).

mikeller, 2017-11-05:
Ok, this application sounds like it would need an 'blackbox log always on' option (which would be useful for other situations as well). Could probably be done by changing `blackbox_on_motor_test` into `blackbox_mode`, with `normal`, `motors_running`, `always`. @vikings4276: Are you up for some coding? ;-)

jflyper, 2017-11-05:
Oh ... I think there's no magic to accomplish what you want to do.
You may want to create a custom binary. I think changing `inMotorTestMode()` to `true` at line 1640 in `blackbox/blackbox.c` would serve your purpose.

https://github.com/betaflight/betaflight/blob/master/src/main/blackbox/blackbox.c#L1638-L1647
```
1638         if (blackboxConfig()->on_motor_test) {
1639             // Handle Motor Test Mode
1640             if (inMotorTestMode()) {
1641                 if (blackboxState==BLACKBOX_STATE_STOPPED)
1642                     startInTestMode();
1643             } else {
1644                 if (blackboxState!=BLACKBOX_STATE_STOPPED)
1645                     stopInTestMode();
1646             }
1647         }
```

mikeller, 2017-11-05:
@jflyper: See my comment https://github.com/betaflight/betaflight/issues/4488#issuecomment-341960199 (GitHub got the timeline wrong for some reason): I think there is value in adding this as a permanent option instead of doing it in a custom build.

jflyper, 2017-11-05:
I would add that option after automagic erase of oldest log is implemented...

vikings4276, 2017-11-05:
I have created my own binary that starts recording data upon power-up, but blackbox_decode and the blackbox explorer cannot open the file (possibly because the logging occurs through power-down which means stopInTestMode() is never called, which I assume means the file is never closed.  Is there another way to extract this data into a .csv?

mikeller, 2017-11-05:
@jflyper: What's the reason for your suggested course of action? To me, these two features look quite independent. 'logging always on' looks like the easier one to implement, and with the size of today's SD cards, erasing old log files to make space does not seem to be a super urgent requirement.

jflyper, 2017-11-05:
@mikeller A blackbox may have underlaying flash device.

@vikings4276 
> which I assume means the file is never closed

Naze uses a flash device for logging, right?

Then, there is no need to close (at least for the current firmware) the log, as records are simply appended to the beginning of fresh region of the flash. Configurator downloads and saves entire written region into one file.

mikeller, 2017-11-05:
@jflyper:

> A blackbox may have underlaying flash device.

Sure, but the configuration of when logging is enabled and when not should be independent of the flash device that is used. In the case of a flash device being used, there should probably be a periodic flush of the file buffer, to minimise the amount of data that is lost when the system loses power while logging (this will also be beneficial for 'normal' logging in the case of a power loss caused by a crash.

> Configurator downloads and saves entire written region into one file.

Could it be that this does not work because the write counter isn't persisted periodically?

vikings4276, 2017-11-05:
I am not using the internal flash, I’m using an openlog (serial) device connected to UART2.  The internal logger does not have enough storage capacity.
One solution I was considering was to open and close new files after x time of recording, so a sudden power loss only results in corruption of the last file.  Any way I can close the file and open a new one (without appending or power cycling)?

mikeller, 2017-11-05:
@vikings4276: In this case, the fact that you are loosing files (if this is the case) is not a Betaflight problem, but a problem that exists with the openlog device you are using. (Have you ever managed to get files written onto the log device, or could it be that your logger does not receive any log data at all?) Anyhow, closing / opening files on the logging device is not something that Betaflight has any control over - to the logging device the flight controller is just a device that sends a stream of data, that is then written to the storage device, and there is no way to send control information over this link.

vikings4276, 2017-11-05:
The logger does record the data files, but I cannot read it or parse it with the blackbox tools because I get an error: “Missing header data, are you sure this is a valid file?”   Although I can open the file in a text editor and the header data seems to be there.

Is there another way to parse this (binary) file into a readable csv?

jflyper, 2017-11-06:
What does CLI command `version` say?

vikings4276, 2017-11-06:
I created a fork from the current dev build on github version 3.3, if that's what you're asking.  For personal usage, I updated version.h using major rev 9 and minor rev 1, now it says:  

Betaflight / NAZE (AFNA) 9.1.0 Nov  5 2017 / 18:40:22 () MSP API: 1.37

mikeller, 2017-11-06:
@vikings4276: Have you tried opening the logs with the latest version of the blackbox explorer from GitHub? Also, not sure if the blackbox explorer tries to match the log format to the version that is reported by the firmware, so try setting this back to what it should be.

Jahnkeanater, 2017-11-07:
I'm not sure a FC accelerometer will work in that application. A rocket can accelerate at over 10g. If you are fine with the data during the motor burn being nonsense then its fine. I'm not sure how sensitive the accelerometer is in betaflight but it is possible to turn the sensitivity down so that max is 16g. They make less sensitive sensors for applications like this.

You might have better results building a simple Arduino data logger like this one. http://www.instructables.com/id/Arduino-Rocket-Datalogger/

jflyper, 2017-11-07:
So true...

vikings4276, 2017-11-07:
I tried to do it this way because I have an extra FC laying around.  I know there are better solutions out there...

PR #4498 (2017-11-07): Added option to have blackbox always on.
Fixes #4488.
closed 2017-11-07 by mikeller