betaflight/betaflight #4491: Blackbox log AmperageLatest not shown when ESC sensor

adrianmiriuta, 2017-11-05:
When using ESC sensors 
AmperageLatest does not get logged into blackbox logs.

mikeller, 2017-11-05:
@adrianmiriuta: What version is this? Do you have a `diff` for the configuration that you are getting this with? A quick look into current master looks like it is set appropriately:

https://github.com/betaflight/betaflight/blob/bee86890c7b9f1c1fad3ba22d17781cabf904f60/src/main/sensors/current.c#L209-L214

adrianmiriuta, 2017-11-06:
I have two quads with ESC sensor
an both the  AmperageLatest  does not get logged into the blackbox log.
One with Kiss24a one with Wraith35a.
The telemetry amperage works !

---------------------------------------------------------------------
# diff
# Betaflight / LUXV2_RACE (LUXR) 3.3.0 Oct 19 2017 / 13:40:52 (231d1bb7b) MSP API: 1.37
name warpx230

feature AIRMODE
feature ESC_SENSOR
feature ANTI_GRAVITY
feature DYNAMIC_FILTER
map RETA1234
serial 0 1024 115200 57600 0 115200
serial 1 0 115200 57600 0 115200
serial 3 64 115200 57600 0 115200
serial 4 32 115200 57600 0 115200
aux 0 0 1 1950 2075
aux 1 1 0 1525 1650
aux 2 2 0 1700 1825
aux 3 6 2 1950 2075
aux 4 7 3 1950 2075
aux 5 13 1 1450 1550
aux 6 26 1 1950 2075
set gyro_sync_denom = 2
set gyro_notch1_hz = 0
set gyro_notch1_cutoff = 0
set gyro_notch2_hz = 0
set gyro_notch2_cutoff = 0
set min_check = 1000
set max_check = 2000
set rc_interp_ch = RPY
set dshot_idle_value = 270
set motor_pwm_protocol = DSHOT600
set vbat_max_cell_voltage = 45
set vbat_warning_cell_voltage = 34
set current_meter = ESC
set battery_meter = ESC
set beeper_dshot_beacon_tone = 1
set imu_dcm_ki = 31
set deadband = 7
set yaw_deadband = 17
set debug_mode = NOTCH
profile 0

set dterm_notch_hz = 0
set dterm_notch_cutoff = 0
set vbat_pid_gain = ON
set setpoint_relax_ratio = 20
set dterm_setpoint_weight = 60
set p_pitch = 61
set i_pitch = 47
set d_pitch = 23
set p_roll = 61
set i_roll = 47
set d_roll = 23
set p_yaw = 61
set i_yaw = 47
set d_level = 50
set horizon_tilt_effect = 50
set horizon_tilt_expert_mode = ON
rateprofile 0

set thr_mid = 9
set thr_expo = 71
set tpa_rate = 75

#
----------------------------------------------------------------
# diff

# version
# Betaflight / OMNIBUSF4SD (OBSD) 3.3.0 Nov  4 2017 / 18:44:10 (7269e97) MSP API: 1.37

# name
name warp270

# resources

# mixer

# servo

# servo mix

# feature
feature -OSD
feature TELEMETRY
feature AIRMODE
feature ESC_SENSOR
feature DYNAMIC_FILTER

# beeper

# map
map RETA1234

# serial
serial 0 1024 115200 57600 0 115200
serial 2 64 115200 57600 0 115200
serial 5 32 115200 57600 0 115200

# led

# color

# mode_color

# aux
aux 0 0 1 1950 2075
aux 1 1 0 1525 1650
aux 2 2 0 1700 1825
aux 3 6 2 1950 2075
aux 4 7 3 1950 2075
aux 5 13 1 1425 1550
aux 6 26 1 1950 2075
aux 7 31 2 925 1050

# adjrange

# rxrange

# vtx

# rxfail

# master
set gyro_notch1_hz = 0
set gyro_notch1_cutoff = 0
set gyro_notch2_hz = 0
set gyro_notch2_cutoff = 0
set mag_hardware = NONE
set min_check = 1000
set max_check = 2000
set rc_interp_ch = RPYT
set serialrx_provider = SBUS
set blackbox_p_ratio = 64
set dshot_idle_value = 330
set motor_pwm_protocol = DSHOT600
set vbat_max_cell_voltage = 45
set vbat_warning_cell_voltage = 34
set current_meter = ESC
set battery_meter = ESC
set imu_dcm_kp = 3000
set imu_dcm_ki = 31
set small_angle = 45
set deadband = 7
set yaw_deadband = 17
set tlm_halfduplex = OFF
set frsky_unit = METRIC
set debug_mode = NOTCH

# profile
profile 0

set dterm_notch_hz = 0
set dterm_notch_cutoff = 0
set vbat_pid_gain = ON
set setpoint_relax_ratio = 20
set p_pitch = 61
set i_pitch = 47
set d_pitch = 41
set p_roll = 61
set i_roll = 47
set d_roll = 41
set p_yaw = 71
set i_yaw = 47
set d_level = 50
set horizon_tilt_effect = 50
set horizon_tilt_expert_mode = ON

# rateprofile
rateprofile 0

set thr_mid = 17
set thr_expo = 67
set tpa_rate = 70

#
---------------------------------------------------------------------------

mikeller, 2017-11-06:
That's odd. There's nothing out of the ordinary with these diffs, and for what I can tell the `amperageLatest` values should get logged. The amperage that is reported by telemetry is `amperage`, not `amperageLatest`, which gets filtered in the case of being read through the ADC.

PR #4499 (2017-11-06): Fixed logging of 'amperageLatest' when non-ADC source is used.
Fixes #4491.
closed 2017-11-06 by martinbudden