betaflight/betaflight #4496: [Feature request] Support for Matek F405-Mini

TheIsotopes, 2017-11-06:
BF actually not support the [Matek F405-Mini](http://www.mateksys.com/?portfolio=f405-mini) ... imho it's the same like Matek F405-OSD but with 32M flash instead of micro sd slot ... pls could someone create a target for it ... thx 👍

mikeller, 2017-11-06:
@TheIsotopes: Is there anything besides the flash chip that does not work with this board when using the MATEKF405 target? Support for the flash chip is already slated to come in 3.2.2: #4382

TheIsotopes, 2017-11-06:
imho i do not think so ... my request relates to this statement from matek on firmware & download site

> F405-MINI FIRMWARE
> Pls download the BF3.2.1 here.  FW_BF3.2.1 listed in GUI can’t support 32M flash on this FC now.
> 
> [Betaflight_3.2.1_MATEKF405 Mini](http://www.mateksys.com/Downloads/FW/betaflight_3.2.1_MATEKF405_mini.hex)

McGiverGim, 2017-11-06:
As mikeller said, in 3.2.2 version it will work. Matek page has a 3.2.1 firmware with the https://github.com/betaflight/betaflight/pull/4382 added to it. This change will be official in 3.2.2 version.

TheIsotopes, 2017-11-06:
@mikeller @McGiverGim 
it's all well and good ... but still missing the appropriate target to create the firmware hex file
the target board identifier is `MKM4` and instead of the sd card support should the flash support to be defined

mikeller, 2017-11-06:
@TheIsotopes: So does this mean that you can confirm that this is working with the MATEKF405 target (obviously without the flash for now)?

jflyper, 2017-11-07:
I have a MATEKSYS F405-MINI came factory flashed with MKF4, and it detects onboard FLASH.

Let me look into this.

mikeller, 2017-11-07:
@jflyper: I am already talking to Mateksys, target update is incoming.

jflyper, 2017-11-07:
Holy ****!!!
I've found out that CS is PC0 and it's working.
I'm now working on automatic blackbox logging device selection based on detection result 😭

mikeller, 2017-11-07:
I thought we already did that?
But if not, go ahead, that's something that's universally useful and should be target independent (there's other targets with either / or I think).

Mateksys said they'd send me the details for the board. I'll update the target once I got them.

jflyper, 2017-11-07:
> I thought we already did that?

I think it was co-existence of sdcard and onboard flash, not automatic selection of default logging device.

> But if not, go ahead, that's something that's universally useful and should be target independent (there's other targets with either / or I think).

Aaaand, it turned out that it's harder than I thought; logging device device descriptor descriptor referenced from blackbox codes is in `blackboxConfig` and can't be changed on the fly.

We have to define a variable that holds an active logging device, say `blackboxDevice`, independent from `blackboxConfig()->device`, and substitute all (but init) occurrence of `blackboxConfig()->device` to `blackboxDevice`. Thoughts?

mikeller, 2017-11-07:
I thought we'd already done that, to enable to write to a micro SD card if one is present, and onboard flash otherwise (i.e. change `blackboxConfig()->device` to be the _preferred_ device). But it could be that we just talked about it. But yes, I would think that, unless 'NONE` is selected, logging to whatever device is available seems to be reasonable, considering there is nothing else that users can do with the free space on the device.

TheIsotopes, 2017-11-07:
@jflyper hhmmmm ... funny ;-)
shared firmware from mateksys homepage have the MKM4 board identifier

@mikeller is #4502 based on the details of mateksys? ... if so, then imho we can close here

mikeller, 2017-11-07:
@TheIsotopes: Matek probably considered at one point releasing an extra target for the mini, but then realised that this would just increase the amount of work required for target maintenance, and the potential for bugs.

And yes, #4502 is based on information from Matek.

PR #4502 (2017-11-07): Added drivers for Matek F405 mini to MATEKF405 target.
Fixes #4496.
closed 2017-11-07 by mikeller