betaflight/betaflight #4501: [Feature Request] LED Dimming

TempleClause, 2017-11-06:
It would be great to be able to dim LEDs. In the LED tab it would be cool if one could assign a channel, like the color modifier channel, to dim the LEDs. 

Is somebody working on this? Otherwise, I can potentially try to develop that. I think it should be doable for me. I believe the Value part of the HSV color controls the brightness?

mikeller, 2017-11-06:
@TempleClause: This is already supported: https://github.com/betaflight/betaflight/wiki/LED-Strip-functionality#how-to-set-up-throttle-functionality-with-leds
Just set the colour choosen for the LED to mid-brightness, the one after it in the palette to maximum, and the one before it to minimum brightness.

TempleClause, 2017-11-06:
@mikeller I don't get how this should work. My end goal is to have one knob to change the colour and one knob to change the brightness. Is that really possible?

TempleClause, 2017-11-06:
https://www.youtube.com/watch?v=Xkxa3dOHtK8&feature=youtu.be

Got a prototype working :-) I'll try to tidy up and make a pull request...

mikeller, 2017-11-06:
Yes, if you want to be able to independently adjust colour and brightness this is not currently supported.

As a suggestion, I would go for making the brightness adjustable for all LEDs at once, since the data structure used for the per LED configuration does not have any space available, and we don't want to extend it for space reasons.

TempleClause, 2017-11-06:
Okay thanks good to know! So a global brightness level override for all LEDs it is. Can you point me in the right direction how and where I can save this value to? I should be able to do the html + css stuff but not sure where I could save the value to? I guess somewhere in the ledConfig? Any help is appreciated!

TempleClause, 2017-11-06:
Okay so after reviewing the code and data structure, I think the best solution is to add another mode_color. As of right now the highest one is 7 which is the color modifier.

(By the way, I think the GUI is very confusing as the color modifier is a global setting but the GUI lets you believe you can change it for every LED). 

So I think the best solution is to add a mode_color like this:
8 0 channel 

Is that okay like this?

mikeller, 2017-11-06:
Not my favourite way of making this configurable, since it is quite convoluted, but at least it is consisted with how the rest of the LED_STRIP configuration is done. After all, `SetModeColor()` is just used as an overly complicated way to set the value for a Throttle / AUX channel in `ledStripConfigMutable()->ledstrip_aux_channel`:

https://github.com/betaflight/betaflight/blob/7a6c4c0ebf1174e77b25565a32b4f450a9c9fea5/src/main/io/ledstrip.c#L1153-L1156

Just add `ledStripConfigMutable()->ledstrip_brightness_aux_channel`, and use `SetModeColor()` to assign it.

TempleClause, 2017-11-07:
I agree 100%. The whole LED code seems to be very unclean and hard to understand. It actually would make sense to rewrite the whole thing. But for consitency reasons it makes the most sense to do it like this.

Maybe it makes a bit more sense to add 
`mode_color 7 1 channel 
`
instead of 
`mode_color 8 0 channel`

What do you think?

mikeller, 2017-11-07:
Yeah, I guess `LED_AUX_CHANNEL` is a good match, and having 0 for colour, and 1 for brightness makes sense.

(Maybe we should add 2 for saturation as well. :grinning:)