betaflight/betaflight #4508: [Feature Request] Flash OSD at specific intervals

chime13, 2017-11-08:
I'd like to be able to have my OSD, or selected OSD elements, toggle on and off at specific intervals as opposed to being on all the time which I tend to find distracting while taking up screen real estate.
I propose:

OSD FlashEnable = (0 or 1) 0 = Disabled(original function) , 1 = Enabled
OSD FlashOn (1 to 10sec) "time display is on"
OSD FlashOff (1 to 60sec)"time display is off"

Example:  For OSD to turn on for 5 seconds, every 15 seconds
OSD FlashEnable = 1 
OSD FlashOn = 5
OSD FlashOff = 15

Vidalcris, 2017-11-08:
Flashing the OSD would be even more distracting, at least for me ;)
You can already use a switch to turn the OSD on/off.

chime13, 2017-11-08:
@Vidalcris "You can already use a switch to turn the OSD on/off."
Only if you have an extra AUX to assign the OSD on/off. 
I only have 2 AUXs with 2 Mix Switches and they're used up already. 
Also - OSD consumes power and not having it on all the time would save some juice. I fly strictly brushed 1s micros, Pilots that fly 2S - 4s quads are not normally concerned with little things that consume power like LEDs and +200mv VTX's for instance.
I'd rather have the OSD only display important messages when needed such as Low battery+voltage and possibly signal strength.

mikeller, 2017-11-08:
@chime13:

> Also - OSD consumes power and not having it on all the time would save some juice

Is this based on actual measurements, and can you give us some numbers?

My suspicion is that the power consumption of the OSD chip is probably more or less constant, independent of if it's used to display text or not. But I might be wrong.

DanNixon, 2017-11-08:
> I'd rather have the OSD only display important messages when needed such as Low battery+voltage and possibly signal strength.

Then enable the WARNINGS element and leave everything else disabled.

chime13, 2017-11-08:
I base my assumptions using both CLI commands "tasks" and "status". When I disable OSD, CPU usage goes down. 
Less CPU% consumed = less current consumed
In my case using an Omnibus F3,  OSD consumes 4.5% of CPU and GYRO/PIDS consuming 84.8% of CPU. Other functions are negligible at mostly .5% and all added up to equal 100% CPU usage.
I "ALSO" could be reading this completely wrong - hehehe

DanNixon, 2017-11-08:
The reason the OSD is said (quite rightly) to draw a noticeable amount of power is because of the video overlay chip (specifically the original Maxim part, the newer Airbot variants are better in this respect).

Just disabling the OSD feature does not turn off this IC and it is unlikely that not drawing will affect its current draw that much. The CPU load will make a negligibly small difference.

If you wanted to make confident assertions about current consumption you really need to measure it.

Vidalcris, 2017-11-08:
Could be cool to be able to enable warnings even if the osd is of by toggle ;)

jflyper, 2017-11-09:
>Could be cool to be able to enable warnings even if the osd is off by toggle ;)

That can't happen.

Overlay chip is initialized during initialization phase if it is enabled.
We can't suddenly initialize them on the fly (sic) when you want to pop up something on your goggles.

One more thing. The change in CPU usage is not related to actual power draw for the FC at all for two reasons; MCU continues to run even when it is idle, and the overlay chip draws power even if it is not initialized by software (without this feature, it will not be able to do video pass-through at all).

mikeller, 2017-11-09:
@jflyper: I think with 'off by toggle' @Vidalcris means 'switched off with the BOXOSD mode switch'. This _is_ possible, and I think there is work underway (or at least planned) to make the function of the OSD switch configurable (similar to what MWOSD has).

tavdog, 2022-05-09:
Pretty stale issue. Time to close ?

closed 2022-05-09 by haslinghuis