betaflight/betaflight #4510: Horizon angle lock

Yamaford, 2017-11-08:
Now that long range flights are becoming more popular I had a idea to allow us to cruise more efficiently to our destination. I'd like to try to use an aux channel to "lock" the pitch angle while in fwd flight. Basically tell the flight controller this angle I'm at now is where  I want you to assume level is. Stick movements would still allow us to move around just as you would on normal horizon mode but when the angle lock switch is set it goes back to that angle on pitch when the sticks are released. I believe this would make for an enjoyable flight. I think it's at least worth exploring.

martinbudden, 2017-12-02:
@Yamaford , can't you use the trims on your transmitter to do this?

Yamaford, 2017-12-04:
@martinbudden Probably but not "Set to this angle" at the flick of a switch and certainly not back to default just as fast.  With the trims you would be pecking away at it each direction. I suppose a POT might work better. Maybe I'll give that a shot when it gets above freezing.

digitalentity, 2017-12-04:
@martinbudden @Yamaford I use a mix on my TX to offset the pitch stick - works good, until you forget to turn the mix off when switching to ACRO. I'm thinking about implementing "level offset" as a mode and make it controllable via a pot myself, but I didn't do any implementation yet.

closed 2018-07-10 by stale[bot]