betaflight/betaflight #4513: 3D is hard to arm now

Dave50W, 2017-11-08:
To arm in 3.2 you now have to have throttle dead centre that is very hard to do with goggles on or with a very low deadband.

Can we change it back to how it was in 3.1.7 where you have throttle down all the way hit arm switch raise throttle until it starts up at centre stick then off you go?

Pilots are programming 3 way switch on their Tx so the middle position has 1500-2000 to make it easier to arm. Then flicking into 3rd position armed and back to 1000-2000 ( For + & - 3D )

This is dangerous because if you go past the middle position (armed 1500-2000) straight to 3rd pos you could have a quad armed with full throttle @ 1000

mikeller, 2017-11-08:
@dannixon might have some ideas.

DanNixon, 2017-11-08:
@Dave50W #4516 should fix the issue. Let me know if you want a build (and for what board) to test it.

Dave50W, 2017-11-08:
a build would be great thanks mate I run Impulsercf3

mikeller, 2017-11-08:
@Dave50W:

[betaflight_3.3.0_IMPULSERCF3.zip](https://github.com/betaflight/betaflight/files/1455511/betaflight_3.3.0_IMPULSERCF3.zip)

Dave50W, 2017-11-08:
Bench test works as it should :)

One bug beeper starts when armed not a big issue tho may even serve as a warning your armed.

Do you know if future release will revert back to this arming method?

Thanks very much this is much better.

mikeller, 2017-11-08:
@Dave50W: Thanks for testing. If you get to fly it too any time soon, please report back.

Do you use `acc_cal_on_first_arm`? That would explain the beeping.

I think since this change is a safety improvement, it should go into 3.2.2, which is due to be released next weekend.

Dave50W, 2017-11-08:
@mikeller Yes I will fly it tomorrow and report back.

I think the beeping is a warning you aren't at mid stick, my ACC is off

mikeller, 2017-11-08:
Ah, true, that would cause it to beep in this case. Probably not a bad thing to happen, it tells you that you have armed, but the props don't spin (yet). Should help with avoiding accidental arming and then getting a nasty surprise when moving throttle, which was what the change that 'broke' this addressed.

closed 2017-11-10 by mikeller

mikeller, 2017-11-10:
@Dave50W: Here's a build of to-be-released 3.2.2. Can you please test if arming with 3D works as it should on this:

[betaflight_3.2.2_IMPULSERCF3.zip](https://github.com/betaflight/betaflight/files/1462920/betaflight_3.2.2_IMPULSERCF3.zip)

Dave50W, 2017-11-11:
@mikeller sure I’ll update tonight and fly tomorrow.

Was unable to make the field yesterday

Dave50W, 2017-11-12:
@mikeller 3.2.2 works great with the switch arming for 3D and having the tone as an armed warning is working great also.

mikeller, 2017-11-12:
Thanks for testing, @Dave50W!

PR #4516 (2017-11-10): Improve 3D arming checks
Fixes #4513

See commit messages for the implemented behaviors for each 3D mode combination.

Ideally I'd like someone to give this a test before merging.