betaflight/betaflight #4517: Turtle Mode in 3D mode (deactivated) does spin the motors reversed

Zappelphilipp, 2017-11-08:
Version: AirbotF4 3.2

In case 3D mode is enabled in the Configuration tab and mode "DISABLE 3D" with "FLIP OVER AFTER CRASH" is active together, the motors spin in the wrong direction. This means if you crash and want to flip over again, it just would want to eat itself in the ground. 

Sidequestion: I hope 3D mode would be fixed with the new DSHOT commands but instead it looks like we need to configure our ESCs again as bidirectional which leads to the problem of halfed stick resolution which then leads to wrong PID behavior. see: https://www.reddit.com/r/Multicopter/comments/67fdso/betaflight_3d_setup_behaves_weird/

my question: I do not remember exactly, but it was a release candidate of 3.2 which had a mode like "reverse motors" which allowed to reverse the motors but behave exactly like normal without setting the escs to bidirectional mode. This is exactly what i am looking for 3D flight. just flip a switch and reverse the motors without reconfigure blheli and fiddling around with pids. Does some one know where to find the code?

mikeller, 2017-11-08:
The thing that is stopping Dshot commands from being used for motor reversal is that ESCs have to be disarmed for Dshot commands to be accepted. This makes the whole 'disarm all ESCs / reverse direction / arm ESCs' sequence quite brittle, and there would always be the risk of not all ESCs getting the direction change, rendering the craft unflyable when trying to go inverted.

I think the PID issue should be addressed by appropriate scaling instead.

Zappelphilipp, 2017-11-13:
okay, i see. thank you for the explanation.

closed 2017-12-10 by mikeller

ddrcoder, 2018-11-25:
I'm encountering the same issue with v3.5.2. Can we reopen this issue, or should I open a new one?

mikeller, 2018-11-25:
@ddrcoder: This was fixed (for 4.0) in 7026.

PR #4729 (2017-12-10): Crashflip (Turtlemode) 3d fix
Makes the turtles play nice and not fight in 3d mode.

The dshot command to set temp forward and reverse was not playing well when 3d mode for blheli_s escs (probably others).

Tested on the switched 3d mode

Might fix #4517