betaflight/betaflight #4518: FC not boot - Super_S F4 remap LED_STRIP 1 to D02. Not tested on other F4's.

Szpero, 2017-11-08:
Hi,

I'm stick with Super_S F4 FC and led strip handling.

OMNIBUSF4SD and REVO firmware tested.
Super_S LED pin is connected to D02 and REVO with default settings couldn't handle with led strip.

remap LED_STRIP1 to A00 working correctly

remap to D02 hanging FC on startup

If LED_STRIP is inactive in configurator you can remap to D02 but activating it later causing FC freeze on boot.

mikeller, 2017-11-08:
Well, looks like there is no target definition for this board, and it obviously does not match the hardware for the OMNIBUSF4SD. Ask the manufacturer to publish the correct target definition for their hardware.

Szpero, 2017-11-08:
Recommended target is REVO and batch is delivered with REVO on board. All additional functions are inactive.
Anyway:
Hardware side: D02 pin looks physicaly Free - leading to LED pad.
Software: D02 FREE on resource and resource list. Remaping LED strip (didnt tested motors yet) to D02 hanging chip. A00 - not. Other PWM related pins allowing me to remap.

It's quite strange for me why only D02 causing so such problem :/

Ech... perhaps again I need to use my soldering skills and solder tinny wire to STM :)

mikeller, 2017-11-08:
PD2 does not have a timer or DMA. This will not work, unless you update the target definition to have a timer / DMA.

Szpero, 2017-11-09:
Thanks a lot! This was enough enough for me. After checking CubeMX I almost fell from my chair. This is not design flaw in Super_S - it's design fail.

I need to solder tin wire to CPU.
I'm closing the case.

closed 2017-11-09 by Szpero