betaflight/betaflight #4531: Some I2C MAG does not work

McGiverGim, 2017-11-10:
I've a Omnibus F4 Pro (V3). Two months ago I added a Radiolink SE100 (M8N + HMC5983) to my quad. It worked perfectly but after some hours the hardware was dead, I don't know why. I bought another, the same model, but this does not work. Betaflight does not detect the MAG.
```
# status
System Uptime: 19 seconds
Current Time: 0000-01-01T00:00:00.000+00:00
Voltage: 0 * 0.1V (0S battery - NOT PRESENT)
CPU Clock=168MHz, GYRO=MPU6000, ACC=MPU6000, BARO=BMP280
SD card: Manufacturer 0x3, 15558144kB, 06/2017, v8.0, 'SC16G'
Filesystem: Ready
Stack size: 2048, Stack address: 0x10010000
I2C Errors: 0, config size: 1840, max available config: 16384
CPU:9%, cycle time: 125, GYRO rate: 8000, RX rate: 49, System rate: 9
Arming disable flags: RXLOSS CLI MSP
```
First, I thinked it will be a hardware issue, so I tried with iNav, and iNav detected it perfectly out of the box, so the hardware is working. 
I Have observed that the board in the old version (that worked in Betaflight) and the new (that don't work) are slightly different being the same model.
Here you have the complete diff:
```
# diff

# version
# Betaflight / OMNIBUSF4SD (OBSD) 3.3.0 Nov  9 2017 / 18:53:40 (6f46ba7) MSP API: 1.37

# name
name migmul martian

# resources
resource MOTOR 5 NONE
resource MOTOR 6 NONE
resource SERIAL_TX 3 NONE
resource SERIAL_TX 11 A01
resource SERIAL_RX 3 NONE
resource SERIAL_RX 11 A08
resource I2C_SCL 2 B10
resource I2C_SDA 2 B11

# mixer

# servo

# servo mix

# feature
feature SOFTSERIAL
feature GPS
feature TELEMETRY
feature DYNAMIC_FILTER

# beeper

# map

# serial
serial 0 4160 115200 57600 0 115200
serial 5 2 115200 57600 0 115200
serial 30 16384 115200 57600 0 115200

# led

# color

# mode_color

# aux
aux 0 0 0 1950 2050
aux 1 1 1 1250 1350
aux 2 1 1 1450 1550
aux 3 1 1 1650 1850
aux 4 2 1 1350 1450
aux 5 3 1 1250 1350
aux 6 3 1 1650 1850
aux 7 6 1 1750 1850
aux 8 10 1 1250 1350
aux 9 11 1 1250 1350
aux 10 11 1 1650 1850
aux 11 13 1 1250 1350
aux 12 26 0 1950 2050
aux 13 28 1 1050 1150
aux 14 32 5 1950 2050
aux 15 33 4 950 1200
aux 16 34 4 1800 2050

# adjrange
adjrange 0 0 2 1950 2050 18 3

# rxrange

# vtx

# rxfail
rxfail 5 s 1500

# master
set gyro_notch1_hz = 0
set gyro_notch1_cutoff = 0
set gyro_notch2_hz = 0
set gyro_notch2_cutoff = 0
set mag_declination = -110
set max_check = 1950
set serialrx_provider = IBUS
set motor_pwm_protocol = MULTISHOT
set failsafe_delay = 3
set failsafe_off_delay = 20
set failsafe_throttle = 1200
set failsafe_procedure = AUTO-LAND
set bat_capacity = 1800
set vbat_detect_cell_voltage = 55
set ibata_scale = 333
set small_angle = 180
set disarm_kill_switch = OFF
set gps_provider = UBLOX
set gps_sbas_mode = EGNOS
set gps_auto_baud = ON
set deadband = 3
set yaw_deadband = 5
set pid_process_denom = 1
set osd_rssi_alarm = 0
set osd_cap_alarm = 1600
set osd_tim1 = 0
set osd_tim2 = 1793
set osd_vbat_pos = 2433
set osd_rssi_pos = 46
set osd_tim_1_pos = 54
set osd_tim_2_pos = 2455
set osd_flymode_pos = 2145
set osd_throttle_pos = 2424
set osd_vtx_channel_pos = 376
set osd_crosshairs = 2250
set osd_ah_sbar = 2250
set osd_ah_pos = 2250
set osd_current_pos = 2400
set osd_mah_drawn_pos = 2370
set osd_craft_name_pos = 2440
set osd_gps_speed_pos = 2264
set osd_gps_lon_pos = 2129
set osd_gps_lat_pos = 2097
set osd_gps_sats_pos = 2119
set osd_home_dir_pos = 2126
set osd_home_dist_pos = 2092
set osd_compass_bar_pos = 2154
set osd_altitude_pos = 2294
set osd_pid_roll_pos = 391
set osd_pid_pitch_pos = 455
set osd_pid_yaw_pos = 487
set osd_debug_pos = 391
set osd_power_pos = 321
set osd_pidrate_profile_pos = 344
set osd_avg_cell_voltage_pos = 76
set osd_pit_ang_pos = 257
set osd_rol_ang_pos = 289
set osd_battery_usage_pos = 302
set osd_disarmed_pos = 2411
set osd_nheading_pos = 106
set osd_nvario_pos = 1
set osd_esc_tmp_pos = 82
set osd_esc_rpm_pos = 83
set osd_stat_max_dist = ON
set osd_stat_min_batt = OFF
set osd_stat_min_rssi = OFF
set osd_stat_max_alt = ON
set osd_stat_endbatt = ON
set debug_mode = FFT
set vcd_video_system = 2

# profile
profile 0

set dterm_lowpass_type = PT1
set anti_gravity_gain = 2000
set setpoint_relax_ratio = 50
set dterm_setpoint_weight = 105
set p_pitch = 65
set d_pitch = 40
set p_roll = 50
set d_roll = 36
set p_yaw = 90

# rateprofile
rateprofile 0

set rc_expo = 40
set rc_expo_yaw = 10
set thr_mid = 40
set roll_srate = 65
set pitch_srate = 61
set yaw_srate = 60
set tpa_rate = 0
```
But I've tested with `defaults` too, only adding this:
```
resource I2C_SCL 2 B10
resource I2C_SDA 2 B11
```
Things that I've tested:
- Ask for help in slack (thanks to @jflyper for the effort) without luck. So I generate this issue to see if we can find more people with the same problem.
- Betaflight 3.1.7, 3.2.1 and master without luck.
- iNav -> WORKING
- Disable the i2c overclock, and modify the i2c timeout in the code, without luck
- Compare the iNav code with the Betaflight code but my knowledge is very limited, so I didn't see nothing.
Maybe someone with more knowledge can compare the iNav and Betaflight code to see where the difference can be?

jflyper, 2017-11-10:
I2C mag repeatedly come up as not working properly in various symptoms (including "it works on iNav"), but it is usually the case that only those with proper measurement apparatus (e.g., DSO, logic analyzers - LAs) really help trouble shooting.

@McGiverGim Do you have an LA?

McGiverGim, 2017-11-10:
No, sorry, I don't :(

digitalentity, 2017-11-10:
Hm... Usually it's the other way arouind - hardware is reported working in Betaflight and not working in INAV :smile: 

In INAV it's usually an edge cases with marginal hardware quality. Some compass sensors will only work at lower I2C speeds (200 kHz). As far as I know Betaflight doesn't have the code to select low I2C speeds.

McGiverGim, 2017-11-10:
Taking a look at the code, Betaflight works at 800 and 400, and iNav has 100 and 200 too, but I think by default it is configured to work at 400.

jflyper, 2017-11-10:
You want to try 200?

I'm building for default 200 now.

McGiverGim, 2017-11-10:
I can try, for sure. If the changes in the code are minimum, you can told me where to change and what values you want to try and I can compile the code by myself, if it is easier for you. I don't want to give you a lot of work ;)

jflyper, 2017-11-10:
Yeap. That it, line 417. Change 400000 to whatever you want, 200000, 100000, ...

McGiverGim, 2017-11-10:
I need to disable the overclock too I think?

https://github.com/betaflight/betaflight/blob/8b334e5674f3a53ac2cd359d7ab2eb037d901dbb/src/main/target/common_fc_pre.h#L31-L32

I will remove the if completelly and put to 200000 the value always.

jflyper, 2017-11-10:
Good finding!

McGiverGim, 2017-11-10:
Some advance:
- Removing the if and using 200000 always, MAG not detected, and the configurator shows one I2C error.
- Removing the if, and using 400000, THE MAG IS DETECTED!!!! but the configurator shows the I2C error counting constantly incrementing.
- Letting the if, and putting the overclock to false, to let it automatically to 400000, MAG not detected and no errors.
- Removing the if, and using 400000, and putting the overclock to false, MAG detected but the configurator shows the I2C error counting constantly incrementing.

So the constant overclock does not seem to work. Any other ideas to try? :)

McGiverGim, 2017-11-10:
More info :)

Using 400000 and changing:
https://github.com/betaflight/betaflight/blob/8b334e5674f3a53ac2cd359d7ab2eb037d901dbb/src/main/drivers/bus_i2c_impl.h#L25-L27
The default timeout to the long timeout... the MAG is detected and I've no errors. I need to test if it works, if the north is detected right, but is an advance :)

McGiverGim, 2017-11-10:
Result of some tests: it does not seem to work. When I start the FC the configurator always point to the north, ignoring the position of the quad.

So the MAG is detected, there're no errors, but the data does not seem to be readed correctly.

McGiverGim, 2017-11-11:
After a lot of tests, now I think that the mag is faulty. With the changes, betaflight detects it, but the sensors tab of the mag always is 0, so I think they were not readed.
I've installed iNav, that detects the MAG out of the box... and the problem seems the same.  It does not detect the values in the sensors tab.
So if nobody has another idea, I think that I need to buy another MAG :( and I will close this issue.
Sorry for the time spended by all trying to help me.

jflyper, 2017-11-11:
😭

McGiverGim, 2017-11-11:
I have bought another, the same model, pigheaded that I am... The thursday will be here and we will see what happens...

catchra, 2017-11-12:
Hi please keep up the good work i have the same problem and would love to be able to use my mag but programing is beyond me.
The problem is still in the new ver. 3.2.2
and i can confirm that for me inav works with my neoM8N+HMC5883 with no problems but betaflight doesn't even if i use `set mag_hardware = HMC5883`

mikeller, 2017-11-12:
@catchra: At least telling us what board you are using, or how you are connecting your I2C device, whould give us a chance to help you with your problem... ;-)

catchra, 2017-11-12:
I am using the omnibus f4 v3 with sdcard on firmware 3.2.2
as for the mag it is connected to SCL/TX3 and SDA/RX3
on the ports tab i made no changes to uart3
on the configuration tab i turned on the mag
and in the CLI i used `set mag_hardware = HMC5883` and than `save`
these are the same setting i used in Inav

in the betaflight flasher i am using OMNIBUSF4SD at version 3.2.2

in Inav flasher i am using OMNIBUSF4SD at version 1.6-ALPHA2
with show unstable set to on

McGiverGim, 2017-11-12:
You must put this commands in cli:
```
resource I2C_SCL 2 B10
resource I2C_SDA 2 B11
save
```

catchra, 2017-11-12:
Thank you that work 

but this is the output from the CLI

```
Entering CLI Mode, type 'exit' to return, or 'help'

# resource I2C_SCL 2 B10

NOTE: B10 already assigned to SERIAL_TX 3.

Resource is set to B10

# resource I2C_SDA 2 B11

NOTE: B11 already assigned to SERIAL_RX 3.

Resource is set to B11
```

it did not work until i used the commands that you gave me.

McGiverGim, 2017-11-12:
Yes, don't worry. They are warnings.

McGiverGim, 2017-11-13:
To let more info about this problem, until I receive the new unit, I have found an iNav user with the same problem and the same product:

https://github.com/iNavFlight/inav/issues/2299

One of the comments say that the MAG maybe is a QMC5883 and not a HMC5983L. The photo shows an "DA 5883 7002". Is there a way to know if is HMC or QMC?

jflyper, 2017-11-13:
@McGiverGim The "DA 5883" looks like a QMC5883; there are numerous Chinese sites listing the DA chip as "_HMC5883 alternative_".

McGiverGim, 2017-11-13:
So I have two problems:
- The first, Betaflight does not support it
- The second, it seems that is not a standard QMC5883, because iNav detects it as a HMC5883 and if a configure it as a QMC5883 it appears as unavailable.
Maybe is a QMC5883 listening to the address of the HMC5883?

jflyper, 2017-11-13:
I think the manufacturer is offering custom chips with non-standard I2C address; data sheet says

_The default I2C address is 0D: 0001101_
_If other I2C address options are required, please contact factory._

http://img.filipeflop.com/files/download/Datasheet-QMC5883L-1.0%20.pdf

jflyper, 2017-11-13:
Here's the whole story.
https://github.com/iNavFlight/inav/issues/1903

McGiverGim, 2017-11-13:
Thanks! I will try to compile inav using QMC and the address of HMC, to see if it works... Maybe this will be the only "customization".

jflyper, 2017-11-13:
Data sheet mentions "_OTP Load_" immediately after POR and Soft Reset. I suspect I2C address is written in the OTP area...

digitalentity, 2017-11-13:
Oh yeah, that QMC5883 chip appears to be a real pain the back. When it's on the default address - all is ok, but if manufacturer configures it for something different - funny things starts to happen.

McGiverGim, 2017-11-13:
Yes, I'm playing a little changing values and registers, mixing the values neede to the HMC5883L and the QMC5883L, to see if I have luck... but not. Making it in a blind way seems like a pain...
If someone has some idea, I can try, but I think I have simply the "incorrect" MAG :(

McGiverGim, 2017-11-13:
After playing with it... I have a dirty code that **SEEMS TO WORK!!!**. But I don't know why and I don't understand a big part of the code. Finally it seems that my QMC5883 emulates almost anything of HMC5883. In theory the commands and registers for QMC5883 are different than HMC5883, but in this version not. Maybe a customized chip?

To make it working, I've changed this three things:

https://github.com/betaflight/betaflight/blob/8b334e5674f3a53ac2cd359d7ab2eb037d901dbb/src/main/drivers/bus_i2c_impl.h#L27
Change the timeout to `I2C_LONG_TIMEOUT`

Substitute this if:
https://github.com/betaflight/betaflight/blob/8b334e5674f3a53ac2cd359d7ab2eb037d901dbb/src/main/drivers/bus_i2c_stm32f10x.c#L414-L418
by `i2cInit.I2C_ClockSpeed = 400000;` in theory this can be achieved modifying the `I2C_OVERCLOCK` constant to `false` but it didn't work for me, maybe I made something wrong.

Without the above two changes, the chip was not detected. With them, it is detected like HMC5883. But the values readed by the MAG are always zero.

To solve the always zero values, I changed this:
https://github.com/betaflight/betaflight/blob/8b334e5674f3a53ac2cd359d7ab2eb037d901dbb/src/main/drivers/compass/compass_hmc5883l.c#L202
By `int ret = (int16_t)(raw[0] << 8 | raw[1]);` So I removed totally the gain. I don't know if this is important or not.

The code left has a lot of operations (like calculate the gain). I don't know if some of them maybe are not valid for QMC5883L but it works.

Can I let the code without the gain part or maybe can I change something to let it calculate in the right way?

McGiverGim, 2017-11-16:
To end with this, I've spoken with the manufacturer (radiolink), this is the most important thing that they say:

> We did the conversioncan and can use it as per HMC5983

They thinked it is compatible, then I told them the problems and solutions that I found, so they say:
> 1、Our new compass conversion doesn't support betaflight.
> 2、 It can works perfectly with apm and pixhawk now . 
> 3、 You can delete calculates the GAIN .

I think is not interesting make changes to the code to support this model (I don't know if others brands will do the same in a future), any thougts? I will use a custom firmware for me by the moment.

jflyper, 2017-11-16:
Meh... I don't think it's worth the effort to support random chimera... 😩

McGiverGim, 2017-11-16:
I don't know if is better to close the issue or let it open for some time (maybe others can find this info useful).

@jflyper what do you think?

ledvinap, 2017-11-16:
It does make sense to check calculated gain values and replace obviously invalid values with something reasonable ...

I2C speed reduction by disabling overclock should work, it should be fixed if it does not.

I2C_DEFAULT_TIMEOUT  is strange. @McGiverGim : Can you log bus transactions? Saleae or similar digital analyzer?

McGiverGim, 2017-11-16:
@ledvinap 

About the gain, the iNav code (that has support for native QMC5883) does not calculate the gain for this compass, but it is calculated for the HMC. I don't know why.

The I2C speed reduction changing the I2CX_OVERCLOCK defines, does not work. I've made several tests and this value is not readed. Maybe when passing the values from one variable to another in the code the value is lost. I'm looking at the code, but I have not found the place. This is a bug for sure.

Finally, no, I don't have any tool that let me log anything :(

jaisor, 2017-11-27:
I'm having similar issues but my sensor values seem to range way higher than they should in sensor data. Omnibus F4 SD with Ublox NEO-M8N. The values change as I rotate the craft so it seems like it is reading the mag sensor, however it always tends to point to South, I think because of the high Z value? I've tried calibrating and changing MAG orientation without success. I'm getting the build environment ready to start tinkering with the gain values and other ideas from above. Any suggestions are appreciated.
[image]

jflyper, 2017-11-27:
@jaisor Start by opening up your GPS puck and read the marking on the mag chip.

jaisor, 2017-11-27:
NEO-M8N-0-01, not sure I can read much of the chips markings, attached is a photo
[image]

Here is the BF status:
`System Uptime: 56 seconds
Voltage: 0 * 0.1V (0S battery - NOT PRESENT)
CPU Clock=168MHz, GYRO=MPU6000, ACC=MPU6000, BARO=BMP280, MAG=HMC5883
SD card: None inserted
Stack size: 2048, Stack address: 0x10010000
I2C Errors: 0, config size: 2084, max available config: 16384
CPU:8%, cycle time: 127, GYRO rate: 7874, RX rate: 49, System rate: 9
Arming disable flags: RXLOSS CLI`

I tried setting the magzero values as McGiverGim mentioned above and it did improve the reported sensor ranges, the compass heading now updates much better when I rotate the craft. Could this indeed be a calibration-not-working issue?

McGiverGim, 2017-11-27:
Your mag (U3 in the image) is an original HMC5883. So your problem is totally different.

andrejpodzimek, 2017-12-16:
@McGiverGim Thanks a lot for this report and for the investigation. I have [this Radiolink SE100](https://www.banggood.com/Radiolink-M8N-GPS-Module-UBX-M8030-for-Naze32-APM-CC3D-SP-F3-Naze32-Flip32-PX4-Flight-Controller-p-1051291.html), even though one can never be sure which one it is, given the vast number of Aliexpress, eBay and Banggood variants. :stuck_out_tongue_closed_eyes:

After my past interactions with Betaflight developers who kindly helped me set up my unusual configurations, I already have an oscilloscope. :sunglasses: That said, can I help? Because I have either the same problem as you or a similar one. [I've already asked on rcgroups](https://www.rcgroups.com/forums/showthread.php?2884546-Omnibus-f7/page37#post38817923) about my current issues with an `OMNIBUSF7V2` board &mdash; one with softserial and one with the compass.

What makes me somewhat doubtful about the similarity of my issue is the fact that your [hack](https://github.com/betaflight/betaflight/issues/4531#issuecomment-344032911) doesn't work in my case. It just doesn't have any effect whatsoever. Yet the whole issue may still be worth investigating.

@jflyper @digitalentity @ledvinap What and when should I measure with my oscilloscope? Should I capture what happens on SDA and SCL when the board boots on USB? Any other debugging suggestions?

andrejpodzimek, 2017-12-16:
A glance inside the puck / hexagon, so that we know what device it is (in my case).
[image]

McGiverGim, 2017-12-16:
Yes it is the qmc5883, my changes must work for you. Others users in iNav has treated them and worked. Maybe is a problem with omnibusf7v2, I don't know. 

Let's see if others will give you instructions for the oscilloscope. 😉

andrejpodzimek, 2017-12-17:
Nope, sadly the hack has no effect in my case. :disappointed: But there may be a number of other reasons for that. The most importantly, we may have the same GPS puck, but definitely we have a different board. [What I have](http://www.readytoflyquads.com/flip-32-f7-omnibus) is an `OMNIBUSF7V2`.

    # status
    System Uptime: 93 seconds
    Voltage: 5 * 0.1V (0S battery - NOT PRESENT)
    CPU Clock=216MHz, GYRO=ICM20608G, ACC=ICM20608, BARO=BMP280
    SD card: Startup failed
    Stack size: 2048, Stack address: 0x20010000
    I2C Errors: 2, config size: 2140, max available config: 32768
    CPU:35%, cycle time: 100, GYRO rate: 10000, RX rate: 49, System rate: 9
    Arming disable flags: RXLOSS CLI MSP

andrejpodzimek, 2017-12-17:
>> 2、 It can works perfectly with apm and pixhawk now .

Only noticed that^^^ now. So in that case the answer is [somewhere here](https://github.com/PX4/Firmware/blob/master/src/drivers/hmc5883/hmc5883.cpp). But I can't spot it at the first glance; there's no obvious branch like "support the non-standard stuff here".

andrejpodzimek, 2017-12-18:
Also, porting [this driver](https://github.com/iNavFlight/inav/blob/master/src/main/drivers/compass/compass_qmc5883l.c) from iNav to Betaflight may help. I looked into it for a while and it's not hard to do; just a few constants to copy with it and a small API difference in `qmc5883Read()` which I just hacked around, hoping that I would at least get the compass initialized, albeit with nonsensical values that miss some normalization or whatnot. Well, it builds and I have a QMC5883 option in the configurator, but that's all I have.

When I set QMC5883, the only thing that happens is that I start getting 7 I2C errors instead of 2-3. That's not exactly helpful. :cry: I also tried to remove the `QMC5883_ID_VAL` check (and an equivalent hack for `HMC5883`), but nope, it just won't get detected in any case. Tried 200 / 400 / 800 kHz, but nope. Address 30 and 13 tried for both cases, to no avail. Quite frustrating.

andrejpodzimek, 2017-12-18:
So on iNav, mag gets detected out-of-the-box, without changing any config options. But when setting the driver explicitly, it's `HMC5883`, **not** `QMC5883`. So it's a difference in the `HMC5883` driver and the `QMC5883` porting experiment was pretty much in vain. This is not a `QMC5883` device. And it's unusable, because it shows just three zeros all the time.

Porting of iNav's `HMC5883` into Betaflight is harder than for the `QMC5883` case. Lots of constants and renamed `struct` members would need inspection case by case. :fearful: And because it currently doesn't work anyway (showing zeros), I'm not sure it's worth it.

McGiverGim, 2017-12-18:
@andrejpodzimek what you have seen, is what I saw before:
- The radiolink QMC5883 has been "altered" to work as an HMC5883, the adress, registers, command, etc. are the same, so this is the reason because iNav detects it as an HMC5883.

But the "copy" is not exact, some things does not work, at least known:
- The mag gain calculation (we need remove the use of the gain)
- The timeout (we need to change it by a LONG timeout)
- The speed (we need to disable the overclock of the I2C bus)

But with this changes, it works (I used it and it works very, very good). I don't know if you have a branch with the changes, to see if they're correct. If not, I can build one hex for you, to test if it works.

andrejpodzimek, 2017-12-19:
Attempts to re-test that stuff with today's master didn't go exactly well. :angry: I filed #4779.

[Here's my branch](https://github.com/andrejpodzimek/gammaflight/tree/oldmaghack) with (what I believe) is [your hack applied](https://github.com/andrejpodzimek/gammaflight/commit/5c4b7c2912d2476f9c751f94691180bda25e7431). I had to cherry-pick the hack into a state before de5b0d53ffff22037a1d96409aa88ffcf51423ce to avoid #4779, so I've tested it on what things looked like ~2 days ago. (The current master head, [with your hack](https://github.com/andrejpodzimek/gammaflight/commit/be43b605a7dd03b0ac455783f37cbd542bb17310) or without it, won't reboot once I apply the diff mentioned in #4779.)

So, does the hack correspond to something that worked for you? Presumably, I've also tested it with ~10 different additional ad-hoc tweaks (200 kHz on I2C, skipping the device ID check in the driver etc.). None of those tweaks worked for me. :sob:

McGiverGim, 2017-12-19:
@andrejpodzimek I've take a look to your changes, and they seems good for me :(

The hack was tested by several users in the iNav thread, and worked for all, so maybe is another kind of incompatibility with your board. Maybe is a stupid question, but, are the I2C pins mapped with the resource command correctly?

andrejpodzimek, 2017-12-19:
Well, I hope so. Swapping SDA and SCL doesn’t help and the two pins are taken simply from the default mapping for UART3, which is shared with I2C on this board.

andrejpodzimek, 2017-12-19:
@McGiverGim Actually your point about the `resource` command is a great one! I recall that I only tried to swap the pins without your hack, not together with your hack. Now I've found an obvious problem: [**this**](https://github.com/betaflight/betaflight/blob/b924ac0c9ceb772d4f544834a8700df126506f5c/src/main/target/OMNIBUSF7/target.h#L88-89) versus [**this**](https://github.com/iNavFlight/inav/blob/912d131556302cf6ea67d59308e43dfe65ca8026/src/main/target/OMNIBUSF7/target.h#L91)! `UART3` is the serial port shared with I2C. And only one of the two headers can be right. :rofl: I configured the I2C pins in Betaflight based on the Betaflight header and [this](http://s3.amazonaws.com/zcom-media/sites/a0iE000000GWIo4IAH/media/catalog/product/8/-/8-11-2017_1-03-57_pm.png) schematic:
[image]

So this may be a combination of 2 issues:
  * Your hack is needed, or else it won't work.
  * The pins have to be configured correctly, or else it won't work.

That^^^ leaves us with 4 possible combinations and thus far I've only explored 3 of them. I'll test the 4th one once I get back home; looking forward to that!

andrejpodzimek, 2017-12-20:
1. I was completely wrong (again), as @jflyper pointed out in @4784.
1. **I got magnetometer data** by applying @McGiverGim's hack to iNav instead of to Betaflight.

This is weird. The hack doesn't work for me in Betaflight at all. [Here's my iNav version of it.](https://github.com/andrejpodzimek/gammaflight/commit/9b02f455c7bf54c69f232c3b008bdc9f7664e4fa) **It works** and I have some non-zero data. That is not to say that I've tested the sensor properly in any way or sense, but it simply works somehow:

[image]

Calibration also worked:
```
magzero_x = 131
magzero_y = -334
magzero_z = -44
```

So, what's the magic powder that Betaflight is lacking? The driver looks different, but I have no clue what magic numbers to focus on. [This Betaflight hack](https://github.com/andrejpodzimek/gammaflight/commit/5c4b7c2912d2476f9c751f94691180bda25e7431) doesn't work for me; no I2C there.

jflyper, 2017-12-20:
@andrejpodzimek 

If you are working with F7, you have to manipulate `bus_i2c_hal.c` instead of `bus_i2c_stm32f10x.c`.
It's around line 255:
```
251     if (pDev->overClock) {
252         // 800khz Maximum speed tested on various boards without issues
253         pHandle->Init.Timing = 0x00500D1D;
254     } else {
255         pHandle->Init.Timing = 0x00500C6F;
256     }
```

I think the `Timing` member goes to `I2C_TIMINGR` register, but as I look at the reference manual, this code does not manipulate prescaler value (both zero).

McGiverGim, 2017-12-20:
@jflyper is right, I didn't think in the two files.

What I did in my code was, in my `target.h` undef all the `I2Cx_OVERCLOCK` defines and define them again with a value of `false`.  This will serve for both files. The `pDev->overClock` is hardcoded to this defines in build time.

andrejpodzimek, 2018-01-03:
@jflyper @McGiverGim  So I've just tried the additional hack (`pHandle->Init.Timing = 0x00500C6F` in `bus_i2c_hal.c` regardless overclocking) and even set all the `I2C?_OVERCLOCK` defines to `false` in `common_fc_pre.h`. But it didn't help at all; still no compass. (Tested with current `master`.)

jflyper, 2018-01-03:
@andrejpodzimek Like I noted in the previous comment, I think you have setup some descent prescaler value for `Timing`.

And ... did you have a logic analyzer and dss? I wouldn't deal with something like this with proper measurement devices. It's just a waste of time guessing and half-blindly hacking things up.

yetipuk, 2018-01-23:
Guys, I am sorry if it's not the right place to ask. I've been struggling to make my GPS and Mag working for 3 weeks now  (plus 3 days struggling to build custom firmware using all related threads on RCG and GitHub).  

Please advise:
1. Board: OMNIBUS F4 SD
2. GPS: Ublox Neo-M8N,
3. Mag: chip says L883-2122 (I guess that is HMC5883...)
4. Inav: after I lowered timing to 200 it got finally detected, calibrated and seemed to be working but counting i2c errors at a rate around 2 errors every ten secs...
5. BetaFlight and Cleanflight: Nothing helps... Assigning serial 3 TX and RX to NONE and assigning i2c 2 to B10 and B11 - nothing. Manually setting mag_hardware to HMC5883 - nothing. The compass is not detected. 

In Betaflight after I assign I2C SCL and SDA to B10 & B11 the Gyro and Acc are affected in some way: first they are not detected and after a couple oа times the board connected/disconnected to USB -  they seem to start working properly, but sometimes they shut down again. 

I have problems setting up the windows-ubuntu-sygwin-bash... to compile custom builds (I am working on it). 

Please compile a 200KHZ i2c speed build for me. Or give me any other advice. 
Looking forward to hearing from you, guys.

jflyper, 2018-01-23:
@yetipuk 
Open up the GPS puck, take a hi-res pic of the PCB inside, the side without the antenna and upload the pic here.

yetipuk, 2018-01-23:
[image]

[242 more replies]

closed 2026-09-26 by blckmn