betaflight/betaflight #4535: FC DYSF4PRO locks up with feature LED_STRIP

AirBreak69, 2017-11-11:
In the latest builds my DYSF4PRO locks up if the feature LED_STRIP gets enabled. The LEDs are on in the locked state.
I tested the code corresponding to the "BorisB_BetaFlight" builds 1787, 1793, 1797 and 1798:
1787 was the last working one. 
To avoid some questions: I used "full chip erase" and CLI "defaults" prior restoring my CLI settings.

mikeller, 2017-11-11:
This is affecting OMNIBUSF4  as well, bisecting.

jflyper, 2017-11-11:
Strange. Master is okay here with OMNIBUSF4 with Omnibus F4 AIO (gen1).

mikeller, 2017-11-11:
Weird. The bisect came up with #4455. Mine is an early model OMNIBUSF4, and it clearly locks up with the latest master (below) and the following steps:
- reflash with 'full chip erase';
- connect, go to 'Configuration', enable 'LED_STRIP', 'Save and Reboot';
- 4 LED flashes after reboot, USB port is not recognised.

[betaflight_3.3.0_OMNIBUSF4.zip](https://github.com/betaflight/betaflight/files/1463176/betaflight_3.3.0_OMNIBUSF4.zip)

jflyper, 2017-11-11:
Try enabling SmartAudio.

mikeller, 2017-11-11:
@jflyper: Yep, that 'fixes' it.

jflyper, 2017-11-11:
@AirBreak69 You may want to test the up to date master. It will probably be 1800.

closed 2017-11-11 by jflyper

AirBreak69, 2017-11-11:
Can confirm: latest master fixes it.
Thanks to all.

jflyper, 2017-11-11:
Happy props :)

PR #4538 (2017-11-11): Prevent led strip code from calling vtx api when vtx is absent
PR status: Ready to merge

Fixes: #4535 

Current mod just prevents other `vtxCommonXXX` from being called.

~Point of consideration is whether we can ditch everything in `applyLedVtxLayer` by checking `vtxCommonDeviceRegistered` at the top. Since it returns if `!active` at line 619, this may be sensible.~

EDIT
Made it return as early as possible.