betaflight/betaflight #4548: Allow fast rearm after failsafe

danarrib, 2017-11-11:
The Standard failsafe procedure is to drop the quad (stop the motors and let it fall). But in some cases, you can regain the control of the quadcopter. 

If you takeoff and just fly above your head as high as you can until you get a failsafe, the quadcopter will eventually fall back to the radio signal radius, so this is a good opportunity to be able to rearm it to be able to control it again.

So my proposal/request is:
- After a Failsafe disarm, when RX reports signal again, and the ARM switch is still on ARM position, then arm the quadcopter again.

Steps to reproduce the problem on bench:
1) Arm your quadcopter
2) Turn off the radio transmitter
3) Quadcopter should wait a second, then disarm itself and starts beeping the failsafe beep.
4) Turn on radio transmitter again. If it display some switch position warning, just confirm or cancel to make it transmit.
5) Quadcopter will stop beeping, but it won't spin its motors, even if you slide the throttle stick to minimum.

Tested Version:
BetaFlight 3.2.1 on LUX Flight Controller, with Taranis X9D Plus and X4R-SB receiver (using SBUS).

Diff:
```
# diff
# Betaflight / LUX_RACE (LUX) 3.2.1 Oct 15 2017 / 19:39:05 (434db9a36) MSP API: 1.36
name FPV Sampa Racer
resource MOTOR 1 C09
resource MOTOR 2 C08
resource MOTOR 3 C07
resource MOTOR 4 C06

feature ANTI_GRAVITY
feature DYNAMIC_FILTER
serial 0 64 115200 57600 0 115200
serial 1 0 115200 57600 0 115200
serial 2 32 115200 57600 0 115200
aux 0 0 0 1300 2100
aux 1 1 2 1300 1700
aux 2 2 2 1700 2100
aux 3 13 1 1700 2100
aux 4 28 0 1700 2100
aux 5 35 1 1275 1675
set gyro_sync_denom = 2
set gyro_notch1_hz = 0
set gyro_notch1_cutoff = 0
set gyro_notch2_hz = 0
set gyro_notch2_cutoff = 0
set rssi_channel = 8
set dshot_idle_value = 200
set motor_pwm_protocol = DSHOT600
set current_meter = ADC
set ibata_scale = 252
set pid_process_denom = 1
profile 0

set dterm_lowpass_type = PT1
set dterm_notch_hz = 0
set dterm_notch_cutoff = 0
set i_pitch = 55
set d_pitch = 40
set i_roll = 45
set d_roll = 35
rateprofile 0

set rc_expo = 15
set rc_expo_yaw = 15
set roll_srate = 80
set pitch_srate = 80
set yaw_srate = 80
```

wind0r, 2017-11-11:
"Automatic Arming Quads" will never ever happen.
Just double switch you arming switch and you are done.

If you feel fancy and want life risky:
You can do some Taranis/OpenTx magic which will Arm the quad after telemetry is back.
DONT DO THAT!

danarrib, 2017-11-11:
I think I didn't explain myself properly. I'm not proposing automatic arm. The quad was already armed when failsafe kicked in. The quad will only stays on the state it was before lost RX signal (and only if transmitter is still transmitting ARM switch position on the same state).

Also, I tested here... Even if I quick switch the ARM switch to off and on, the quad do not arm after failsafe. I must switch it off ARM mode, then wait a second, then arm again. This second can be the difference between crash and don't crash.

wind0r, 2017-11-11:
"Stay Armed" will not happen.
Imagen you failsafes.. The quad dropped. Some Kid picked it up and you walking close to it getting in range again and Quad arms again.

As soon as your link is back you just need to switch the arm switch twice, without any big delay.

If you have failsafe issus you should not asking for adding extremly risky stuff to betaflight.
You should fix your antennas, your antenna position or simply not flying that far away.

You still can change the Sourcecode how every you want.
I am just saying that such a feature is way way way to risky to add to a firmware used by plenty of people who mostly do not even read the easiest tutorial.

danarrib, 2017-11-11:
Ok, understood. Thanks.

closed 2017-11-11 by danarrib