betaflight/betaflight #4551: Openpilot REVO (F4) only Motor 2 not working in DShot Mode

guerradr, 2017-11-11:
Motor 2 on my Hexacopter does not work in DShot Mode. But I can flash the motor controller with BLHELI to different firmwares with the BLHELI Configurator on the Motor 2 Port. I had the issue on BF 3.1.7 and upgraded now to BF 3.2.2 with the same result.
I can use all other modes exept DShot on Motor 2 Port. All other motor ports work fine with DShot and other Protocols like Oneshot and PWM. For testing I used the Motor Page in the Betaflight Configurator.

I'm using an original Openpilot REVO (F4) board.

mikeller, 2017-11-12:
That is to be expected, since motor 2 has a resource conflict with motor 6. If you want to use Dshot on a hexacopter with this board (if that is not entirely impossible due to resource constraints), you'll at least have to use a custom built firmware for this.

guerradr, 2017-11-12:
Thank you for the fast reply. In this case I will remain on Multishot.

closed 2017-11-12 by guerradr