betaflight/betaflight #4554: [Feature Request} Vertically adjustable crosshairs for OSD

chime13, 2017-11-12:
For builds that have very aggressive camera tilts (>30degrees) - What do you think about having the ability to move the OSD crosshairs vertically (center-down) either by GUI or by stick commands to better align with true level vs image center.
You could calibrate the crosshairs manually when the craft is stationary on a level surface. By doing this using OSD you have a level reference without having to enable the accelerometer for art horz.
Currently, my craft is now is looking at my face while it’s aiming at my toes - lol
It just would make it easier to fly slower for landings when the camera is tilted 45 degrees.

jflyper, 2017-11-12:
This could be very easily done.

1. Create a new config variable `ahPitchOffset` in `osdConfig_s`.
2. Add this variable to `attitude.values.pitch` when pitchAngle is calculated:
```
 512             int pitchAngle = constrain(attitude.values.pitch + osdConfig()->ahPitchOffset, -maxPitch, maxPitch);
```
3. Done

DanNixon, 2017-11-12:
Having all elements positionable (including AHI and crosshairs) was something done in #3417 (although work seems to have ceased on that...).

jflyper, 2017-11-12:
Oh, that's what he wanted to do!

icosamuel, 2018-02-26:
Will the crosshair and AH be configurable independently? I personally like my crosshair center screen, but the AH is just lying to me with my 45deg uptilt. Is this something to look forward to?

DanNixon, 2018-02-26:
@icosamuel Yes. Actually there is little stopping at least making them positionable right now. I'll take a look tonight.

PR #5326 (2018-03-03): Positionable OSD crosshairs and AHI
Fixes #4554.

Makes the position of following OSD elements configurable:
- crosshairs
- artificial horizon
- artificial horizon sidebars

Also removes the need to enable artificial horizon to have the sidebars.
closed 2018-03-03 by mikeller