betaflight/betaflight #4558: Current master breaks OSD screen

Arakon, 2017-11-12:
12c052c103c7f7a3475e0a5b0cc7de2cc53c98b7

> 
> Firmware is transmitting a different number of statistics (202) to what the configurator is expecting (12)
> decode @ osd.js:1039
> bluebird.min.js:29 Unhandled rejection TypeError: Cannot read property 'src' of undefined
>     at Object.OSD.generateTimerPreview (chrome-extension://jeepdpmfbbolmfplhocnmnpmbfhgidao/tabs/osd.js:239:40)
>     at Object.preview (chrome-extension://jeepdpmfbbolmfplhocnmnpmbfhgidao/tabs/osd.js:629:20)
>     at Object.decode (chrome-extension://jeepdpmfbbolmfplhocnmnpmbfhgidao/tabs/osd.js:1087:29)
>     at chrome-extension://jeepdpmfbbolmfplhocnmnpmbfhgidao/tabs/osd.js:1195:21
>     at r (chrome-extension://jeepdpmfbbolmfplhocnmnpmbfhgidao/js/libraries/bluebird.min.js:31:7722)
>     at i._settlePromiseFromHandler (chrome-extension://jeepdpmfbbolmfplhocnmnpmbfhgidao/js/libraries/bluebird.min.js:30:13044)
>     at i._settlePromise (chrome-extension://jeepdpmfbbolmfplhocnmnpmbfhgidao/js/libraries/bluebird.min.js:30:13847)
>     at i._settlePromise0 (chrome-extension://jeepdpmfbbolmfplhocnmnpmbfhgidao/js/libraries/bluebird.min.js:30:14548)
>     at i._settlePromises (chrome-extension://jeepdpmfbbolmfplhocnmnpmbfhgidao/js/libraries/bluebird.min.js:30:15878)
>     at r._drainQueue (chrome-extension://jeepdpmfbbolmfplhocnmnpmbfhgidao/js/libraries/bluebird.min.js:29:3148)
>     at r._drainQueues (chrome-extension://jeepdpmfbbolmfplhocnmnpmbfhgidao/js/libraries/bluebird.min.js:29:3209)
>     at drainQueues (chrome-extension://jeepdpmfbbolmfplhocnmnpmbfhgidao/js/libraries/bluebird.min.js:29:1238)
>     at <anonymous>

mikeller, 2017-11-12:
Yes, that is a known open issue, since the configurator needs a change to accommodate #4487. (For next time, please report configurator issues in https://github.com/betaflight/betaflight-configurator/issues.)

Arakon, 2017-11-12:
I had reported it here because it broke the stable configurator as well, so it I assumed it would be a firmware issue.

closed 2017-11-12 by Arakon

mikeller, 2017-11-12:
As a general remark, there is no expectation that the stable configurator should work with the latest firmware from the repository (i.e. it is not forward compatible for versions of the firmware that are built after its release). The latest configurator from the repository _should_ work with the latest firmware, except in cases like this one, where there are a couple of days of delay for changes in the firmware to be reflected in the configurator (but the CLI still works).