betaflight/betaflight #4573: drift problem with MPU6000

klaudius666, 2017-11-13:
Get drift problems on lizard95 based on MPU6000.

All is working like if accelerometer was not taken in account and after 5 min of flight acc drift become very important. if i land and arm, level is perfect and begin slowly drifting.

i took a look to function imuIsAccelerometerHealthy thinking that (cause of vibration noise) this function return false in most case and for a long period (then leveling is just trusting on gyro.... but will drift in time).

when computing accmagniture you avoid using square root 
by doing  (accx*accx+accy*accy+accz*accz)*100/(acc_1G*acc_1G), and the right formulae without optim should be
sqrt(accx*accx+accy*accy+accz*accz)*100/acc_1g. 

i' suppose that's why you compare with 81=9x9 for 90% and 121=11*11 for 110%.

I think it could be cool to be able to change this 10% tolerance on acc reading, cause perhaps on small quadcopters it could avoid this continuous drift effect.

devalos, 2017-11-18:
Running into similar left roll and forward pitch drift on two micros both running F3 boards with:

Processor: STM32F303
Sensor: SPI Sensor MPU6000

Very hopeful that the drift is just a calculation issue with the 3.2 release.

When connecting to the BF Configurator, on the setup screen the roll and pitch start off fine but over the next 3-5 min their values begin to creep up to 1-2 pitch  and 3-4 roll.  Clicking the Calibrate Accelerometer button does nothing.

I can use stick commands to issue a clibrate accelerometer command, but the drift just returns.

This behavior seems to have been introduced with 3.2.1 and is present in 3.2.2.

RipperDrone, 2017-11-19:
Have it on 1 micro with PikoBLX as well since I flashed BF3.2 - can only fly acro for now...

knockoutperformance, 2017-11-22:
Yeah, I'm drifting after a few minutes of flight. Running a dysf4pro with isolation gromets and vibration standoffs. Checked my logs, not much for vibrations. I'm going to tighten up that percentage error and test it for drift. Comparing at 9x10 and 10x11 for 90 and 110. Will also test 95 and 105

knockoutperformance, 2017-11-23:
Still eventually starts drifting in horizon mode. 90 and 110 helped, but not much. 95 and 105 not any better. Tested on an sprf3 deluxe with dshot1200 enabled also, same drift issue. I'm not using mag,GPS,sonar.

martinbudden, 2017-11-24:
@klaudius666 , @RipperDrone , can you confirm you didn't have this problem on 3.1.7?

klaudius666, 2017-11-24:
hi martin and big thx for your work on betaflight. 

No the problem was occuring with 3.1.7 too.  I was hoping it was fixed in version 3.2.1 because you changed some initializations. 
Really. I think this problem is not a faultly acc. I think that someting in scheduler or some other test avoid acc to be taken in account for too long time and the attitude is only deducted from gyro integration.

Everywhere on internet we can see a lot of people having problems with level and horizon on Lizard95 and mini quad using F3+MPU6000 or 6500. The other not complaining are just using acromode.

Just landing and rearming, then attitude is ok.... but slowly drift within a minute.
it's like if it was trimming acc during time. This never occurs on my 250 quads, event with strong vibrations...

I saw that some acc (Mpu6000) can have 512*8 (or *4 can't remember) for raw acc xyz.
it's 13 bits (if signed). 
when computing 32 bits accmagnitude (x*x+y*y+z*z) it could take 25 bits.... and the multiply by 100 can drain 7 more bits....
could'nt this lead to 32 bit overflow ? (sorry if it's crazy question cause i ve not big knowlege of acc drivers of betaflight, is 512 equivalent to 1G ? what is acc range 2G to 8G etc...? ).

Can u confirm that if function imuIsAccelerometerHealthy always return false, then level will only rely on gyro integration ? 

why does not any acc drivers return the same value (ACC_1G differs from one MPU to an other).
In my mind The drivers should return values with a known normalized range accepting maximum +-4G, it could avoid many  use of divide by acc_1G everywere

Thx

klaudius666, 2017-11-24:
512 is 512x8

RipperDrone, 2017-11-24:
Strange, got drift issues with my micro 82mm Shuriken, PIKOBLX FC. It's slow drift as you say, acro mode is perfectly fine, it's just in horizon mode. It only occurs on this micro, all other (15!) 4-6" are ok. 

Running BF3.2, I think it was ok until BF 3.0.1...

martinbudden, 2017-11-25:
It does seem possible that there could be overflow in the `imuIsAccelerometerHealthy` calcualtion.

Could someone try this PR and see if it fixes the problem https://github.com/betaflight/betaflight/pull/4657

RipperDrone, 2017-11-25:
For me it's still hard to understand why gyro readout/horizon leveling should be target specific, only occurring on some FCs and only on Micros. Or is it because Micros have higher vibration so filters act differently? (My Shuriken is very delicate structure so definitely higher vibes). Could it also be linked to filter defaults having changed since 3.0.1?

martinbudden, 2017-11-25:
@RipperDrone , a possible explanation is that smaller drones experience higher g-forces when manoeuvring and so the acc values are too high to be used more often.

RipperDrone, 2017-11-25:
Will try to re-install my build environment and try this commit out. Any recommendations on a lean build environment without virtual networks messing up my home network? (This is why I uninstalled Docker...)

RipperDrone, 2017-11-25:
Argh - seems I'm back to start with getting the build setup to work. What's missing - this is my Docker logfile...?
[BF Build Log Docker.txt](https://github.com/betaflight/betaflight/files/1503022/BF.Build.Log.Docker.txt)

RipperDrone, 2017-11-25:
Here is where the errors start...
%% (size optimised) config_streamer.c
%% (speed optimised) adc.c
%% (speed optimised) buf_writer.c
%% (speed optimised) bus.c
%% bus_i2c_busdev.c
%% (size optimised) bus_i2c_config.c
%% bus_i2c_soft.c
In file included from ./src/main/drivers/accgyro_mpu.c:38:0:
./src/main/drivers/accgyro.h:26:5: error: unknown type name 'sensorReadFuncPtr'
     sensorReadFuncPtr read;                                 // read 3 axis data function
     ^
./src/main/drivers/accgyro.h:27:5: error: unknown type name 'sensorReadFuncPtr'
     sensorReadFuncPtr temperature;                          // read temperature if available
     ^
./src/main/drivers/accgyro.h:35:5: error: unknown type name 'sensorReadFuncPtr'
     sensorReadFuncPtr read;                                 // read 3 axis data function
     ^
In file included from ./src/main/drivers/accgyro_mpu.c:45:0:
./src/main/drivers/accgyro_mpu.h:190:39: error: unknown type name 'extiConfig_t'
 mpuDetectionResult_t *detectMpu(const extiConfig_t *configToUse);
                                       ^
./src/main/drivers/accgyro_mpu.c:67:14: error: unknown type name 'extiConfig_t'
 static const extiConfig_t *mpuIntExtiConfig = NULL;
              ^
./src/main/drivers/accgyro_mpu.c:77:39: error: unknown type name 'extiConfig_t'
 mpuDetectionResult_t *detectMpu(const extiConfig_t *configToUse)
                                       ^
./src/main/drivers/accgyro_mpu.c: In function 'detectMpu':
./src/main/drivers/accgyro_mpu.c:89:5: warning: implicit declaration of function 'delay' [-Wimplicit-function-declaration]
     delay(35);
     ^
./src/main/drivers/accgyro_mpu.c: In function 'mpuIntExtiInit':
./src/main/drivers/accgyro_mpu.c:239:48: error: request for member 'tag' in something not a structure or union
     IO_t mpuIntIO = IOGetByTag(mpuIntExtiConfig->tag);
                                                ^
./src/main/drivers/accgyro_mpu.c:248:22: error: 'OWNER_MPU' undeclared (first use in this function)
     IOInit(mpuIntIO, OWNER_MPU, RESOURCE_EXTI, 0);
                      ^
./src/main/drivers/accgyro_mpu.c:248:22: note: each undeclared identifier is reported only once for each function it appears in
./src/main/drivers/accgyro_mpu.c:248:33: error: 'RESOURCE_EXTI' undeclared (first use in this function)
     IOInit(mpuIntIO, OWNER_MPU, RESOURCE_EXTI, 0);
                                 ^
./src/main/drivers/accgyro_mpu.c:248:5: error: too many arguments to function 'IOInit'
     IOInit(mpuIntIO, OWNER_MPU, RESOURCE_EXTI, 0);
     ^
In file included from ./src/main/drivers/accgyro_mpu.c:33:0:
./src/main/drivers/io.h:105:6: note: declared here
 void IOInit(IO_t io, resourceOwner_e owner, uint8_t index);
      ^
Makefile:295: recipe for target 'obj/main/PIKOBLX/drivers/accgyro_mpu.o' failed
make[1]: *** [obj/main/PIKOBLX/drivers/accgyro_mpu.o] Error 1
make[1]: *** Waiting for unfinished jobs....

mikeller, 2017-11-25:
@RipperDrone: Erm, I think you need to update your sources. PIKOBLX  was renamed to FF_PIKOBLX  a while ago.

RipperDrone, 2017-11-25:
ok, will pull complete git repo again and retry...

RipperDrone, 2017-11-25:
Darn, progress but seems the linker cannot finish / FLASH overflow. Is this version too large to fit into PIKOBLX flash memory? Plus there's some warnings related to VTX...

[Docker Build Log 2.txt](https://github.com/betaflight/betaflight/files/1503088/Docker.Build.Log.2.txt)

RipperDrone, 2017-11-25:
Where can I find which OPTIONS=<> settings each FC needs? Seems I have to 'opt out' some features to make the image fit into PIKOBLX flash?

mikeller, 2017-11-25:
> /usr/lib/gcc/arm-none-eabi/4.9.3/../../../arm-none-eabi/bin/ld: region `FLASH' overflowed by 24869 bytes

That is a lot of overflow. Are you trying to do a debug build? If so, this will probably not work for F3 because of the flash space limitation, you just about need an F4 for it.

RipperDrone, 2017-11-25:
Maybe - apologies for not being knowledgeable enough to know what exactly I'm building. I am using the Docker route, and my build batch file looks like this:

docker run -e "PLATFORM=FF_PIKOBLX" --rm -it -v //y/"Meine Backups"/_Files/GitHub/betaflight:/opt/betaflight mikeller/betaflight-build

RipperDrone, 2017-11-25:
so no OPTIONS switch except the one in your script (betaflight-build?)...?

mikeller, 2017-11-25:
Ok. For one, you are still using the old docker image from when before it was moved into the Betaflight project. Try

```
docker run -e "PLATFORM=FF_PIKOBLX" --rm -it -v //y/"Meine Backups"/_Files/GitHub/betaflight:/opt/betaflight betaflight/betaflight-build
```

RipperDrone, 2017-11-25:
Success :-) Flawless - it was due to script from old repo obviously... thx a lot, can test PIKOBLX drift now...

RipperDrone, 2017-11-25:
Finally done checks with commit https://github.com/betaflight/betaflight/pull/4657 on my PIKOBLX Shuriken 82 - drift is still there :-(

Takes some 30secs hovering in Horizon mode, and the center point is off by ~10%, continues to drift further as I keep flying. Both Pitch and Roll axes are affected.

knockoutperformance, 2017-11-25:
dysf4pro on PR #4657, seemed ok at first in the house with little movement on sticks in horzn.

disarmed, went outside, rearmed and flew. about 40 sec and two barrel rolls later(one left , one right), it was like i had a 30mph wind pushing from rear and to left.  gets worse as i countinue to fly. couple more quick flips forward, back, left, right. got real hard to fly then, i had to switch outta horizon. got up to about 100ft, hit horizon again, yee-haaaw! seemed like i had sticks at full left and forward flipping, hit acro mode and did a rough landing, clipping a prop tip.

accidently cleared blackbox, powered off, powered on, rearmed in horizon, and flew again with clipped prop to get some data again. seemed to affect pitch more with the extra vibration.

RipperDrone, 2017-11-25:
Seems to be accumulating errors in one direction per axis only - this is very consistent throughout all our tests...

RipperDrone, 2017-11-25:
Was wondering whether I should try very aggressive (low freq thresholds) or dynamic filtering. Does that make sense? At least I had filters freqs down to 50Hz, still same behaviour...

martinbudden, 2017-11-27:
@RipperDrone , could you try PR https://github.com/betaflight/betaflight/pull/4672

That should provide some improvement.

knockoutperformance, 2017-11-28:
thank you for working on this.
i recompiled and flashed. on setup page in configurator, looks like its spinning at 2000 degrees a sec in all directions, maybe faster, did not dare try to fly.

martinbudden, 2017-11-28:
@knockoutperformance , there was an error in PR #4672 . Can you try PR https://github.com/betaflight/betaflight/pull/4684 ?

RipperDrone, 2017-11-28:
@martinbudden sorry to ask but how can I pick only the individual commit or PR you referenced and do a quick compile? I always download the whole repo at the snapshot time of your commit and do a Docker compile. However, in GitKraken I see the 3.2 maintenance branch and the current master. Can I drag&drop the commit from your test branch into master and then trigger the compilation? Just trying to make this test loop faster…

Now downloading latest commit you recommended for a test on my PIKOBLX…

RipperDrone, 2017-11-28:
Hey, have tries out PR #4686 - drift seems actually fixed :-) At least in a 1min hovering before my battery went out... let's see if @knockoutperformance can confirm...

knockoutperformance, 2017-11-29:
Thank you

tested pr#4684.

tried 2khz 2khz , 4khz 4khz, 8khz 8khz, all 8 min each of hover inside the house, very small amount of drift. if any.

doesnt seem to lean after some time, which would after 2min in the house.

NEEDS to be tested outside, where i can do some flips or rolls.

way to windy outside today to play here.

martinbudden, 2017-11-29:
@RipperDrone , @knockoutperformance , I've merged the fix into the master codeline, since it certainly is better than the current implementation.

Can you test outside and report back if any problems persist?

hydra, 2017-11-29:
@martinbudden great to see progress on this issue which has been a problem for quite some time now.

klaudius666, 2017-11-29:
@martinbudden .great job. Very happy if this issue is gone. got a look at the code you added about this martin. Is it supposed to enhance gyro datas too ? (as i see you do same accumulation on gyro datas too) 

Will test that in next RC on lizard95.
Thx

RipperDrone, 2017-11-29:
THANK YOU, too - it's very stable now :-)

martinbudden, 2017-11-29:
@knockoutperformance , @RipperDrone , good to hear it is working well now. The code has been merged into master, so will be in the 3.3. release.

ledvinap, 2017-11-29:
@martinbudden : Well, I was wrong ... But I'm spill persuaded that there must be something wrong hidden in the code, averaging alone should not make that big difference ...

martinbudden, 2017-11-29:
@ledvinap , I think it is more the improved integration of the gyro rate rather than the averaging of the acceleration value that has made the difference.

ledvinap, 2017-11-29:
@martinbudden : Yes, acceleration is heavily filtered in IMU (few seconds time constant).
But gyro signal is low-pass filtered before integration (and before summing). The should be little aliasing caused by IMU downsampling.  And it is strange that described offset is long-term and in same direction.

Does anyone have blackbox data (ideally with gyro debug enabled) with #4684 applied?

ledvinap, 2017-11-29:
@martinbudden : One change is that sampling time is calculated on gyro read, not on IMU invocation ...

knockoutperformance, 2017-11-29:
@levinap i can try tomorrow afternoon with gyro debug on if it doesn't rain here. Did you want any filters enabled/disabled?

I did notice today, that  if I set to .5khz sampling, in the sensor gui tab gyro data is offset around 10.00-13.00, instead of 1.00 to -1.00. Like the scaling is off?

martinbudden, 2017-11-29:
> One change is that sampling time is calculated on gyro read, not on IMU invocation ...

Yes, I think that's the significant change. It means gyro is integrated with dt=125us, rather than dt=10000us, a hugely significant difference

ledvinap, 2017-11-30:
@martinbudden : Thanks. I totally overlooked that IMU is running only on 100Hz. All my arguments assume that IMU downsampling is possible - IMU update rate is 2x highest frequency in data (Nyquist).

Averaging/summing will provide necessary low-pass filtering, but there may be filters with better performance.
Also IMU performance at this frequency may be noticeably affected (20 deg/update in extreme case is a bit too much for linear approximation), increasing update frequency a bit may be beneficial.

knockoutperformance, 2017-12-01:
@martinbudden There is something goofy still happening, but no where near as it was before. I was able to fly today with little wind. if i quick  flip in one direction, it will drift that direction until i quick flip the opposite direction, then its level again(roll and/or pitch). 

i zipped the cli dump and datalog with gryo debug on

[testing_drift.zip](https://github.com/betaflight/betaflight/files/1522393/testing_drift.zip)

knockoutperformance, 2017-12-01:
here is another where at one point it seemed like it thought it was upside down on a decent, flipped and ended up pointed in opposite direction on its own by just touching throttle.
[BTFL_BLACKBOX_LOG_butters_20171201_095922.zip](https://github.com/betaflight/betaflight/files/1522476/BTFL_BLACKBOX_LOG_butters_20171201_095922.zip)

fiveangle, 2017-12-02:
Still major ACC drift on mini F3 20mm clone "OMNIBUS" (Fortinet v1.0) target that came in [ARFun95](https://www.banggood.com/Arfun-95-Pro-95mm-F3-OSD-5_8G-40CH-25mW-FPV-Racing-Drone-w-600TVL-CMOS-Camera-PNP-BNF-p-1218480.html) running https://github.com/borisbstyle/betaflight/commit/dd3a40f036692cbbaad3279187eb84a0a3c710fa.  I keep meaning to try and add an OpenLog device to capture BB logs but haven't got around (kept hoping a breakthrough would surface from others).  As it seems these mini F3 clones are the lions share of these "drift" issues, I wonder if we're fighting a janky MPU6000 HW clone problem.

Will try to get to it someone soon.

martinbudden, 2017-12-02:
@knockoutperformance , the bias after a flip is interesting.

My immediate thought is that it is due to using the rectangular rule for numeric integration of the gyro values. The rectangular rule overestimates the area under the curve when the gyro rate is increasing and underestimates the area under the curve when the gyro rate is decreasing, so leaves a bias after a increase/decrease cycle. This bias could be largely eliminated by using the trapezium rule for integration. @ledvinap - your thoughts?

Anyway, I'll make a PR in the next few days and we can test this hypothesis.

martinbudden, 2017-12-02:
@fiveangle , it's possible it's a dodgy gyro, but I think it's more likely to be other things. Small quads are inherently noisier than larger quads and noise causes drift. Some questions:

1. Is the FC soft mounted?
2. What are your gyro filter settings? Have you tried increasing them?

And yes, getting the blackbox logs is essential for proper analysis.

fiveangle, 2017-12-02:
The gyro appears to work perfectly (acro mode is fine) but only self-level mode is unusable as it will drift away from true normal force direction within seconds of launch and continue to get worse.

Quad is minimal 90mm frame with FC direct mounted via nylon standoffs and integral to stucture holding top plate on so definitely suspecpitlble to full vibration.

[image]

Drift in self-level mode occurs using stock BIQUAD gyro filters with or without dynamic filter enabled (as well as PT1).  P&R PIDs on this small X are square at 32-20-23.  I'm not familiar with how the gyro interacts with self-level mode.  I thought the accelerometer was calibrated and then was supposed to output delta vector from the calibration vector stored in the chip.  It's obviously not doing that correctly, or the output is being misinterpreted.  Can BB log the raw accelerometer data ?  If so, that would at least tell us if the data is bad or if BF is accumulating the error.

I can't seem to get my old crusty OpenLOG device to accept the new blackbox sketch so this may be a lost cause.  It's running the very first original blackbox INO from @nicthedude when he originally got BB working (was installed on my Flip32+/Blackout).  I'll see if it will work but as I understand it, it's running very low UART speed so may not be able to capture with enough resolution to tell us what's up.

fiveangle, 2017-12-03:
tested 3.1.5 and same drift issue

basicjump, 2017-12-03:
Hello to all. Have the same problem with accel drift on Lizard95. But get another one interesting watch: open the "sensors" graphs at BF and watch accel output. Then try to change something to get buzzer "beep!". When buzzer works I see changes accel output.

ledvinap, 2017-12-03:
@martinbudden : I don't think integration is problem - with averaging, value used in integration is average value of previous interval, which is IMO equivalent to trapezoidal rule.

It's quite hard to study logs without ground truth. @knockoutperformance - would it be possible to upload video too?

One thing I noticed in logs is that acceleration is outside of accepted range (0.90g - 1.10g) for quite long time - accelerometer reading is ignored and attitude update is based only on gyro integration.

ledvinap, 2017-12-03:
It may be best to drop Mahony and port some Kallman based attitude estimator ...

knockoutperformance, 2017-12-03:
@ledvinap I do not have an on-board camera system. I could have the wife use her dslr camera to video me flying, if that would work? She will not let me strap her $1k camera to my drone, i asked, lol. Or if you got an better idea..

ledvinap, 2017-12-04:
@knockoutperformance : Filming it would be great. Please try to record arming beep if possible.

The acc range is probable cause. You may try keeping quad steady for few seconds after flip, it may eventually resync attitude estimator ...

ledvinap, 2017-12-04:
@martinbudden : It would be interresting to integrate https://github.com/sfwa/ukf/blob/master/examples/ahrs/ahrs.h into blackbox tools - it will provide alternative attitude estimate for comparison ...

DieHertz, 2017-12-04:
Interesting, I'll look into it

basicjump, 2017-12-04:
https://youtu.be/QuKoOdPDFpE
How buzzer affects on accel

[86 more replies]

closed 2018-10-20 by stale[bot]