betaflight/betaflight #4574: Custom Airplane seems to be the only mixer that uses smix and sends servo pulses

taileron, 2017-11-13:
Betaflight feels absolutely great for flying single copters (4 servos) but activating baro mode controls pitch instead of throttle is there a trick to change it ? (BF 3.22 Oversky Micro 32 F1 FC )

mikeller, 2017-11-13:
There is very little testing done for non-multirotor mixer types in Betaflight, so this is not really surprising, unfortunately. Any help and contribution of fixes for this is appreciated!

taileron, 2018-01-16:
... still lots of fun but I couldn´t find a way to tune yaw pid for windy situations when it goes into severe yaw oscillation so there is a need to increase yaw d-term significantly ... where is that currently hidden cause the cli yaw-d parameter doesn´t work? (roll and pitch was tuneable perfect with high d values)

mikeller, 2018-01-16:
Betaflight currently does not support d-term for yaw, as it was found to be unneeded for multirotors. From flying tricopters that tend to be affected by yaw oscillations, I found that adjusting `acc_limit_yaw` down can counteract this to some extent.
As an alternative, iNav does have a lot more focus on fixed wing craft.

taileron, 2018-01-16:
tried iNav 1.3 .... 1.8 and older Cleanflight (1.14.2 MW PID) already having yaw relatively tuned but absolutely no flying fun with roll and pitch stability especially at output saturation and when flying in stronger wind.
inav does great and easy missions and gps rth and flying forward at no wind but absolutely not fun in acro and horizon modes.

mikeller, 2018-01-16:
I don't know if there is a substantial reason not to put the yaw d-term back into the calculations, and just default it to 0. After all, even the parameter to set it is still there. @martinbudden what do you think?

taileron, 2018-01-23:
it´s solved, there is a very good tolerance for extremely high i-terms in yaw, so lowering p and set iterm limit above 300 so only a bit slower reaction on yaw is left cause roll and pitch benefits from dterm_setpoint_weight >2. 
Baro-mode works by replacing fw althold with mc althold code in altitude.c

taileron, 2018-03-27:
Tried to test it in a duo-copter using differential thrust, works great in hovering but as soon as there is higher airspeed, all pids need to be reduced by 5 ... 10 times TPA does this partially but in hover the pid gains should be high with upper throttle and at increasing airspeed lower pids are needed even with low throttle. Is there a trick to use TPA with an aux channel instead of throttle or an alternative way to inflight reduce all pids to a defined value?

taileron, 2018-04-05:
that seems to work very good:
float auxPDgain = (((float)rcData[AUX5])/1000.0f)-1.0f;
float auxIgain = (0.5f+auxPgain)/1.5f;

as an additional multiplicator like tpafactor 0 ...1 into pid.c line 488, 538 and .33 ... 1 for i-term 595 to control PID gains via aux channel

taileron, 2018-05-02:
anyway it would be great to add temporally a yaw d value. Which line in pid.c can be changed to add a higher d component to yaw (hardcoded) 
... was not able to get it running anymore with calculating a yaw d component enabled

jflyper, 2018-05-03:
I think this one has a impact on what you are doing here:
[GPS RESCUE] added gps for altitude estimation #5764

taileron, 2018-05-03:
All about althold is solved now, but i have oscillations before yaw P and I gains come closer to the perfect position. My vehicle flies with nose 90 degree up so to be able to use angle mode yaw has to be roll and finally yaw is the most complex to stabilize axis for this vehicle cause different airspeeds change the whole situation e.g. pitch it´s relatively easy to tune with a very high d-term

closed 2018-06-09 by stale[bot]