betaflight/betaflight #4576: beeper_dshot_beacon_tone command not working

1banger, 2017-11-14:
"beeper_dshot_beacon_tone" command not working. 

Using CL Racing F4S FC. 
4in1 Blheli_32 esc.
FW is betaflight_3.2.2_CLRACINGF4.hex

``# version
# Betaflight / CLRACINGF4 (CLR4) 3.2.2 Nov 11 2017 / 00:24:23 (77782e602) MSP API: 1.36

# name

# resources

# mixer

# servo

# servo mix

# feature
feature TELEMETRY
feature AIRMODE
feature ESC_SENSOR
feature DYNAMIC_FILTER

# beeper

# map
map TAER1234

# serial
serial 3 2048 115200 57600 0 115200
serial 5 64 115200 57600 0 115200

# led

# color

# mode_color

# aux
aux 0 0 0 1300 2100
aux 1 13 1 1300 2100
aux 2 27 3 1300 2100

# adjrange

# rxrange

# vtx

# rxfail

# master
set gyro_notch1_hz = 0
set gyro_notch1_cutoff = 0
set gyro_notch2_hz = 0
set gyro_notch2_cutoff = 0
set rssi_channel = 16
set serialrx_provider = CRSF
set dshot_idle_value = 600
set motor_pwm_protocol = DSHOT1200
set bat_capacity = 1300
set beeper_dshot_beacon_tone = 1
set tlm_halfduplex = OFF
set osd_units = IMPERIAL
set osd_vbat_pos = 2433
set osd_rssi_pos = 2074
set osd_tim_1_pos = 54
set osd_tim_2_pos = 2455
set osd_flymode_pos = 333
set osd_throttle_pos = 225
set osd_vtx_channel_pos = 377
set osd_ah_sbar = 200
set osd_current_pos = 385
set osd_mah_drawn_pos = 353
set osd_craft_name_pos = 362
set osd_gps_speed_pos = 218
set osd_gps_lon_pos = 82
set osd_gps_lat_pos = 65
set osd_gps_sats_pos = 51
set osd_home_dir_pos = 302
set osd_home_dist_pos = 303
set osd_compass_bar_pos = 266
set osd_altitude_pos = 247
set osd_pid_roll_pos = 423
set osd_pid_pitch_pos = 455
set osd_pid_yaw_pos = 487
set osd_debug_pos = 1
set osd_power_pos = 321
set osd_pidrate_profile_pos = 345
set osd_warnings_pos = 329
set osd_avg_cell_voltage_pos = 76
set osd_pit_ang_pos = 257
set osd_rol_ang_pos = 289
set osd_battery_usage_pos = 392
set osd_disarmed_pos = 138
set osd_nheading_pos = 311
set osd_nvario_pos = 279
set osd_esc_tmp_pos = 82
set osd_esc_rpm_pos = 83
set vcd_video_system = 2

# profile
profile 0

set dterm_lowpass_type = PT1
set dterm_notch_hz = 0
set dterm_notch_cutoff = 0

# rateprofile
rateprofile 0

# ``

Can see beeper switch working in configurator. Not sure what else to do. I have set this up before
with no issues on other FC's

mikeller, 2017-11-14:
What is the firmware version on the BLHeli32 ESCs?

1banger, 2017-11-14:
32.1. reverted back to 32.0 with no change in issue

mikeller, 2017-11-14:
That won't work then, DShot commands require 32.12 or better.

1banger, 2017-11-14:
Latest BLHeliSuite_32 only has 32.1 as latest available FW. And that does not work either. Can't find anything newer than 32.1

mikeller, 2017-11-14:
Ask the supplier of your ESCs then, or ask on https://github.com/bitdump/BLHeli/issues.

NectarCollector, 2017-11-16:
@1banger BLHeli_32 has the FW bundled with the configurator. If you want to update to 32.2, the 32.2 BlHeli Software will come bundled with the 32.2 hexes.

See post #1879 for the 32.3 configurator, note, still in testing: https://www.rcgroups.com/forums/showthread.php?2864611-BLHeli_32-Power-to-perform/page126#post38505701