betaflight/betaflight #4578: Magnetic Declination removed

McGiverGim, 2017-11-14:
In this PR, the correction of the magnetic declination was removed, and there was a discussion about removing the magnetic declination:
https://github.com/betaflight/betaflight/pull/3799
The problem was that it seems it was using the local variable magneticDeclination (that it is never configured) but the reality, is that it needs to use the magneticDeclination of mag_t (compass.h) that is configured from the config, if I'm not wrong:
https://github.com/betaflight/betaflight/blob/971abbd9f64c2e9504ec33c9c27246fdccc55ca2/src/main/sensors/compass.c#L252-L254
I don't know nothing about MAG so, am I wrong?
If I'm not wrong, the only thing needed to make it "work" is add the magnetic declination from compass to the YAW value in the `imuUpdateEulerAngles()` function. Is that right?

ledvinap, 2017-11-14:
@McGiverGim : IMO magneticDeclination should be added to (2d projected) compass reading, before using it to adjust attitude estimate
https://github.com/betaflight/betaflight/blob/971abbd9f64c2e9504ec33c9c27246fdccc55ca2/src/main/flight/imu.c#L285
That way rotation matrix will be correct. Applying it to Euler angles only will result in inconsistency between matrix and EA, possibly leading to problems

McGiverGim, 2017-11-14:
I thinked that the "attitude yaw" will be the place to add the magnetic declination, this is what is readed to show the compass in osd, for example:
https://github.com/betaflight/betaflight/blob/971abbd9f64c2e9504ec33c9c27246fdccc55ca2/src/main/flight/imu.c#L367-L377

ledvinap, 2017-11-14:
There are three attitude representation on CF - orientation quaternion, orientation matrix (directly derived from quaternion and used in quaternion update) and Euler angles. Getting them out of sync will cause subtle problems and hard-to-find bugs (for example compass shows north relative to HEADFREE reference)

Also magnetic declination is difference between true and magnetic north, IMU update expects true north as input (or body frame is assumed relative to true north), so correction should be applied first.

McGiverGim, 2017-11-14:
Understood, thanks for the explanation. I'm not able to reach at this level of complexity so this must be left to other with more knowledge.

You can freely close the issue or let it open. Thansk again!

ledvinap, 2017-11-14:
@McGiverGim : It's simple - [hx,hy,0] is magnetic vector rotated/projected to global ground frame (north-east-up). All that is necessary is to rotate [hx,hy] by magneticDeclination befor computing `ez_ef`.
Storing sin(declination)/cos(declination) is probably easiest implementation.

Another possibility is to use magnetic declination as reference for computing correction error ([mx,my,0] pointing to magnetic north), but with that change to error calculation will be desirable  - use unit size vector [mx,my,0] instead of vector size derived from [hx,hy]. This is probably desirable - currency magnetic correction scales with square of projected vector size (`|[hx,hy,0]|`), scaling linearly with projected vector  size will probably provide better results.

Then it will be `[bx, by, 0]` = `[cos(declination), -sin(declination), 0]` (I hope i got declination direction right) and `ez_ef` will need more terms of cross product:
`ez_ef = hx*by-hy*bx`

McGiverGim, 2017-11-14:
Hehehe, is great that you think is simple 😨 

I finished the university 15 years ago, and my work has been far from mathematics since then 😄  I don't remember almost anything about vectors, etc. I will try to understand your explanation, but I don't promise nothing 😋